SpenceKonde/megaTinyCore #534: Wire Library Issue - Slave Hang

tetsudou, 2021-10-03:
Good time of day Spence!

I ahve added this to discussions, but probably this is a better place.

I have a rather weird bug going on in my I2C installation. I have several (8) ATTiny1604 slaves connected to an Arduino Nano master. Three of them run Code A and the rest run Code B (the difference is the peripheral it drives). What happens is that randomly, when the master is attempting to contact B-type slaves, one of these slaves would cut out after receiving the R/W bit and pull both SDA and SCL low as shown in the picture (yellow = SCL, purple = SDA). It would then proceed to completely hang, blocking the bus.

I am using external pull-ups (560 Ohm to ensure sharp edges).

[image]

If only A-type slaves are contacted (even with B-type slaves on the line), the code runs fine indefinitely (at least for a couple of hours). If at least one slave running Code B is contacted, it invariably eventually causes the line to cut out. 

I have tried versions 2.3.2, 2.4.2 and 2.4.3 of the core. On the former two, the failure happens after some time (after 50-70 messages). On the latter, it always happens on the first or the second message.

What could be the reason for this behaviour? 

Thanks in advance!

Code_A.ino:

```
#include <Wire.h>
#include <Servo_megaTinyCore.h>

Servo myservo;

int address = 2;//different for each slave

int pos = 42;
int servoPin = 10;
int lightPin = 1;
int lstate = 1;
int px = 1;
int x = 0;
char message[10];
boolean action = true;

void setup() {
  for (int i = 0; i < 10; i++) {
    message[i] = '0';
  }
  delay(2000);
  pinMode(lightPin, OUTPUT);
  digitalWrite(lightPin, 1);
  myservo.attach(servoPin);
  myservo.write(42);
  while (pos > 15) { //15
    pos--;
    myservo.write(pos);
    delay(10);
  }
  myservo.detach();
  Wire.setClock(100000);
  Wire.begin(address);                // join i2c bus with address #8
  Wire.onReceive(receiveEvent); // register event
  //Serial.begin(9600);           // start serial for output
}

void loop() {
  if (px == 1) digitalWrite(lightPin, lstate);
  else digitalWrite(lightPin, lstate);
  if (px != x) {
    myservo.attach(servoPin);
    if (x == 1) {
      while (pos < 75) { //75
        pos++;
        myservo.write(pos);
        delay(10);
      }
    }
    else if (x == 0) {
      while (pos > 15) { //15
        pos--;
        myservo.write(pos);
        delay(10);
      }
    }
    myservo.detach();
    px = x;
  }
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany) {
  int i = 0;
  while (0 < Wire.available()) { // loop through all but the last
    char c = Wire.read(); // receive byte as a character
    message[i] = c;
    i++;
  }
  if (message[0] == 'l') {
    if (message[1] == 0) digitalWrite(lightPin, 0);
    else if (message[1] == 1) digitalWrite(lightPin, 0);
    lstate = message[1];
  }
  else if (message[0] == 's') {
    x = message[1];
  }
}
```

Code_B.ino:
```
#include <Wire.h>

int address = 8;//different for each slave

int red = 10;//Red: C = 30uF
int amb = 8;//Amber: C = 30uF
int grn = 0;//Green: C = 2uF
int call = 1;//same as yellow
int red2 = 9;//Red: C = 30uF
int amb2 = 4;//Amber: C = 30uF
int grn2 = 2;//Green: C = 2uF
int call2 = 3;
char message[10];
boolean action = false;

void setup() {
  delay(1000);
  for (int i = 0; i < 10; i++) {
    message[i] = '0';
  }
  pinMode(red, OUTPUT);
  digitalWrite(red, HIGH);
  pinMode(red2, OUTPUT);
  digitalWrite(red2, HIGH);
  Wire.setClock(100000);
  TWI0.SADDR = address;
  Wire.begin(address);                // join i2c bus with address #8
  Wire.onReceive(receiveEvent); // register event
}

void loop() {
  pinMode(red, OUTPUT);
  digitalWrite(red, HIGH);
  delay(500);
  pinMode(red, INPUT);
  digitalWrite(call, LOW);
  delay(500);
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany) {
  int i = 0;
  while (0 < Wire.available()) { // loop through all but the last
    char c = Wire.read(); // receive byte as a character
    message[i] = c;
    i++;
  }
  //if (message[0] == '1') {
  if (message[0] == '1') {
    digitalWrite(grn, LOW);
    pinMode(grn, INPUT);
    digitalWrite(amb, LOW);
    pinMode(amb, INPUT);
    digitalWrite(call, LOW);
    pinMode(call, INPUT);
    delay(300);
    pinMode(red, OUTPUT);
    digitalWrite(red, HIGH);
  }
  else if (message[0] == '2') {
    digitalWrite(grn, LOW);
    pinMode(grn, INPUT);
    digitalWrite(amb, LOW);
    pinMode(amb, INPUT);
    delay(300);
    pinMode(red, OUTPUT);
    digitalWrite(red, HIGH);
    pinMode(call, OUTPUT);
    digitalWrite(call, HIGH);
  }
  else if (message[0] == '3') {
    digitalWrite(grn, LOW);
    pinMode(grn, INPUT);
    digitalWrite(red, LOW);
    pinMode(red, INPUT);
    digitalWrite(call, LOW);
    pinMode(call, INPUT);
    delay(300);
    pinMode(amb, OUTPUT);
    digitalWrite(amb, HIGH);
  }
  else if (message[0] == '4') {
    digitalWrite(amb, LOW);
    pinMode(amb, INPUT);
    digitalWrite(red, LOW);
    pinMode(red, INPUT);
    digitalWrite(call, LOW);
    pinMode(call, INPUT);
    delay(300);
    pinMode(grn, OUTPUT);
    digitalWrite(grn, HIGH);
  }
  //}
  //else if (message[0] == '2') {
  else if (message[0] == '5') {
    digitalWrite(grn2, LOW);
    pinMode(grn2, INPUT);
    digitalWrite(amb2, LOW);
    pinMode(amb2, INPUT);
    digitalWrite(call2, LOW);
    pinMode(call2, INPUT);
    delay(300);
    pinMode(red2, OUTPUT);
    digitalWrite(red2, HIGH);
  }
  else if (message[0] == '6') {
    digitalWrite(grn2, LOW);
    pinMode(grn2, INPUT);
    digitalWrite(amb2, LOW);
    pinMode(amb2, INPUT);
    delay(300);
    pinMode(red2, OUTPUT);
    digitalWrite(red2, HIGH);
    pinMode(call2, OUTPUT);
    digitalWrite(call2, HIGH);
  }
  else if (message[0] == '7') {
    digitalWrite(grn2, LOW);
    pinMode(grn2, INPUT);
    digitalWrite(red2, LOW);
    pinMode(red2, INPUT);
    digitalWrite(call2, LOW);
    pinMode(call2, INPUT);
    delay(300);
    pinMode(amb2, OUTPUT);
    digitalWrite(amb2, HIGH);
  }
  else if (message[0] == '8') {
    digitalWrite(amb2, LOW);
    pinMode(amb2, INPUT);
    digitalWrite(red2, LOW);
    pinMode(red2, INPUT);
    digitalWrite(call2, LOW);
    pinMode(call2, INPUT);
    delay(300);
    pinMode(grn2, OUTPUT);
    digitalWrite(grn2, HIGH);
  }
  //}
}
```

SpenceKonde, 2021-10-04:
Those calls to delay - that's not being handled *IN THE ISR* is it? It kind of looks like it is.... if so that is your problem; ISRs should run as quickly as humanly (well, AVRly) possible, you set flags in the ISR, do the time-sensitive stuff and get out, , and then you can handle the command when you don't have the rest of the devices on the I2C bus staring at you glaring at you and pointing to their watch.  

Delay should absolutely never be used in any code that will execute as an ISR. That is sure to cause problems (this is far from the only one if I'm understanding what it does... I can name a few other problems you may not have noticed yet,....)

tetsudou, 2021-10-04:
Wow thanks for a quick reply! Yes, I wouldn't call myself the most competent programmer, so thank you 😅 And yeah any other issues would be good!

With regards to the issue though, would that have an effect on the hang? I have always imagined the data gets stored in a register, and the ISR then pulls it from there when called.

Speaking of those delay statements on the ISR, the reason I have put that code there is because I have tried methods to put that code into the main loop and failed. For example, the ISR would set `action = true`. The loop would only execute if `action` is true and set it to false at the end. Another was to check if the message has updated in the loop and to execute the code (similar to code A). Any of these result in the slave failing to respond to I2C completely or hang. In fact, the reason Code A's loop has the meaningless two lines at the front is because without them the code never executes for some reason.

Thanks in advance! Sorry if it's something really dumb 😅

SpenceKonde, 2021-10-04:
Is the variable you had it setting declared volatile? A variable changed by
isr buy read outside of it must be declared volatile, otherwise the
compiler can optimize away reads or turn it into a constant if "reachable"
code (it doesn't understand when interrupts can change what can and can't
run) never changes it. By declaring it volatile you force the compiler to
check the actual memory location every time it is read or written (that is
how all the special function registers are declared to it as well,
otherwise their access could also get optimized away
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***