SpenceKonde/megaTinyCore #540: attiny402 device signatures. Missing pymcuprog device file and boards.txt entry for existing workaround

MadGarra, 2021-10-30:
Hi, 

    I know it is an old issue, and probably no affecting too much users, but just in case you like to take a look at it...

    I'm using release 2.4.2, and found two problems when trying to use some old Attiny402 with the worng signature. 
    The first is related with the compillation. Apparently, there is a missing entry in the boards.txt file, 8-pin non-bootloader ###chip section. --> "atxy2.menu.chip.402w.build.attiny" is missing. It is correctly included in the optiboot section (atxy2o.menu.chip.402w.build.attiny=402)
    After adding atxy2.menu.chip.402w.build.attiny=402 I was able to compile the code without problems

SpenceKonde, 2021-11-02:
Good catch. I never got any of the bad chip and it also looks like they aren't in the CI rotation either.

MadGarra, 2021-11-08:
I ordered them on Nov/2019 for “evaluation”, as an alternative to Attiny25/85, but never really used. Now, with the silicon supply problems I was planning to give them a second chance.
	
By the way, if you are interested, I can send you a pair just in case you like to add them to a “Freak” silicon collection :-)

closed 2021-11-09 by SpenceKonde

MATTERealise, 2022-06-17:
I'm head-bashing my way through bringing up a custom CH340 UPDI programming board. It seems like https://github.com/SpenceKonde/megaTinyCore/blob/1203928c7510e8dcc825cc2f8469e960d5efd82e/megaavr/tools/libs/pymcuprog/deviceinfo/devices/attiny402w.py#L83 is still showing the correct (and therefore wrong for this workaround) device ID.

commit 1203928c (2021-11-09): Fix #540, and names of exported map and lst.
commit c26bd2f0 (2022-06-17): fix #540 for t402w (wrong sig t402)

I'm amazed these chips are still making the rounds.... Microchip shipped them like 5 years ago. 
Note that some bogus 8-pin tinyAVRs are also making the rounds on aliexpress; they read as a 20 pin part - but unlike the supported wrong-sig parts, they have other bugs and are not sold by legitimate distributors. Those are counterfeit and are not supported nor will they be.