SpenceKonde/megaTinyCore #545: attiny 412 programming: UPDI initialization failed.

phil-barrett, 2021-11-09:
Arduino 1.8.15, Win 10, megaTinyCore 2.4.2 installed via board manager.

Programmer Selected: SerialUPDI with serial 4.7K or diode 230400 baud
Programmer is CH340, first version. 470 ohm resistor on UPDI pin.  diode between TX and RX. I don't know how to verify my serial programmer works correctly. I verified that the diode orientation is correct.  It does pass the "loopback" test without the diode. 
Target board is my own design.  Works with Atmel ICE.
Error trace back from Arduino: (sorry for the crappy formatting, Arduino IDE kinda sucks.)

> cmd /C "C:\\Users\\philba\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-nm" --numeric-sort --line-numbers --demangle --print-size --format=s "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_642728/Blink.ino.elf" > "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_642728/Blink.ino.map"
"C:\\Users\\philba\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-size" -A "C:\\Users\\philba\\AppData\\Local\\Temp\\arduino_build_642728/Blink.ino.elf"
Sketch uses 632 bytes (15%) of program storage space. Maximum is 4096 bytes.
Global variables use 10 bytes (3%) of dynamic memory, leaving 246 bytes for local variables. Maximum is 256 bytes.
C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1/python3 -u C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2/tools/prog.py -t uart -u COM5 -b 230400 -d attiny412 --fuses 2:0x02 6:0x04 8:0x00 -fC:\Users\philba\AppData\Local\Temp\arduino_build_642728/Blink.ino.hex -a write 
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.0 - June 2021
Using serial port COM5 at 230400 baud.
Target: attiny412
Set fuses: ['2:0x02', '6:0x04', '8:0x00']
Action: write
File: C:\Users\philba\AppData\Local\Temp\arduino_build_642728/Blink.ino.hex
Traceback (most recent call last):
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2/tools/prog.py", line 282, in <module>
    main()
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2/tools/prog.py", line 131, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2/tools/prog.py", line 198, in pymcuprog_basic
    args_start)
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\tools\libs\pymcuprog\pymcuprog_main.py", line 544, in _start_session
    backend.start_session(sessionconfig)
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\tools\libs\pymcuprog\backend.py", line 362, in start_session
    sessionconfig.interface_speed)
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\tools\libs\pymcuprog\programmer.py", line 83, in setup_device
    options=self.options)
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\tools\libs\pymcuprog\nvm.py", line 42, in get_nvm_access_provider
    accessprovider = NvmAccessProviderSerial(transport, device_info, baud=frequency)
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\tools\libs\pymcuprog\nvmserialupdi.py", line 53, in __init__
    self.avr = UpdiApplication(port, baud, self.dut)
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\tools\libs\pymcuprog\serialupdi\application.py", line 79, in __init__
    datalink.init_datalink()
  File "C:\Users\philba\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2\tools\libs\pymcuprog\serialupdi\link.py", line 44, in init_datalink
    raise PymcuprogError("UPDI initialisation failed")
pymcuprog.pymcuprog_errors.PymcuprogError: UPDI initialisation failed
pymcuprog.pymcuprog_errors.PymcuprogError: UPDI initialisation failed

SpenceKonde, 2021-11-09:
Is that diode a schotky? If not, a 4.7k resistor will work better - I could
never make it work with a silicon diode.  Most ch340 adapters have a
resistor in series with the TX pin so if yours does, try without the 470
ohm resistor (which is not in any way required for serial UPDI - all we
need is at least 470-1k total between the two, and I keep starting to think
I need to lower the value of the series resistor for general use.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

phil-barrett, 2021-11-09:
It was a 1N4148 like the UPDI docs call out.  But, that isn't a schottky by the way.  It's just a fast switching silicon diode.  I do have some real schottky diodes I could use.  But, it sounds like the 4.7K resistors works so I'll go with that. Put it between RX and TX on the serial adapter, right?

Thanks for getting back to me.  I know you've got lots on your plate and appreciate your attention.

SpenceKonde, 2021-11-09:
ARGHHHHH which document still has the wrong partnumber? I thought I went
through and fixed that everywhere when I realized I'd screwed up on the
part number....

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

phil-barrett, 2021-11-09:
It's [here](https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md).  

Tell you what, once I get it working with the resistor, I will try out a schottky diode and let you know.

SpenceKonde, 2021-11-10:
Thanks - fixed.
If you have a 'scope, it is much easier to debug - generally the cause of
programming problems is that the balance of "pulling line high" and
"pulling line low" is wrong and either the serial adapter isn't able to
pull the updi line low enough for the target to recognize a low, or the
target isn't able to pull it low enough for the adapter to recognize a high
- both of which are very easy to see on a scope. With the 4.7k resistor, it
*barely* works with a 470 ohm, series resistor which is why I use and
recommend the schottky diode. (the series resistor is needed for the
simplest jtag2updi programmer from nano, because a lot of those get built
with no series resistor of their own and that really shouldn't be connected
directly because it's possible in theory for the two to be out of sync and
fight over the line.

Just like every other programming issue, the errors you get back are
practically useless. All it can see is "Huh, I didn't get the response I
expected", just like the STK500 sync error for bootloader uploads. There's
no information there, because all it's getting is what character the serial
adapter sees. We could do a better job with the debugging messages, as
there is one piece of information we do get - we can tell if we're seeing
the character we sent... but the code is a bit of a dumpster fire (it's
derived from pymcuprog, which is written by people at Microchip who don't
know python either) - it's agonizing to work with and when I talked about
the verbosity of the verbose mode the guy who was the python expert on the
project still hadn't accepted that pymcuprog's promise "It'll be updated
and tested by Microchip, it'll always be up to date and..." was completely
empty. The version as received uploads at 500 bytes/second, 1k at most, I
had to hack up the link layer unrecognizably to get good performance, so
it's not mergable anyway, but I just had to manually synthesize device
specs for parts that had been out for almost a year so it's not being
updated either. And they just don't care about performance in house, and/or
are using some other "serial adapter" (MCU-as-serial-adapter I think) that
has less of a latency problem. Probably a mixture of the two.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

SanFable, 2021-11-17:
Hello,

I have similar problem using new Attiny3226 using working arduino nano setup as a jtag2updi (I have bracket and on other attiny it works).

```
avrdude: AVR Part "attiny3226" not found.

Valid parts are:
  uc3a0512 = AT32UC3A0512
  c128     = AT90CAN128
  c32      = AT90CAN32
  c64      = AT90CAN64
  pwm2     = AT90PWM2
  pwm216   = AT90PWM216
  ...
```

Any solution?

SpenceKonde, 2021-11-18:
Attiny3226 support requires 2.4.3 (github) avrdude.conf for uploads through
avrdude.  Uploads using the recommended upload tool, serialupdi, should be
fine

 There are two major changes that are half implemented in the core itself
totally unrelated to that so it is awkward for me to do a release and I
have a ton a documents that refer to 2.4.3 as being the first release with
new features, so I would need to find all those and correct them if I built
a new branch with the features rolled back to 2.4.2 state and that whole
process sucks when one text file is missing 3 entries and we're deep into
dev of next versiob because the ide doesn't let you put . I think
serialupdi should w
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2021-12-12:
worth noting that 2.5.0 (formerly 2.4.3) is coming out VERY soon.

SpenceKonde, 2021-12-21:
No comments from OP in over a week. Closing under assumption that this is resolved.

closed 2021-12-21 by SpenceKonde

dejavu1987, 2023-04-20:
Well I have tried using an smd switching diode and a the resistor of 1k and my CH340 adapter to program my first ATTiny424, and it worked without a hitch, I tried twice and worked like a charm.

Now today a month later I tried literally same Hardware, same sketch, and same config, and it is not working anymore.
Did anything change in the library, I can still see the same version in the terminal tho `Version 1.2.3 - Jan 2022`.

Really confused.
I am getting same `pymcuprog.pymcuprog_errors.PymcuprogError: UPDI initialisation failed`

I have recorded a video which is on youtube which shows that the configs look similar too.
````
Using serial port /dev/cu.usbserial-210 at 230400 baud.
Delaying next op after each page write command by 1.0ms (-wd).
Target: attiny424
Set fuses: ['0:0b00000000', '2:0x02', '6:0x04', '7:0x00', '8:0x00']
```
Here is the video https://www.youtube.com/shorts/w51bsm1D_dc 
Should I change resistance value? The fact that it worked before makes me think it might have changed somewhere else.

SpenceKonde, 2023-04-21:
drop the resistor back to 220 ohms or so and it should work on a CH340, I use those all the time without issue,

dejavu1987, 2023-04-22:
Okay realised, my CH340 had Tx and Rx in reverse. 
Woked both with 1k and 330 res.
In addition, I was using another board where UPDI pin was connected with a .1uF cap to the ground for some reason, that was also one reason it did not work.

wholder, 2023-08-12:
To barge in on this thread, In an attempt to minimize issues with diode voltage drop, I tried using a tri-state buffer (sn74lvc1g125) where the enable pin (active low) is tied to the input pin and then wired to the TX line of a CH340N.  Likewise, the output pin is connected to the RX pin and then to the UPDI pin on an ATtiny212.  However, this seems not to work at all.  That is, the ATtiny212 does not respond to any queries I try.  I was able to use the diode scheme earlier, but found that I kept getting garbled responses (parity errors, etc.) on the RX line.

The signal output from the sn74lvc1g125 looke fine on my scope, so I'm not sure what the problem might be.  It seems like having a tri-state output from the TX line would be ideal, but I must be missing something.  Any thoughts?

hmeijdam, 2023-08-13:
How you wired it current can't flow from the UPDI pin into the TX pin.

Which I believe is the case with the diode schematic, given how it's polarity is positioned.
```
internal resistor in adapter: 2.2k >= Ra

--------------------                                 To Target device
                DTR|                                  __________________
    internal    Rx |--------------,------------------| UPDI---\/\/---------->
  Tx---/\/\/\---Tx |-------|<|---'          .--------| Gnd    470 ohm (100 ~ 1k)
    resistor    Vcc|---------------------------------| Vcc
  typ 1-2k      CTS|                     .`          |__________________
                Gnd|--------------------'             If you make a 3-pin connector, use this pinout
--------------------

```

Leo-Ka, 2024-07-22:
I have Attiny1616. Here is my solution: https://github.com/SpenceKonde/megaTinyCore/discussions/713?new_signup=true?new_signup=true#discussioncomment-10117450