SpenceKonde/megaTinyCore #549: ArduinoModbus/ArduinoRS485 Serial Issues

DigiMaxIO, 2021-11-12:
First off, thank you for your great work @SpenceKonde!

I am working on a project for the ATTINY3217 which utilizes the [ArduinoModbus library](https://github.com/arduino-libraries/ArduinoModbus). This Modbus Library uses the [ArduinoRS485 library](https://github.com/arduino-libraries/ArduinoRS485) to function. The RS485 library is supposed to inherit the Hardware Serial library from the core but for some reason, it simply does not work-> no data comes out of the TX pin, etc. 

It does however work appropriately if I use the @xukangmin's TinyCore [https://github.com/xukangmin/tinycore](TinyCore)
but unfortunately, this core isn't set up for PlatformIO IDE yet which is what I am using.

Also Note: If I try to use these libraries in the official Arduino IDE with MegaTinyCore, It does not compile (this does not occur if using PlatformIO), and the error is not one I was able to figure out, see here:

;-------------------------------------------------------------------------------------------------------------------------------
/var/folders/9w/fl9jynkn51sbjgyvhn0sf7qc0000gn/T//ccBXpSWx.ltrans0.ltrans.o: In function ModbusRTUServerClass::poll():
<artificial>:(.text+0x1d7a): undefined reference to operator delete(void*, unsigned int)
<artificial>:(.text+0x1d90): undefined reference to operator delete(void*, unsigned int)
<artificial>:(.text+0x1da6): undefined reference to operator delete(void*, unsigned int)
collect2: error: ld returned 1 exit status
Using library ArduinoRS485 at version 1.0.2 in folder: /Users/username/Documents/Arduino/libraries/ArduinoRS485 
Using library ArduinoModbus at version 1.0.6 in folder: /Users/username/Documents/Arduino/libraries/ArduinoModbus 
exit status 1
Error compiling for board ATtiny3227/3217/1627/1617/1607/827/817/807/427/417._
;-------------------------------------------------------------------------------------------------------------------------------

Any tips/help getting this working is much appreciated!

Thanks!

SpenceKonde, 2021-11-13:
UGH!  I hate when it gives offsets that we can't see (.text+1d7a - that'd be real helpful if you had the object files and could read machine code, but IIRC 
I will say that 1)  No testing is conducted on third party IDEs, so I can't comment on why platformIO lets it build when arduino doesn't - Are they using the same version of megaTinyCore? They don't do a great job of keeping that updated in their distributions; more importantly though, platformIO doesn't make sure to include everything that the Arduino IDE does, and so I only understand the environment of the IDE - I have no idea whether or confidence in what settings it is using correctly and what ones it might be silently dropping
2. That library looks like a monstrosity. 
3.  One of the next big projects involves supporting a few missing options on the hardware serial ports. Including to control an RS485 transciever. 
4. I have a feeling that this is a matter of them assuming one thing about how the Arduino wrappers behave which matches some cores but not my core (I have gone beyond the Arduino API in many places because of the disappointing design decisions evident in the official core.
4. That tinyCore you referred to is also incompatible with all hardware that accurately identifies its platform by arduino standards, and any library that does that and which it thinks is compatible, is likely incompatible. There are a lot of problems with that core, not least the fact that the maintainer doesn't seem to be actively maintaining it. 

Can you please post code that reproduces the arduino error? I have no idea what modbus is, I need example code that I can generate the error on and examine.

DigiMaxIO, 2021-11-15:
@SpenceKonde I happened to come across this issue that had already been raised previously, [HERE](https://github.com/SpenceKonde/megaTinyCore/issues/323#issuecomment-779386491). Changing my clock setting to 20Mhz fixed the issues with the library in PlatformIO.

closed 2021-11-15 by DigiMaxIO

SpenceKonde, 2021-11-16:
What clock speed was it using that was generating that error? 

Or were you not defining ANY clock speed (which yeah, nothing will compile like that, because code all over the place assumes F_CPU is defined)  I thought I checked for that and threw a #error if it wasn't supplied. And I'm almost certain I do for CLOCK_SOURCE, which should be 0 for internal oscillator and 2 for external, plus MILLIS_USE_TIMERx where x is NONE, A0, B0, B1 (only on parts with TCB1), D0, RTC, or RTCXTAL - to set the millis timekeeping source or tell it not to provide millis). Plus, with 2.4.3, Wire requires TWI_MORS or TWI_MANDS to specify whether you want have both slave and master enabled at the same time... The Optiboot boards also define USING_BOOTLOADER which is used at a few key moments. 

And optionally CLOCK_TUNE_INTERNAL to turn the internal oscillator to a different speed than what it's fused for (see documentation - they can be pushed way farther in either direction than the datasheet says - most 2-series work at 32 and basically all 0/1/2 run fine at 24 at room temperature and 5v)). Next version will have a "referenceless" tuning sketch too that I think will be nearly as good... and will require a define to configure how attachInterrupt() works (so you can make it not take over every pin vector). 

Developing for the  Arduino IDE, when I put in a new option, I can add it to board.txt/platform.txt and know that the default value *will be the default value*. But with platform IO, all bets are off, and I have no way to tell people what values they should use, (and nobody reads my docs _anyway_)

DigiMaxIO, 2021-11-17:
I was originally running at 16Mhz, and setting it to 20Mhz is what fixed my baud timing issue. 

The only error that was generated was when I tried to use the Modbus library on the Arduino IDE... Had nothing to do with the clock issue from what I could tell.

The way things are configured in PlatformIO is so that each project has a configuration file (platformio.ini file). Within that file, you can configure all sorts of settings including the clock settings. I always make sure to set this, however, I believe PlatformIO has a default clock setting that it sets too if nothing is added manually (all other MCU's I've tried with PlatformIO work this way too).

Here is an example (Note the clock setting in Bold): 
>[env:ATtiny3217]
platform = atmelmegaavr
board = ATtiny3217
framework = arduino
board_build.mcu = attiny3217
**board_build.f_cpu = 20000000L**
debug_tool = pkobn_updi
upload_protocol = pkobn_updi
upload_port = /dev/cu.usbmodem224302
monitor_port = /dev/cu.usbmodem224302
debug_port = /dev/cu.usbmodem224302

I've read your Docs quite a few times now, they're great, but just because you read something in Docs doesn't mean you'll know what's applicable to your particular problem. Docs can be pretty damn close to perfect, but you still have human beings using them.