SpenceKonde/megaTinyCore #550: ATtiny1614/ tone() function / unwanted debug information to the Serial console

6v6gt-duino, 2021-11-14:
Core version megaTinyCore 2,4,2
Arduino IDE 1.8.16
Attiny1614 / No bootloader

This output:

`
2048
1219
0
2048
1219
0
2048
1219
0
`
appears when used with this sample code;

```
const uint8_t buzzer = 0 ;
const uint8_t led = 2 ;

void setup() {
  Serial.begin( 115200 ) ;
  pinMode( buzzer, OUTPUT ) ;
  pinMode( led, OUTPUT ) ;
}

void loop() {
  tone( buzzer , 2048 ) ;
  delay( 1500 ) ;
  noTone( buzzer ) ;
  digitalWrite( led, HIGH ) ;
  delay(500) ;
  digitalWrite( led, LOW ) ;
}
```

hmeijdam, 2021-11-14:
How do you get the output with this code? I see no Serial.print statement.

SpenceKonde, 2021-11-14:
Yikes. Thanks. Wonder how long that's been that way...  It's either since July or March of 2021.

hmeijdam, 2021-11-14:
And I just reproduced same error on a 404 and 416

commit 3b6af0a2 (2021-11-14): Fix #550
closed 2021-11-14 by SpenceKonde