SpenceKonde/megaTinyCore #553: serialUPDI not compatible with OTA updates

specternecter, 2021-11-22:
I have been working on OTA uploading to an ATtiny3224 via serialUPDI over Bluetooth and I found a major problem that appears to effect all Bluetooth devices. I tried a total of 7 different Bluetooth-UART modules from different manufacturers, and couldn't get serialUPDI to work with any of them. When I tried a traditional USB to UART module it worked. So I tried the various Bluetooth modules again with a COM Port monitor running and I found where the issue lies. However, I don't know python that well so I don't know how to fix it myself. I recently developed an ISP over Bluetooth programmer from scratch in C, so I understand these types of protocols and everything; I just don't speak python.

At first I thought the issue was related to how serialUPDI appears to open and close the port every time it sends data. No Bluetooth module likes rapid disconnect/reconnect but it turns out most will accept it in this case, albeit with a little effort sometimes. This repeated opening and closing of the port is likely part of the issue but the actual problem becomes pretty clear by watching the serial port monitor.

The beginning of the incoming data from the serial port should look like this:

 00                                                

 55 C3 08                                          

 55 C2 06                                          

 55 80 40                                          

 55 C2 06                                          

 55 C9 02                                          

 55 E6 74 69 6E 79 41 56 52 20 50 3A 30 44 3A 31   
 2D 33 4D 32 20 28 30 30 2E 35 39 42 30 45 2E 30   
 29 00 

and the outgoing data back to the serial port should look like this:

 00                                               

 55 C3 08                                          

 55 C2 06                                          

 55 80                                            

 55 C2 06                                          

 55 C9 02                                          

 55 E6  

The above exchange is what I would get from a USB-UART converter, which works without issue. However, ALL the various Bluetooth modules behave like this...

The entire incoming data from the Bluetooth serial port:

 00                                               

 55 C3 08                                         

 55 C2 06                                          

 55 80                                            

and the entire outgoing response data:

 00                                               

 55 C3 08                                         

 55 C2 06                                        

 55 80                                            

Obviously this is a failed upload and if we compare this data with what we should see in a working exchange, we notice that the data coming into Bluetooth should be "55 80 40" but instead it's getting just "55 80".

So to get a little more information on the issue I created a simple Android app to act as a middle-man. The serial port on the computer sends data to the app, and the app sends it to the Bluetooth module and vice-versa. The reason for doing this is so the app could intercept the incomplete "55 80" and instead send the proper "55 80 40" and see what happens.

The incoming data from the serial port was now:

 00                                               

 55 C3 08                                          

 55 C2 06                                          

 55 80 (here we add the 40 and send it over to the Bluetooth module)      
                                 
 55 C2 06                                          

 55 C9 02                                          

 55 E6

and the response data is now:

 00                                               

 55 C3 08                                          

 55 C2 06                                         

 55 80 40                                           

 55 C2 06                                         

 55 C9 02                                         

 55 E6  

Now, a couple things become apparent by looking at this exchange. The important thing is that we can see that Bluetooth is receiving the commands just fine, but it's not receiving the following data. It's getting separated and the data is actually getting completely dropped. As I stated earlier; I tested many different types of Bluetooth modules and they all behave this way. It appears that it's constantly connecting and disconnecting, and that the command is coming in but the following data is slightly behind and being sent out during the period of disconnection, and therefor getting lost.

By looking at how the middle-man app was implemented and the response it gave, I can also see that the delay between the command and data is probably necessary, but is there a way to hold a much more steady connection with the port? This erratic connect/disconnect appears to render serialUPDI over Bluetooth impossible. Which would be sad considering you can pick up any number of Bluetooth modules for less than $3 and could then wirelessly upload and debug right "out of the box". It's a much more convenient method, but leaves me with the options of writing my own programmer from scratch in C, get much more proficient in python, or hope that there’s an easy fix to serialUPDI that would support both wired and wireless uploads.

SpenceKonde, 2021-11-22:
> I just don't speak python.

Don't worry, me neither. 
Nor do the people who wrote pymcuprog (according to one of the few people involved in the process who does). 

Opening and closing the connection? It does that some at the beginning only. 

As you've probably noticed, since RX and TX are connected, the RX line sees everything sent over TX - we expect to read back all those bytes we sent, plus the response, if any.

55 C3 08 - write UPDI ctrl B, - disable collision detection. 

55 C2 06 - write UPDI ctrl a, lower the guard time to very near the minimum and turn off the interbyte delay. 

55 80 40 - read the UPDI status a register. The 80 is sent by the computer, the 0 is sent by the target. THIS IS THE FIRST TIME YOU ARE EXPECTING A RESPONSE - response signature isn't sent for STCS for whatever reason and that's intended. I wouldn't have designed it that way, but I don't work for Microchip. You get no response. 

55 C2 06 Here, again we're setting the guard time lower and making *sure* ibdly is off (it clobbers the read speed, and the code was turning it back on when I didn't expect it). 

55 C9 02 adjust the speed of the UPDI clock  to 8 MHz so we have plenty of margin at 230400 baud

55 E6 Here we are requesting the System Information Block. Once more, we get nothing back. 

The response you were getting before by the way with wired connection is 
tinyAVR P:0D:1-3M2 (00.59B0E.0)
nvm Programming v0, Debug interface v1, 3M2 UPDI osc die rev 0, - and I'm nor privy to meaning the rest f the cryptic numbers)

Looks to me like there's no communication whatsoever, you need to look at the hardware level.... 

I'm not sure whether bluetooth serial adapters have all the functionality needed for this...  How are you getting that information from in the middle? Anything that's on the bluetooth side doesn't tell you all that much...  you don't know if the information is getting mangled on it's way out of the bluetooth serial adapter if so which bytes (ones near start and end are indicative of port opening/closing, possibly - but I don't think that's it, because even the read SIB failed, and that was sent 6 bytes after the last reconnect. , or if the device is trying to respond but the bluetooth adapter isn't recognizing it..... 

The right way to spy on the serial connection is to use an oscilloscope to see what's coming through - the logic levels can be an issue because this sort of setup has to thread the needle - both sides must be able to drive the line low enough for the other side to see as low, and the line must go high enough for both sides to recognize as a high, and I have seen every permutation if that fail. 

Failing that, tying the UPDI line to the RX line of a normal wired serial adapter, (and ground of the bluetooth adapter+target to the wired adapter ground, of course),might at least tell you something - though that would only be illuminating if you saw something different, and if you didn't see anything different you'd need to put 'scope in it to figure out why. 

I for one am not optimistic that the bluetooth adapters are similar enough to real ones - the whole tying TX to RX through a resistor or schottky diode is a bit outside of the normal use cases, and all the bluetooth serial adapters I've used didn't support baud rate configuration from the incoming connection, and had a whole extra layer of restrictions. I didn't think the ones I used could change baud rates like SerialUPDI has to... Also, like, SerialUPDI has been optimized extremely aggressively for speed, because that's what people were complaining about. It does require favorable conditions for operation. it's definitely not meant to OTA, and I suspect to make it work over bluetooth, you're going to need some intelligence at the other end - it's not going to work with just 

You might be better off building from jtag2updi, which though it is a flaming bag of dog poop, it is expected to function on a normal serial port not one hacked to work in a half-duplex mode it was never designed for.. (though god that codebase is awful, it's someone's "look at how good at C++ I am" project, using all the C++y features to the max, I really struggled with it. .)

SpenceKonde, 2021-11-22:
Unless evidence that this is an issue with serialupdi itself comes to light, I think this is most likely an unavoidable fact of life and a limitation of bluetooth serial adapters w/rt this sort of weird single-wire mode, and as such does not represent a defect in the core, but I'll leave this open for now.

specternecter, 2021-11-23:
Thank you so much, that was incredibly informative. My next step was to look into the actual UPDI protocol to figure out what data was coming in and out, but I was trying to avoid looking too deep into it unless I had to write a programmer. Which I very clearly need to, but at least you gave me a quick crash course on the starting protocol. The particular Bluetooth module I favor (and mostly used for these attempts) cannot change baud rates like you mentioned. It can be changed through a special protocol anywhere between 1200 and 1000000 baud, but not in the traditional way that a computer would attempt. It is now clear to me after further investigation that the issue is hardware related and that serialupdi over Bluetooth would only work with another AVR in between the Bluetooth module and the target, such as in the ISP over Bluetooth programmer I developed.  Thanks again

SpenceKonde, 2021-11-23:
Yeah - On the plus side, you've already got an AVR and a bluetooth serial
module for the BT ISP, so making that speak UPDI is just a matter of
software; (The challenge of course is to figure out a way to make avrdude
instruct your thing which to do. Someone who's comfortable in C++ might
find jtag2updi less frustrating than I did, and might even be able to
figure out how the hell it can possibly ever get into a bad state with the
timeout I implemented. The code for jtag2updi is a non-Arduino-dependent
program (though it can be built with the IDE) that compiles to like 5k so
it could be a viable starting point. Integrating it with existing code
would be a job (on the plus side, it doesn't use any interrupts - I
considered building it for a tinyAVR, along with an ISP programmer, but
building jtag2updi with the section start linker option placing it after
the end of the ISP programming sketch (which had reset vector pointed to
it, and then checking (and clearing) reset cause flags, since you can tell
whether a reset was a WDT reset or a software reset that way, and you can
easily trigger both from software on modern AVRs.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

specternecter, 2021-11-23:
Yeah don't get me started on the way jtag2updi is written because it's downright disrespectful lol. I think if I were to gather information on how to make the IDE talk to Bluetooth, I'd actually be better off looking at serialupdi. I do somewhat know python because I've written some python code myself; I'm just not all that proficient in it and trying to comb through serialUPDI is not going to be easy, but likely a lot easier than trying to interpret jtag2updi. I'm not worried about the UPDI protocol itself because I've done work on a lot of different protocols. I even have an ATmega1284 that programs an ESP32 by selecting the desired firmware from the onboard micro-sd card. I thought a lot about this last night and I think I have a better solution that would work for both my projects as well as being very useful to others. This solution would bypass needing to work with the actual protocol (saving time) as well as getting the IDE to talk to it.

My ISP over Bluetooth board uses an ATmega328pb, which I selected because it has a second UART that shares pins with SPI. So it has the standard 6-pin ISP header but MOSI and MISO can also be used for TX/RX. It has a micro slide switch that allows you to select ISP mode or UART mode. UART mode functions just like any other Bluetooth to UART module but with some other really cool features added in.

What I'm envisioning is actually using serialupdi as it is to handle the communication between the IDE and Bluetooth (or simple UART). Let's forget that my board can run in ISP mode for a minute and also pretend it uses SoftwareSerial for communication with the target rather than HardwareSerial. SoftwareSerial maxes out at 38400 baud, but this isn't a problem whatsoever here. We can even pretend it doesn't have Bluetooth and that it's just HardwareSerial there. So we have an AVR with a HardwareSerial port and a SoftwareSerial port. I'm going to call this imaginary board SULLY.

The thing I really never liked about serialupdi is the way it hacks half-duplex UART. If you want to switch between debug and UPDI you either have to manually move the lines, or as I did, have a DPDT switch do it. SULLY's HardwareSerial port would be what talks to the computer, either by USB-UART or Bluetooth-UART. Ideally I would like to have SULLY detect the starting sequence of UPDI and automatically switch into UPDI mode when the IDE starts serialupdi, but a button or switch to pull a pin low would work too if this proves too challenging to get right. Sully would require the standard 3 pins (VCC, UPDI & GND) but also have an RX and TX for connection with it's SoftwareSerial port. A UPDI enable pin may be necessary if the automatic detection doesn't work right, but I think I can get it.

We're in UPDI mode now, however we got here. serialupdi is going to send SULLY some data from the IDE. Sully is going to translate that data to the UPDI pin (and obviously the target). It gets the response from the target and sends it back to the IDE. serialUPDI is happy. We're all happy. Getting that pin to speak UPDI is going to be the most challenging part, but nothing I can't handle. No need to learn the UPDI protocol, just how to speak it. So now we have what is essentially an AVR acting as a translator between UART and UPDI. The really cool thing about SULLY is that SoftwareSerial port. If the target's UART port (hardware or software) is connected to it, there's no need to switch lines. When UPDI is finished uploading you can just open up the serial monitor and debug without the hassle. SULLY's UART ports are separate. You're limited to 38400 baud or less because it's SoftwareSerial, but you really don't need anything faster for debugging. SULLY's HardwareSerial port is handling the UPDI upload and the speed would just be limited by your selected AVR and design parameters. You would just have to use an AVR that has at least 1 Hardware UART port.

specternecter, 2021-11-23:
I could even look at the UPDI reset sequence and add a reset pin that'll trigger a reset in the target. Now you have the capabilities of a reset button. I HATE how the ATtiny3224 doesn't allow this without disabling UPDI

specternecter, 2021-11-23:
Getting the UPDI pin to "speak UPDI" appears much easier than I thought. The AVR304 pdf seems to have everything I need.

SpenceKonde, 2021-11-28:
Yeah that was a major bummer. The way people typically get a "reset pin" is with "ersatz reset", that is, an interrupt (optionally a prioritized one) triggered from low level on pin, and which just fires software reset. Not foolproof, but it's sometimes viable.

SpenceKonde, 2021-11-28:
Closing as this issue does not represent a defect in the core.

closed 2021-11-28 by SpenceKonde