SpenceKonde/megaTinyCore #557: Tool avrdude is not available for your operating system

denis-stepanov, 2021-12-05:
I get the message `Tool avrdude is not available for your operating system` when I try to install megaTinyCore in Arduino IDE.

Arduino 1.8.16. on Ubuntu 18.04 32bit CPU. I guess 32bit is a problem.

SpenceKonde, 2021-12-05:
This is an Arduino problem. The latest version of avrdude is not supplied by Arduino for 32-bit linux, apparently. And they don;t use arduino18 in any official cores apparently, so they don't care that it doesn;t work there, 

It looks like Arduino dropped support 32-bit linux; from their packages index 
```
        {
          "name": "avrdude",
          "version": "6.3.0-arduino17",
          "systems": [
            {
              "size": "219631",
              "checksum": "SHA-256:2a8e68c5d803aa6f902ef219f177ec3a4c28275d85cbe272962ad2cd374f50d1",
              "host": "arm-linux-gnueabihf",
              "archiveFileName": "avrdude-6.3.0-arduino17-armhf-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-armhf-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "229852",
              "checksum": "SHA-256:6cf948f751acfe7b96684537f2291c766ec8b54b4f7dc95539864821456fa9fc",
              "host": "aarch64-linux-gnu",
              "archiveFileName": "avrdude-6.3.0-arduino17-aarch64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-aarch64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "279045",
              "checksum": "SHA-256:120cc9edaae699e7e9ac50b1b8eb0e7d51fdfa555bac54233c2511e6ee5418c9",
              "host": "x86_64-apple-darwin12",
              "archiveFileName": "avrdude-6.3.0-arduino17-x86_64-apple-darwin12.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-x86_64-apple-darwin12.tar.bz2"
            },
            {
              "size": "254271",
              "checksum": "SHA-256:accdfb920af2aabf4f7461d2ac73c0751760f525216dc4e7657427a78c60d13d",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avrdude-6.3.0-arduino17-x86_64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-x86_64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "244550",
              "checksum": "SHA-256:5c8cc6c17db9300e1451fe41cd7178b0442b4490ee6fdbc0aed9811aef96c05f",
              "host": "i686-linux-gnu",
              "archiveFileName": "avrdude-6.3.0-arduino17-i686-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-i686-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "328460",
              "checksum": "SHA-256:e99188873c7c5ad8f8f906f068c33600e758b2e36cce3adbd518a21bd266749d",
              "host": "i686-mingw32",
              "archiveFileName": "avrdude-6.3.0-arduino17-i686-w64-mingw32.zip",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-i686-w64-mingw32.zip"
            }
          ]
        },
```
Yet for Arduino18. it's not. 
```
        {
          "name": "avrdude",
          "version": "6.3.0-arduino18",
          "systems": [
            {
              "size": "220677",
              "checksum": "SHA-256:2e25c9e99c255d595a1072679a88ecddfa12c223b18510760bb867039f35efa5",
              "host": "arm-linux-gnueabihf",
              "archiveFileName": "avrdude-6.3.0-arduino18-armhf-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-armhf-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "231047",
              "checksum": "SHA-256:4f88bb50d2235182ed7aa9e0a1d08e4bb956378ac9569b8e1141e37ed314fb2d",
              "host": "aarch64-linux-gnu",
              "archiveFileName": "avrdude-6.3.0-arduino18-aarch64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-aarch64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "280072",
              "checksum": "SHA-256:df1dfd18e2e287c47232605cd4fa41751eb70df8c300aeb7a00a3a09b524a1b8",
              "host": "x86_64-apple-darwin12",
              "archiveFileName": "avrdude-6.3.0-arduino18-x86_64-apple-darwin12.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-x86_64-apple-darwin12.tar.bz2"
            },
            {
              "size": "254460",
              "checksum": "SHA-256:1ae46972b276b8a54c459f87c4ff326abdad0be2b1a293d73bf86e47765eddc3",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avrdude-6.3.0-arduino18-x86_64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-x86_64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "329515",
              "checksum": "SHA-256:0781f4183e91a9783c2330035520144ab76b8f75c0a9f7a25877c063bc984c4d",
              "host": "i686-mingw32",
              "archiveFileName": "avrdude-6.3.0-arduino18-i686-w64-mingw32.zip",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino18-i686-w64-mingw32.zip"
            }
          ]
        },
```

I can probably define a new version in name that points to the same files, and falls back to 17 for linux x86....

SpenceKonde, 2021-12-05:
It should now work on latest versions, and hopefully this didn't break anything for everyone else. 

Delete the local copy of platform. platform_drazzy.com_index.json from wherever it's buried on Linux, and try to install latest version and it should work.

SpenceKonde, 2021-12-05:
Let me know how this goes for you

SpenceKonde, 2021-12-11:
Great, thanks for confirmation!

closed 2021-12-11 by SpenceKonde