SpenceKonde/megaTinyCore #558: UPDI/Reset pin option missing from Tools menu for ATtiny404 etc

technoblogy, 2021-12-06:
The documentation:

https://github.com/SpenceKonde/megaTinyCore#list-of-tools-sub-menus

suggests that there should be a Tools -> UPDI/Reset pin option, but I'm not getting it with the **ATtiny3224/1624/1614/1604/824/814/804/424/414/404/241/204** option.

SpenceKonde, 2021-12-06:
That is correct. There are no options other than the default (UPDI) that do not require an HV programmer to reprogram. They're only shown on the optiboot boards because there if you bootload it, you can use optiboot to upload code.  I don't expose that option when there is no way you could use it without an exotic HV programming tool (very few people have these) because people will brick their boards and complain. There are lines in boards.txt you can uncomment if you have one.

SpenceKonde, 2021-12-06:
On parts with more pins, the menu is shown, but only contains UPDI and the alt reset pin option for the non-optiboot configurations. Sadly the 14-pin parts didn't get  the alt reset pin (it's PB4, which they unfortunately don't have........

closed 2021-12-06 by SpenceKonde

technoblogy, 2021-12-07:
OK, thanks!