SpenceKonde/megaTinyCore #561: ModuleNotFoundError: No module named 'hid'

denis-stepanov, 2021-12-11:
Sorry to bother you again.
I am trying to program the 1624 controller with the following settings:

Board: ATtiny3224/1624/824/424 w/Optiboot
Chip: ATtiny1624
Programmer: SerialUPDI with 4.7k resistor or diode (230400 baud)

I have a python error:
```
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.0 - June 2021
Using serial port /dev/ttyUSB0 at 230400 baud.
Target: attiny1624
Set fuses: ['2:0x02', '6:0x04', '8:0x02']
Action: write
File: /tmp/arduino_build_32448/atty_hello.ino.hex
Connecting to any uart{upload.with_programmer_not_available_for_bootloader_boards}
Traceback (most recent call last):
  File "/home/dns/.arduino15/packages/megaTinyCore/hardware/megaavr/2.4.2/tools/libs/pyedbglib/hidtransport/cyhidapi.py", line 4, in <module>
    import hid
ModuleNotFoundError: No module named 'hid'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dns/.arduino15/packages/megaTinyCore/hardware/megaavr/2.4.2/tools/prog.py", line 282, in <module>
    main()
  File "/home/dns/.arduino15/packages/megaTinyCore/hardware/megaavr/2.4.2/tools/prog.py", line 131, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "/home/dns/.arduino15/packages/megaTinyCore/hardware/megaavr/2.4.2/tools/prog.py", line 176, in pymcuprog_basic
    backend.connect_to_tool(toolconnection)
  File "/home/dns/.arduino15/packages/megaTinyCore/hardware/megaavr/2.4.2/tools/libs/pymcuprog/backend.py", line 150, in connect_to_tool
    self.transport = hid_transport()
  File "/home/dns/.arduino15/packages/megaTinyCore/hardware/megaavr/2.4.2/tools/libs/pyedbglib/hidtransport/hidtransportfactory.py", line 46, in hid_transport
    from .cyhidapi import CyHidApiTransport
  File "/home/dns/.arduino15/packages/megaTinyCore/hardware/megaavr/2.4.2/tools/libs/pyedbglib/hidtransport/cyhidapi.py", line 9, in <module>
    raise ImportError(MSG)
ImportError: This transport class requires cython and hidapi to be installed:
> pip install cython
> pip install hidapi

the selected serial port 
 does not exist or your board is not connected
```

Programming does work when I select the same board without Optiboot.
Arduino 1.8.16, core 2.4.2.

SpenceKonde, 2021-12-11:
Ugh, I was expecting a more graceful failure than that. What that was designed to do is generate the a message like

"Unknown option for port {upload.with_programmer_not_available_for_bootloader_boards}" 

I can't make the Arduino IDE grey out the Upload using Programmer option, and and I don't know how any devices ever get the with-bootloader binaries created, but they are never generated for these parts. 

However, when the sketch is compiled for use with Optiboot, it will not function when optiboot is not present. Likewise, if the sketch uis compiled for use without optiboot, it will not work on a board that does have optiboot on it. 

This is a consequence of Microchip's design decisions: Classic AVRs put the bootloader at the very end of the flash, and there was a fuse you could set that would make it jump to there, instead of 0x0000 on reset. Among other things, that allowed the binary to be identical whether or not there was a bootloader - the reset vector was at 0x0000.

NOW the bootloader is at THE BEGINNING of the flash. That means:
* the compiler and linker need to be told to start 512b from the first address, (it's a 512b  bootloader). Everything is offset 512b. 
* If a sketch compiled for a bootloader is uploaded to a device without one, without setting the fuses, execution will start at 0x0000. There's nothing there, so it will skid along the 00xFFFF instructions (which are a skipif, but the condition won't be true at startup) until it hits 0x0200 (512b in. That's the reset vector, so (assuming it was compiled with -mrelax, which we do) it will begin running the sketch, initializing the peripherals normall, including millis. When the millis timer rolls over for the first time, however, the reset vector it jumps to is vector 25, the part has > 8k flash so they're 4 byte vectors so execution jumps to address 100 (0x0064 in typical notation). That's in the blank bootloader section, so execution skids along the 0xFFFF's again. This time it's even money whether it's skipping every other one or not, but it doesn't matter, because there are 412 of these instructions between the place it jumped to, and the reset vector, so it will land on the reset vector. The reset vector will then begin initializing the sketch. (note: at this point, the code doesn't knowthis, but the hardware thinks it's running an interrupt!). 2.4.2 and earlier, it will proceed on it's merry way, only this time no interrupts will run, and the sketch will "run" with no interrupts but otherwise normally....But it'as basically not usable. In 2.5.0 and later, the dirty reset will be detected and trigger a software reset, leading to a bootloop (just as useless). Both will be confusing situations to debug. The only way you could end up in that situation is if you had an optiboot board definition selected, and then chose "Upload using Programmer" 
* If the board did have a bootloader, since upload using programmer erases the flash, that would remove the bootloader, and as I mentioned above, the compile process never generates combined bootloader + sketch binaries. That is what one would need to upload if you wanted "Upload using Programmer" to work with Optiboot. 

So because of this - since I don't know how to get combined binaries, but know non0combined ones will never work) I needed to figure out a way to disable the upload using programmer menu option for optiboot boards. Except they don't give us any toolsto really make our lives easier. So what I wound up doing is defining one of the arguments passed to the upload tools as {upload.upload.with_programmer_not_available_for_bootloader_boards}. On on-optiboot boards, I define upload.with_programmer_not_available_for_bootloader_boards as an empty string, so it vanishes and has no effect on behavior.  

My expectation was that it would give an error indicating that an invalid option was passed of that name, which would make clear to the user why it wasn't working: **Upload using Programmer is not supported for optiboot board definitions because it is guaranteed top produce a non-working sketch**

It looks like the reaction of SerialUPDI to that is much worse than it ought to be.

SpenceKonde, 2021-12-11:
The opposite case, where the target may have optiboot already, but we're uploading a non-optiboot-board compiled sketch is handled in a very simple manner: We just erase everything and upload like normal, set the limited number of fuses we set on any UPDI upload: (OSCCFG to set 16 vs 20 base speed, and BOOTEND to set the size of the bootloader section to 0 so that vectors and reset vector are in the correct location and sketch could potentially work.,

denis-stepanov, 2021-12-11:
Thanks for the detailed explanation. I was distracted by the "ModuleNotFoundError" to pay attention to the preceding messages. I was trying to go back from UPDI to Optiboot configuration, assuming the bootloader would be installed with the program. After burning the bootloader as a separate step Optiboot works again.

commit b31ebbee (2021-12-12): Fix #561 , update docs,

Also undo critical bugs intoduced by a linting error
closed 2021-12-12 by SpenceKonde