SpenceKonde/megaTinyCore #570: Improve SoftwareSerial speed and size

MCUdude, 2021-12-20:
Very simple fix. See https://github.com/arduino/ArduinoCore-avr/pull/433 for details.

SpenceKonde, 2021-12-20:
Goddamn.....

Pity this missed the 2.5.0 and 1.4.0 releases...

SpenceKonde, 2021-12-21:
And you can save another 4 words by casting to a uint8_t instead! there's a similar place in read where you can save 3 words too - but that's small potatoes compared to this change. 

What triggers the problem is that the result ends up as an int16_t, so modulo is much harder because the number could be negative. 

(unsigned) % (2<sup>constant</sup>) =  (unsigned) % (2<sup>constant</sup> -1) 

(signed) % (2<sup>constant</sup>) !=  (signed) % (2<sup>constant</sup> -1)

commit 043b2bcc (2021-12-21): Fix #570

Slightly better than the fix going into the standard cores
closed 2021-12-21 by SpenceKonde