SpenceKonde/megaTinyCore #575: relocation truncated to fit: R_AVR_13_PCREL against `no symbol'

ArminJo, 2021-12-21:
With 2.4.2 [this example](https://github.com/Arduino-IRremote/Arduino-IRremote/tree/master/examples/SendDemo) compiles well, with 2.5.2 I get the error you can see [here](https://github.com/Arduino-IRremote/Arduino-IRremote/runs/4599925366?check_suite_focus=true)
```
Linking everything together...
/home/runner/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-gcc -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x0 -mrelax -mmcu=attiny1604 -o /tmp/arduino-sketch-94DF6C3F89A2BA65C1FDC23FEF78B691/SendDemo.ino.elf /tmp/arduino-sketch-94DF6C3F89A2BA65C1FDC23FEF78B691/sketch/SendDemo.ino.cpp.o /tmp/arduino-sketch-94DF6C3F89A2BA65C1FDC23FEF78B691/../arduino-core-cache/core_megaTinyCore_megaavr_atxy4_chip_1604,clock_16internal_d162889679f52b643353cd4ee3f49e58.a -L/tmp/arduino-sketch-94DF6C3F89A2BA65C1FDC23FEF78B691 -lm
/tmp/cc1zFuRf.ltrans0.ltrans.o: In function `UartClass::_poll_tx_data_empty()':
<artificial>:(.text+0x13c): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/tmp/cc1zFuRf.ltrans0.ltrans.o: In function `UartClass::begin(unsigned long, unsigned int)':
<artificial>:(.text+0x132c): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
```

SpenceKonde, 2021-12-21:
Yes I was just alerted to this I will be releasing 2.5.3 with a fix shortly.

SpenceKonde, 2021-12-22:
This is now available in board manager release.

hmeijdam, 2021-12-22:
> This is now available in board manager release.
I think the new JSON for 2.5.3 may not have been pulled to the webserver.

ArminJo, 2021-12-22:
I also cannot find this version in Board Manager.

SpenceKonde, 2021-12-22:
Your diagnosis is correct, I had forgotten to push it.

ArminJo, 2021-12-22:
I am very sorry, but the error still exists 😞 see [here](https://github.com/Arduino-IRremote/Arduino-IRremote/runs/4609412507?check_suite_focus=true)

mechatroniks-git, 2021-12-22:
Confirmed, I'm still getting the PCREL with 2.5.3.

SpenceKonde, 2021-12-22:
Can either of you test with github version? (I can't reproduce the error here)

mechatroniks-git, 2021-12-22:
To use the GITHUB version, do I follow one of these instructions?

- https://github.com/SpenceKonde/megaTinyCore/blob/master/Installation.md

- https://www.electronics-lab.com/project/getting-started-with-the-new-attiny-chips-programming-the-microchips-0-series-and-1-series-attiny-with-the-arduino-ide/

mechatroniks-git, 2021-12-22:
I created a portable installation per your instructions, getting this:

```
panic: no major version found

goroutine 1 [running]:
go.bug.st/relaxed-semver.MustParse(...)
	/home/jenkins/go/pkg/mod/go.bug.st/relaxed-semver@v0.0.0-20190922224835-391e10178d18/parser.go:19
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).loadPlatforms(0x12d39a60, 0x12c2b4c0, 0x12490460, 0xc, 0x12d3a7bc)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210514123546-d710b642ef79/arduino/cores/packagemanager/loader.go:175 +0x14f6
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardwareFromDirectory(0x12d39a60, 0x12d3e428, 0x0, 0x0)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210514123546-d710b642ef79/arduino/cores/packagemanager/loader.go:120 +0x42b
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardwareFromDirectories(0x12d39a60, 0x12405240, 0x3, 0x4, 0x4, 0xcb63b2)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210514123546-d710b642ef79/arduino/cores/packagemanager/loader.go:45 +0x4f
github.com/arduino/arduino-cli/legacy/builder.(*HardwareLoader).Run(0x13db448, 0x12d46000, 0x13db448, 0x0)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210514123546-d710b642ef79/legacy/builder/hardware_loader.go:29 +0x150
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x13db448, 0x12d46000, 0x0, 0x0)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210514123546-d710b642ef79/legacy/builder/container_setup.go:50 +0x2b1
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x12d46000, 0x12449ee8, 0x2, 0x2, 0x0, 0x0)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210514123546-d710b642ef79/legacy/builder/builder.go:191 +0x198
github.com/arduino/arduino-cli/legacy/builder.(*ParseHardwareAndDumpBuildProperties).Run(0x12469f26, 0x12d46000, 0x13db380, 0x12d3e4b0)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210514123546-d710b642ef79/legacy/builder/builder.go:182 +0x8f
github.com/arduino/arduino-cli/legacy/builder.RunParseHardwareAndDumpBuildProperties(...)
	/home/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210514123546-d710b642ef79/legacy/builder/builder.go:214
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/main.go:391 +0x7de
D:\arduino-1.8.19\arduino-builder returned 2

Error compiling for board ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204.
```

SpenceKonde, 2021-12-22:
Can you post the code that is generating this error? 

I don't know how to deal with and have never seen that error;

mechatroniks-git, 2021-12-22:
I think my portable setup is not quite right, but this will fail on my non-portable Arduino, running 2.5.3 from boards manager, not github's latest.

```
#include <Arduino.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2( U8G2_R0, /* reset=*/U8X8_PIN_NONE );

void setup() {
  u8g2.begin();
  u8g2.initDisplay();      
  u8g2.clearDisplay();   
  u8g2.setFont( u8g2_font_crox1hb_tr );
  u8g2.drawStr( 0, 12, "hello there" );
  u8g2.setFont( u8g2_font_courB14_tr );
  u8g2.drawStr( 0, 32, "megatinycore" );
  u8g2.sendBuffer();
  Serial.begin(115200);
  
}

void loop() {
  Serial.print("Loop\n");
  delay(1000);
}
```

```
C:\Users\pc\AppData\Local\Temp\cc3aan2h.ltrans0.ltrans.o: In function `UartClass::_poll_tx_data_empty()':
<artificial>:(.text+0x30e): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
C:\Users\pc\AppData\Local\Temp\cc3aan2h.ltrans0.ltrans.o: In function `u8g2_ClearDisplay':
<artificial>:(.text+0x1900): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204.
```

[image]

SpenceKonde, 2021-12-23:
Okay, I've found the problem and will do another release tonight

SpenceKonde, 2021-12-23:
2.5.4 should now be in boardmanager. It was a stupid error - I'd written 0x8192 instead of 8192

closed 2021-12-23 by SpenceKonde

mechatroniks-git, 2021-12-23:
@SpenceKonde I think the new JSON for 2.5.4 may not have been pulled to the webserver.

SpenceKonde, 2021-12-23:
It is now, I managed to not update it in a slightly different way this time.

commit a341b16c (2021-12-21): Should fix #575 #574
commit 2ca4aca4 (2021-12-22): Fix #575