SpenceKonde/megaTinyCore #579: v2.5.4 install did not include avrdude.conf

EclipseBoom, 2021-12-23:
Just did a clean install of Arduino 1.8.19 and megaTinyCore v2.5.4 on Windows

Install did not include avrdude.conf in the usual location.  I grabbed it from the release ZIP file manually installed it into

C:\Users\me\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/avrdude.conf

After that it works for both "Burn Bootloade"r and the "Upload" button on the IDE.

SpenceKonde, 2021-12-23:
Weird, nothing relating to that was changed

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

EclipseBoom, 2021-12-23:
Maybe it was a glitch on my end.  Don't know.  

I was guessing it might have been missed but not noticed doing a 2.5.3 to 2.5.4 upgrade.

Anyway if there is no problem then fine by me to close the issue.

SpenceKonde, 2022-01-04:
probably related to my big screwup on the json

closed 2022-01-04 by SpenceKonde