SpenceKonde/megaTinyCore #584: Programming issues (split from existing closed discussion.

SpenceKonde, 2021-12-27:
Hi, I am trying latest arduino with latest megaTinyCore with ATtiny1626 and ATtiny3226, both are UPDI-programmable well with schottky and 470ohm (cytron CH340 adapter), even after initial several minutes of reverse polarity and probably smoke on both )), but when I try to select Optiboot version of board and burn the bootloader with (or without) support of DTR autoreset circuit (all placed on breadboard below), the default blink sketch (tested with UPDI) is not flashed it seems. I also initially had problem with COM1 port forced by avrdude.conf, uploading sketch doesnt take selected boarrd COM port, so I patched the global default in conf to my COM10 and then something started to happen - after few tests with 8sec optiboot without DTR (and no success), I tried also the autoreset, now I dont need to power toggle the board but detailed output of uploading lists this only:

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/avrdude.conf"

         Using Port                    : com10
         Using Programmer              : arduino
         AVR Part                      : ATtiny1626
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00
[image]

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 9.1
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9429 (probably t1626)

avrdude done.  Thank you.

_Originally posted by @apws in https://github.com/SpenceKonde/megaTinyCore/discussions/438#discussioncomment-1873564_

SpenceKonde, 2021-12-27:
Can you post the **complete** output of upload attempt that fails? Unfortunately, you've omitted the single most important line, the first line that shows what avrdude command it's actually invoking. I think the IDE must not include this line with "copy error messages" or something, because it is so often left out. 

No patching of avrdude.conf should ever be needed. Please provide details of what you have changed there, and what problem it was meant to resolve/

apws, 2021-12-27:
first issue was forced COM1 for avrdude, so I set my in conf at line 339 here: default_serial     = "com1"; 
now I reverted this and flashed optiboot board version boootloader which appears okay?

SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.0 - June 2021
Using serial port COM10 at 230400 baud.
Target: attiny1626
Set fuses: ['0:0x00', '1:0x00', '2:0x02', '5:0b11111110', '6:0x04', '7:0x00', '8:0x02']
Action: write
File: C:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/bootloaders/hex/optiboot_tx2z_rst.hex
Pinging device...
Ping response: 1E9429
Setting fuse 0x0=0x0
Writing literal values...
Verifying literal values...
Action took 0.03s
Setting fuse 0x1=0x0
Writing literal values...
Verifying literal values...
Action took 0.03s
Setting fuse 0x2=0x2
Writing literal values...
Verifying literal values...
Action took 0.03s
Setting fuse 0x5=0xfe
Writing literal values...
Verifying literal values...
Action took 0.03s
Setting fuse 0x6=0x4
Writing literal values...
Verifying literal values...
Action took 0.03s
Setting fuse 0x7=0x0
Writing literal values...
Verifying literal values...
Action took 0.03s
Setting fuse 0x8=0x2
Writing literal values...
Verifying literal values...
Action took 0.03s
Finished writing fuses.
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.01s
Writing from hex file...
Writing flash...
[==================================================] 8/8
Writing flash...
Action took 0.13s
Verifying...
Verify successful. Data in flash matches data in specified hex-file
Action took 0.04s

apws, 2021-12-27:
and now, after changing my circuit to use PB2=TX,PB3=RX, PB4=DTR autoreset RC and uploading sketch this is returned:

C:\Users\petra\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/avrdude.conf -v -pattiny1626 -carduino -D {upload.serialparams} -Uflash:w:C:\Users\petra\AppData\Local\Temp\arduino_build_496543/megaTinyCore_Blink.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/avrdude.conf"

         Using Port                    : com1
         Using Programmer              : arduino
avrdude: ser_open(): can't open device "\\.\com1": The system cannot find the file specified.

avrdude done.  Thank you.

Problem uploading to board.  See https://support.arduino.cc/hc/en-us/sections/360003198300 for suggestions.

apws, 2021-12-27:
when I patch avrdude.conf default serial port to com10 (my current ch340, as I didnt found in IDE way to change it - Tools selection is always on selected Com10), this is returned with the first command line:

C:\Users\petra\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/avrdude.conf -v -pattiny1626 -carduino -D {upload.serialparams} -Uflash:w:C:\Users\petra\AppData\Local\Temp\arduino_build_496543/megaTinyCore_Blink.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/avrdude.conf"

         Using Port                    : com10
         Using Programmer              : arduino
         AVR Part                      : ATtiny1626
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 9.1
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9429 (probably t1626)

avrdude done.  Thank you.

(excuse me for such unformated post, but here backticks for inserting code dont work well too, its weird)

SpenceKonde, 2021-12-27:
Problem is clear, that part in {} isn't defined and results in the unparaed
markup getting us3e instead of what it resolves to markup coming through
and not generating a simple compile error,

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

apws, 2021-12-27:
So okay, its bug? and will it fix the not uploading hex? Another thing I found was avrdude WARNING about duplicated t3217 ID in avrdude.conf, this id is used in both ATtiny3217 and 3226 definitions - this is something intentional or not? i also tried to fix it in conf, not knowing consequences though, but its weird. Thank you for your support :-)

SpenceKonde, 2021-12-27:
No it is absolutely not intended, an7 time you see one boards name in
another's section it is copy-paste error. I don't think I could design a
format more vulnerable to human error if I tried

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

apws, 2021-12-28:
ya, as a workaround, replacing {upload.serialparams} by -PCOM10 -b230400 in cmdline flashes well against optiboot ...

SpenceKonde, 2022-01-11:
This issue should be corrected in 2.5.6

closed 2022-01-11 by SpenceKonde