SpenceKonde/megaTinyCore #587: CP2102 Programming Problem

GeorgeIoak, 2021-12-28:
I started a [discussion](https://github.com/SpenceKonde/megaTinyCore/discussions/585#discussioncomment-1876828) about this issue but thought it warranted creating an issue.

I have a CP2102 serial adapter and to do some initial quick testing I soldered a ATtiny1626 to a breakout board and added a BAT54C and 470Ω series resistor on a breadboard. During my initial troubles I separately installed pymcuprog using pip and ran some testing from the command line. Using the original pymcuprog I can reliably flash my part:

```
C:\Users\george\Box\Design\Fresenius Jet\Design\Software\Test Code\readTempVcc>pymcuprog -t uart -u COM5 -b 230400 -d attiny1626 -f readTempVcc.ino.t1626.20c0.mB1.pfF.v254.hex write --verify
Pinging device...
Ping response: 1E9429
Writing from hex file...
Writing flash...
Verifying flash...
OK
Done.
```
If I try to flash through Arduino 1.8.16 with the default SerialUPDI programmer at 230400 baud it fails unless I drop the speed to 57600. The traceback when it fails is:

```
Writing flash...
[                                                  ]Traceback (most recent call last):
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/tools/prog.py", line 282, in <module>
    main()
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/tools/prog.py", line 131, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/tools/prog.py", line 254, in pymcuprog_basic
    pagewrite_delay=args.writedelay)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4/tools/prog.py", line 141, in run_pymcu_action
    status = func(backend, *args, args_pymcu)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\pymcuprog_main.py", line 340, in _action_write
    _write_memory_segments(backend, result, args.verify, blocksize=args.blocksize, pagewrite_delay=args.pagewrite_delay)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\pymcuprog_main.py", line 387, in _write_memory_segments
    backend.write_memory(segment.data, memory_name, segment.offset, blocksize=blocksize, pagewrite_delay=pagewrite_delay)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\backend.py", line 526, in write_memory
    self.programmer.write_memory(data=data, memory_name=memory_name, offset=offset_byte, blocksize=blocksize, pagewrite_delay=pagewrite_delay)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\programmer.py", line 190, in write_memory
    self.device_model.write(memory, offset, data, blocksize=blocksize, pagewrite_delay=pagewrite_delay)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\nvmserialupdi.py", line 175, in write
    self.avr.nvm.write_flash(offset_aligned, chunk, blocksize=blocksize, bulkwrite = bulk, pagewrite_delay=pagewrite_delay)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\serialupdi\nvm.py", line 117, in write_flash
    return self.write_nvm(address, data, use_word_access=True, blocksize=blocksize,  bulkwrite=bulkwrite, pagewrite_delay=pagewrite_delay)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\serialupdi\nvm.py", line 199, in write_nvm
    self.execute_nvm_command(nvmcommand)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\serialupdi\nvm.py", line 76, in execute_nvm_command
    return self.readwrite.write_byte(self.device.nvmctrl_address + constants.UPDI_NVMCTRL_CTRLA, command)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\serialupdi\readwrite.py", line 64, in write_byte
    return self.datalink.st(address, value)
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\serialupdi\link.py", line 332, in st
    return self._st_data_phase([value & 0xFF])
  File "C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog\serialupdi\link.py", line 278, in _st_data_phase
    raise PymcuprogError("Error with st")
pymcuprog.pymcuprog_errors.PymcuprogError: Error with st
pymcuprog.pymcuprog_errors.PymcuprogError: Error with st
```
This is on a Windows 10 PC and I don't touch a single wire when I switch between trying the Arduino IDE Programming (which calls the modified pymcuprog `Version 1.2.0 - June 2021` and running the original pymcuprog from the command line. Writing the fuses and erasing the flash seems to pass:
```
Finished writing fuses.
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.01s
```
but the actual writing fails and appears to fail on the first write.

SpenceKonde, 2021-12-28:
Try a 10k pullup to Vcc on the UPDI line. That's usually whatyou need when
using a serial adapter that uses 3.3v, 5v tolerant logic levels and it
fails at high speeds but works at low speeds.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

GeorgeIoak, 2021-12-28:
Sorry, I forgot to mention that I do have a 10kΩ PU on UPDI

GeorgeIoak, 2021-12-28:
As a test I opened another command prompt and navigated to your tools directory and issued the same programming statement that I used in the original pymcuprog and it worked:

```
C:\Users\george\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.4\tools\libs\pymcuprog>pymcuprog -t uart -u COM5 -b 230400 -d attiny1626 -f "c:\Users\george\Box\Design\Fresenius Jet\Design\Software\Test Code\readTempVcc\readTempVcc.ino.t1626.20c0.mB1.pfF.v254.hex" write --verify
Pinging device...
Ping response: 1E9429
Writing from hex file...
Writing flash...
Verifying flash...
OK
Done.
```

So maybe the issue is something about handing off between erasing and writing the flash? This certainly doesn't seem to be a hardware issue since I'm able to write the flash every time when I issue pymcuprog from the command line.

GeorgeIoak, 2021-12-28:
Just some additional information I thought I would add. I was reading through the issues at pymcuprog and saw that the most recent test version added a switch to erase before programming so I installed that version and tried it out. That also seems to work just fine from the command line:

```
...

  Attempting uninstall: pymcuprog
    Found existing installation: pymcuprog 3.10.2.133
    Uninstalling pymcuprog-3.10.2.133:
      Successfully uninstalled pymcuprog-3.10.2.133
Successfully installed pyedbglib-2.19.8.92 pymcuprog-3.11.3.142

C:\Users\george\Box\Design\Fresenius Jet\Design\Software\Test Code\readTempVcc>pymcuprog write -t uart -u COM5 -b 230400 -d attiny1626 -f readTempVcc.ino.t1626.20c0.mB1.pfF.v254.hex --verify --erase
Pinging device...
Ping response: 1E9429
Erasing device before writing from hex file...
Writing from hex file...
Writing flash...
Verifying flash...
OK
Done.
```

From the command line I do sometimes see this warning `pymcuprog.serialupdi.link - WARNING - Check failed` but then it proceeds to ping the device and finishes all the other steps.

Perhaps this warning is happening during the Arduino programming and that is causing `prog.py` to fail? The only thing I haven't experimented with from the command line is programming the fuses and then erasing/writing the flash.

GeorgeIoak, 2021-12-30:
I'm not sure what could be wrong with the CP2102 (especially since it works fine from the command line, just not from Arduino) but I got some CH340G based adapters and those work in Arduino, even up to 460800 (aka TURBO). I used the same jumper wires for both adapters and left the breadboard setup completely the same.

Looking at the signals in the logic analyzer doesn't show any obvious issues why Arduino doesn't like the CP2102 but the results are very consistent.

If this board also failed to program from the command line then I'd ignore the issue and call it a bad adapter but since it works on the command line but not from Arduino I find this odd.

SpenceKonde, 2022-01-11:
This should be addressed in 2.5.6, use the write delay option if it is failing with "Error with ST" at higher speeds.

closed 2022-01-11 by SpenceKonde