SpenceKonde/megaTinyCore #589: Platfromio serialudpi

br74649, 2021-12-29:
Hi 
I'm trying to change from jtag2udpi to serialudpi 

As per docs this is the preferred supported method

I use platfromio as my ide

Is there a way to get prog.py working and using the embedded python that comes with megatinycore, it doesn't seem to like anything I'm throwing at it, I suppose it's trying to run the upload command with the platformio python not the core one

Tia 

Tony

br74649, 2021-12-29:
i figured it out i think, so without using the upload_flags i just put this in the platformio.ini and it works

[env:ATtiny1614]
platform = atmelmegaavr
board = ATtiny1614
framework = arduino
upload_protocol = custom 
upload_command = C:\Users\tony_\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1/python3 -u C:\Users\tony_\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.4.2/tools/prog.py -t uart -u COM18 -b 460800 -d attiny1614 --fuses 2:0x01 6:0x04 8:0x00 -f$SOURCE -a write

SpenceKonde, 2022-01-04:
Thanks for updae - I don't know platformIO so   \o/

I'm sure there's a better way to generalize it, but I don;t know it.

closed 2022-01-04 by SpenceKonde