SpenceKonde/megaTinyCore #592: Error in atxy2.menu.chip.402w.build.attiny=402bs in boards.txt?

SimonMerrett, 2021-12-31:
I get a build error with wrong signature versions of the ATtiny402 (yes, I have several left from my cursed section of tape). 

`avr-g++: error: device-specs/specs-attiny402bs: No such file or directory`

If I change line 1275 in boards.txt from `atxy2.menu.chip.402w.build.attiny=402bs` to `atxy2.menu.chip.402w.build.attiny=402` but leave the line 1278 as `atxy2.menu.chip.402w.upload.workaround=w` I get a successful build and upload. 

megaTinyCore version 2.5.4, Arduino IDE 1.8.7 (yes, I know I need to deal with that)
Please correct me if I'm misreporting - totally likely I've missed something.

SpenceKonde, 2022-01-01:
looks like a typo to me. Will correct.

SpenceKonde, 2022-01-04:
No, it wasn't a typo, it was my wanting to include something to indicate that a bad sig chip was in use when generating the extra assets with export compiled binary... but I forgot that I also use that number to generate the name of the chip.

SimonMerrett, 2022-01-04:
Understandable. Is there a way to fix? I can live with the manual edit, so that question is purely out of interest rather than necessity. Who'd have thought bad signatures would have cost so much time...

SpenceKonde, 2022-01-04:
I did the easiest thing possible, I returned it to 402, and to keep the
name in the asset distincti just reuse the upload.workaround so the
filenames will indicate that it's a t402w, which is good enough

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SimonMerrett, 2022-01-04:
Closing as fix has been committed.

closed 2022-01-04 by SimonMerrett