SpenceKonde/megaTinyCore #593: Conflicting libs?

a7hybnj2, 2022-01-03:
This is more of a place to get the conversation started because I haven't 100% tracked down what is going on yet.

Here is what is happening.

I made a project months ago using 2.2.9 That project consisted of an i2c oled and I used 
```
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
```

Today I updated to 2.5.4 and the olded is 'skewed'. All the lines are offset by a few more than the ones before it. Anyway when I roll back to 2.2.9 and reupload it works perfectly again.

`attiny3216`

SpenceKonde, 2022-01-03:
How is it connected to the screen, SPI or I2C?

a7hybnj2, 2022-01-03:
It is connected via i2c. The `sda` and `scl` pins. On address: `0x3C`... I don't know why I have the `...SPI.h>' file.

SpenceKonde, 2022-01-03:
Uhoh....

Can you try to pinpoint the version where this started failing?  

2.2.9 was still based on the old I2C implementation, and it's been completely replaced now.

a7hybnj2, 2022-01-03:
@SpenceKonde Yes, I can do that but! This particular project has to go back to my client so I will have to build up a little breadboard thing. Give me a day or two and I will test it out.

a7hybnj2, 2022-01-04:
@SpenceKonde Okay. I just tested it out. 2.5.1-2.5.3 cause a fail to compile but 2.5.4 compiles but breaks in the previous way described. 2.4.2 works great. I just used the generic example "ssd1306_128x64_i2c" provided by the "ADAFRUIT SSD1306" with the only modification being the address of my oled being 0x3C vs 0x3D...

SpenceKonde, 2022-01-04:
Aw shit.... That's no good. Got a picture of what it looks like?

Does this library show the same issue? https://github.com/datacute/Tiny4kOLED

@MX682X  - this is not the kind of problem one wants to encounter with a new wire implementation.

a7hybnj2, 2022-01-04:
Correct on 2.4.2
https://user-images.githubusercontent.com/36831279/147999948-adc8a784-f22c-4fa1-a577-a35d4fa58e67.mp4
Garbled up on 2.5.4
https://user-images.githubusercontent.com/36831279/147999998-9b494b66-7e33-4cb5-ad98-1404e243cb03.mp4

I can look at the Tiny4kOLED over the weekend. It does say its for the tiny85 but I guessing I can get it to work on the 3216 since you suggested it.

a7hybnj2, 2022-01-04:
Also, Thanks! This project is sweet and I have been using it with the series 2 stuff since the [beginning](https://github.com/SpenceKonde/megaTinyCore/issues/309). I used to suffer through the jtag2updi but the new serialupdi is so fast and much easier to implement. Just wanted to mention that since I'd been on 2.2.9 until now.

SpenceKonde, 2022-01-04:
The Wire version is the one you want to use - it should work everywhere that has a working Wire library. It was originally made to deal with the whole USI vs SPI thing on the tiny85 and other classic tinies, but they have since expanded it to normal Wire (so it works with anything that has a working Wire library), some other optimized+cut-down classic tinyAVR library.... and a pure bitbang mode that makes a mockery of the standard, but runs super fast and is claimed to work. But, it's the Wire mode that's interesting to us. If something has broken in wire, that's a big problem.

a7hybnj2, 2022-01-04:
Okay. I will try it out as soon as I get my desk cleared off.

MX682X, 2022-01-04:
OK, that is weird. I have an OLED screen too and testd it. Sometimes, there was a an offset of one pixel, but I thought it was because itt was a cheap screen.

MX682X, 2022-01-04:
OK, that is weird. I have an OLED screen (128x64) too and testd it. Sometimes, there was a an offset by two rows, but I thought it was because it was a cheap screen and probably some cheap IC copy. That was one of the reasons, why I haven't tested it with the adafruit library.
I have a setup with an 128DA48 and the OLED. I tried using the u8g2 library for interfacing with it. I tried the constructor for ssd1306 - and had the row offset again. When trying to use the constructor for SH1106 - it works perfectly. 

So what I would ask you to try: install the u8g2 library and tell us, how it works with it. also try both ICs. Just to be sure.

SpenceKonde, 2022-01-04:
I wonder what we're doing differently from the stock Wire that triggers this - it's not plausible that this is the only part impacted by whatever it is... But ugh, filling a screen is so much data it's almost hopeless to try to track it down on a 'scope;

MX682X, 2022-01-04:
I have multiple theories and a couple of idea how to tackle the issue. My first theory was a wrong buffer size (maybe the adafruit library is using a bigger buffer then the wire one, which might lead to lost data). Later I had the thought that it might be the wrong IC, since the adafruit example says `See datasheet for Address; 0x3D for 128x64, 0x3C for 128x32` but the tread opener is using 0x3C. From the videos, it seems like they are using the 64pixel version. But this does not explain why the old one was working but the new one isn't. 
After checking how it works with the u8g2 library, I will write a script that will move a pixel around. Since only 2 pixels have to be updated, the amount of data will be small and trackable. Compare the data from the old Wire version and the new one.

a7hybnj2, 2022-01-09:
@MX682X I just took a look at u8g2. I don't know anything about what you were saying so I just pulled u8g2 from the library manager and I loaded the example "u8g2 - full_buffer - graphics test" and I uncommented this line `U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display` and it works fine.

a7hybnj2, 2022-01-09:
I just loaded the adafruit example again and put in a
```
for(int i=0; i<=display.width()-20; i+=20) {
    display.fillRect(i, 0, 5, 64, SSD1306_WHITE);
    display.display();
    delay(1);
}
```
which should make vertical lines across the screen and this is what it looks like.
[image]

MX682X, 2022-01-10:
@a7hybnj2 You have used the software I2C implementation. You can recognize that by _SW_I2C. I don't think it is using the wire library. try to use following constructors:
`U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);`
and 
`U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);`

I want to know if the IC is in fact correct (there are actually barely any differences according to the u8g2 library.
One of th differences is the acceptable clock. Since I had some trobble with another I2C slave too, that only wanted to work at about 125kHz, even though it was specified below 100kHz, I'd like you to try out different clocks with the adafruit library. The last two ardguments are the clocks. Try out something like this for 50kHz, and maybe try 200kHz. Just to see if there is a difference.
`Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET, 50000, 50000);`

grandaspanna, 2022-01-10:
Maybe a dumb idea, but have you tried pullups on the i2c lines?

a7hybnj2, 2022-01-10:
> U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
and
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

I tried both of these and they both worked... ¯\\_(ツ)_/¯ 

I don't have any idea of what is going on behind the curtain when I am using these libraries, I am sure it is beyond my level. All I know is it used to work on 2.4.2 and doesn't on newer builds which makes me think something had changed that messed it up. I am seeing some of these other packages are working correctly so I could for my simple project just switch to them but I am sure someone is going to be using the adafruite libs in the future because they are very popular.

SpenceKonde, 2022-01-10:
We cannot break the adafruit library, yeah. 

It's losing 4 bytes per 128 pixels. @MX682X - do we by any chance put the device address into the buffer, and thus end up with only 31 bytes for data? That is a bug if so - we must have the stated amount of space for **data** in the buffer.

SpenceKonde, 2022-01-10:
Huh, yeaaaah it looks like that's what we do, and this is exactly the reason that we have 32-byte buffers for everything that could possibly squeeze in that much buffer space! 

Let's see how big the space hit is for adding a byte to the buffers. That's what I'll do for 2.5.5

SpenceKonde, 2022-01-10:
Looks like around 78 bytes. Impressive how much space it burns just to have 1 extra byte in buffer when that makes it not a power of two, but we must have 32 bytes FOR DATA. Assuming checks pass I'll do a release tonight. (both cores are getting a release tonight because there are a bunch of critical issues, some of which I'm surprised haven't been reported yet except on DxCore.

SpenceKonde, 2022-01-10:
58 for master, 54 for slave, and 78 for SFRRanger for some weird reason.

SpenceKonde, 2022-01-10:
Managed to reclaim some of that space by performing intermediate calculations with a smaller datatype when buffer size is under 128b.

closed 2022-01-10 by SpenceKonde

a7hybnj2, 2022-01-11:
Perfect I will try to keep an eye out for 2.5.5 and give it a test when I see it. Thanks!

SpenceKonde, 2022-01-11:
20 bytes on slave devices so cost there is down to 34. 28 bytes on masters, so total cost 30b for masters, except for the anomalous SFR Ranger thing.

MX682X, 2022-01-11:
> do we by any chance put the device address into the buffer

> Huh, yeaaaah it looks like that's what we do

Wait, what? Where? beginTransmission is using a seperate variable in the struct called _clientAddress. Data is being written exclusively by write and is put into the txBuffer. The only possibility I see that *might* cause loosing a byte is the round robin system, but I have not found a case where it applies. But it explains why u8g2 is working, it is using 24byte transfers. We could use defines to tell the adafruit library to use less space though, it checks for a couple defines in its begin function to allocate a buffer.

a7hybnj2, 2022-01-19:
Hey, Just wanted to let you know I tested it with the previous examples there were giving issue and everything appears to be working. Thanks.

commit c48ee1a2 (2022-01-10): Fix #593 and reclaim some of the space the fix cost.