SpenceKonde/megaTinyCore #594: serialupdi baud rate issue

specternecter, 2022-01-04:
I've been working on using a 2-series ATtiny to control programming and debugging automatically. It runs at 

It took me a while to figure out what was going on here but I finally found it. I modified the baud rate in programmers.txt to be 230404 so that I could watch the serial port and this is what I got...

///////////////////////////////////////////////////////////////////////////////////////////////////////////////
000000: Connect (DOWN): 2021-12-29 11:08:24.9249200 +13285267704.9249191
 Device "\??\USB#VID_10C4&PID_EA60#0001#{0100fdd7-be5a-4808-91f5-05002bc60f72}" has been connected to the system

000001: Create (DOWN): 2021-12-29 11:08:43.0307808 +18.1058608
 Process Id: 27288

000003: I/O Request (DOWN): 2021-12-29 11:08:43.0343936 +0.0000320
 IOCTL_SERIAL_SET_QUEUE_SIZE: Set queue size
	· Input 		= 4096
	· Output		= 4096

000004: I/O Request (UP): 2021-12-29 11:08:43.0343968 +0.0000032
 IOCTL: IOCTL_SERIAL_SET_QUEUE_SIZE

000005: I/O Request (DOWN): 2021-12-29 11:08:43.0344096 +0.0000128
 IOCTL: IOCTL_SERIAL_GET_TIMEOUTS

000006: I/O Request (UP): 2021-12-29 11:08:43.0344112 +0.0000016
 IOCTL_SERIAL_GET_TIMEOUTS: Retrieve timeouts
	· Read interval         		= 0
	· Read total multiplier 		= 0
	· Read total constant   		= 0
	· Write total multiplier		= 0
	· Write total constant  		= 0

000007: I/O Request (DOWN): 2021-12-29 11:08:43.0344224 +0.0000112
 IOCTL_SERIAL_SET_TIMEOUTS: Set timeouts
	· Read interval         		= 0
	· Read total multiplier 		= 0
	· Read total constant   		= 1000
	· Write total multiplier		= 0
	· Write total constant  		= 0

000008: I/O Request (UP): 2021-12-29 11:08:43.0344240 +0.0000016
 IOCTL: IOCTL_SERIAL_SET_TIMEOUTS

000009: I/O Request (DOWN): 2021-12-29 11:08:43.0344304 +0.0000064
 IOCTL_SERIAL_SET_WAIT_MASK: Set wait mask
	· Wait Mask		= SERIAL_EV_ERR (0x80)

000010: I/O Request (UP): 2021-12-29 11:08:43.0344320 +0.0000016
 IOCTL: IOCTL_SERIAL_SET_WAIT_MASK

000011: I/O Request (DOWN): 2021-12-29 11:08:43.0344416 +0.0000096
 IOCTL: IOCTL_SERIAL_GET_BAUD_RATE

000012: I/O Request (UP): 2021-12-29 11:08:43.0344416 +0.0000000
 IOCTL_SERIAL_GET_BAUD_RATE: Retrieve Baud Rate
	· Baud Rate		= 230400

000013: I/O Request (DOWN): 2021-12-29 11:08:43.0344448 +0.0000032
 IOCTL: IOCTL_SERIAL_GET_LINE_CONTROL

000014: I/O Request (UP): 2021-12-29 11:08:43.0344464 +0.0000016
 IOCTL_SERIAL_GET_LINE_CONTROL: Retrieve line control
	· Stop bits		= 1
	· Parity   		= No
	· Data bits		= 8

000015: I/O Request (DOWN): 2021-12-29 11:08:43.0344496 +0.0000032
 IOCTL: IOCTL_SERIAL_GET_CHARS

000016: I/O Request (UP): 2021-12-29 11:08:43.0344496 +0.0000000
 IOCTL_SERIAL_GET_CHARS: Retrieve special characters
	· EOF  		= 0
[... 390 lines cut ...]
        :param baud: baud rate
        """
        self.logger.info("Opening port '%s' at '%d' baud", port, baud)
        if passthrough == True:
            try:
                self.ser = serial.Serial(None, manually_passed_baud, parity=serial.PARITY_EVEN, timeout=1, stopbits=serial.STOPBITS_TWO) 
                self.ser.port = port
                self.ser.dtr = False
                self.ser.rts = False
                self.ser.open()
            except SerialException:
                self.logger.error("Unable to open serial port '%s'", port)
                raise
        else:
            try:
                self.ser = serial.Serial(None, baud, parity=serial.PARITY_EVEN, timeout=1, stopbits=serial.STOPBITS_TWO)
                self.ser.port = port
                self.ser.dtr = False
                self.ser.rts = False
                self.ser.open()
            except SerialException:
                self.logger.error("Unable to open serial port '%s'", port)
                raise

...which confirms that this setting the port to 115200 is unnecessary. It really appears to be a bug in serialupdi, in which it's not listening to the baud rate sent by programmers.txt until well after it should

SpenceKonde, 2022-01-04:
The initialization at 115200 baud is intended, I added it. 
230400 baud is pushing the limits of what the chip is capable of at the default UPDI clock speed (IIRC it's out of spec, but seems to work at least under common use cases). It has to start at a lower speed, and then only after it's set the UPDI clock appropriately can it confidently boost the baud rate.
If there are communication problems it will also close and reopen at 300 baud to send a 0x00 to create a break condition, then close and reopen at 115200 to configure the updi clock and then go back to the baud rate the user requested.

SpenceKonde, 2022-01-11:
SerialUPDI options receive an update 2.5.6. 

However, the behavior about which you complain (note that you have not described how this behavior is a problem) is unchanged. It cannot communicate with all adapters at 230400 baud in it' power-on-reset state (I don't know why it works at all on some!) because it defaults to the minimum clock option for the programming interface (so that it can be programmed over UPDI even when running at 1.8V). To ensure operation at 230400 baud will work, it needs to be increased. Obviously, we can't do this at 230400 baud. I did find some adapters today that would not work at 230400 baud without changing from /4 (4 MHz) to /2 (8 MHz), which are the same adapters that don't work at 460800 baud without changing to /1 (16 MHz).

closed 2022-01-11 by SpenceKonde