SpenceKonde/megaTinyCore #604: -D {upload.serialparams} still unparsed, warning t3217 overwrites/duplicity still in board.txt

apws, 2022-01-11:
hi, I dont know how its possible but I burned optiboot and tried upload sketch and got again this:
(update to 2.5.5 required to delete subfolders of arduino15, ya)

Sketch uses 688 bytes (2%) of program storage space. Maximum is 32256 bytes.
Global variables use 4 bytes (0%) of dynamic memory, leaving 3068 bytes for local variables. Maximum is 3072 bytes.
C:\Users\petra\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.5/avrdude.conf -v -pattiny3226 -carduino -D {upload.serialparams} -Uflash:w:C:\Users\petra\AppData\Local\Temp\arduino_build_780129/megaTinyCore_Blink.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.5/avrdude.conf"
avrdude: warning at C:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.5/avrdude.conf:16457: part t3217 overwrites previous definition C:\Users\petra\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.5/avrdude.conf:16002.

         Using Port                    : com1
         Using Programmer              : arduino
An error occurred while uploading the sketch
avrdude: ser_open(): can't open device "\\.\com1": The system cannot find the file specified.

avrdude done.  Thank you.

apws, 2022-01-11:
I now upgraded also arduino from 1.18.16 to 1.18.19 and problem still persists

SpenceKonde, 2022-01-11:
The avrdude.conf issue is completely new to me and a trivial fix. I hadn't seen it internally because I've only ever uploaded to the 3226 parts with SerialUPDI.

The behavior of the IDE with serialparams is whack, it just will not substitute that in. It substitutes in *other* variables, but not that one! changing the name didn't fix it either! (you can see for yourself  that it's defined, and ignored! Doesn't matter if I change the name or try to have another variable include it, and it makes no fucking sense) 
upload.prog_interlock parses. Everything else parses. But that one doesn't. 

I will release a 2.5.6 later today to correct this by putting those parameters into extra_params, but I shouldn't have to fucking do that!

apws, 2022-01-11:
ya, the t3217 is my fault, I already mentioned it but didnt created issue for it, so it was forgot... the serialparams is really weird thing, thanks for your workarounds

commit 9c3df9a3 (2022-01-11): Fix #604
closed 2022-01-11 by SpenceKonde