SpenceKonde/megaTinyCore #613: Reading memory with the SerialUPDI tool - AttributeError

fijam, 2022-01-14:
I'm trying to read the memory of my chip using the SerialUPDI tool provided with megaTinyCore installation (2.5.6).

I'm using Python 3.7.2 as recommended in https://github.com/SpenceKonde/DxCore/blob/master/megaavr/tools/ManualPython.md

```
PS C:\Users\fijam\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.6\tools> .\python3\python.exe prog.py -a read -d attiny212 -u COM6 -f output.hex
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.0 - June 2021
Using serial port COM6 at 115200 baud.
Target: attiny212
Action: read
File: output.hex
Pinging device...
Ping response: 1E9121
Reading...
Traceback (most recent call last):
  File "prog.py", line 282, in <module>
    main()
  File "prog.py", line 131, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "prog.py", line 269, in pymcuprog_basic
    max_read_chunk=None if args.read_chunk <= 0 else args.read_chunk)
  File "prog.py", line 141, in run_pymcu_action
    status = func(backend, *args, args_pymcu)
  File "C:\Users\fijam\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.5.6\tools\libs\pymcuprog\pymcuprog_main.py", line 201, in _action_read
    result = backend.read_memory(args.memory, args.offset, args.bytes, args.max_chunk_size)
AttributeError: 'Namespace' object has no attribute 'max_chunk_size'
```

SpenceKonde, 2022-01-15:
Thanks, yeah, it's been busted for close to a year and nobody noticed lol....

SpenceKonde, 2022-01-17:
Fixed by 34396f7d25ed9b324e6d5c3289d90651070e1e30

closed 2022-01-17 by SpenceKonde