SpenceKonde/megaTinyCore #621: Tool avrdude is not available for your operating system - again

denis-stepanov, 2022-01-18:
I am sorry to say that the issue #557 has resurfaced. The last version I have been able to install was 2.5.4.

Not urgent - I am quite OK with the older versions I have. Thanks.

SpenceKonde, 2022-01-19:
Argh, yeah I forgot to change it in the json update script. I can't patch the json file without doing a new release (I did that last time, and everyone started getting java null pointer exceptions when they tried to update it with board manager)

SpenceKonde, 2022-01-19:
Will be automatically fixed for next release.

closed 2022-01-19 by SpenceKonde

SpenceKonde, 2022-01-19:
You're in luck, someone reported a critical bug so I'm doing a 2.5.8 release now.

denis-stepanov, 2022-01-21:
Thanks. 2.5.9 installs OK.

SpenceKonde, 2022-01-21:
Unfortunately that error message still displays for everyone and I have no idea how to fix that without causing people to get null pointer exceptions under circumstances unclear to me. 

That was a pretty big fuckup on my part, I gotta say. My json file now appears to be in a bad state that cannot be corrected.

denis-stepanov, 2022-01-22:
No, for me it could be installed both on 64 and 32 bit Linux. But there's now a new warning about avrdude (do not have the text at hand) not preventing installation; not clear how serious. I was only testing 2.5.9 programming via optiboot on 64 bit, this was working OK.

SpenceKonde, 2022-01-22:
Yes, and I've gotten half a dozen bug reports about it and I don;t know of any way to fix it that doesn't force some unknown fraction of users to delete their arduino15 folder otherwise they'll get null pointer exceptions when they try to use board manager.

SpenceKonde, 2022-01-22:
Half a dozen in like 24 hours.