SpenceKonde/megaTinyCore #624: analogRead() conflicts with analogReadEnh()

EclipseBoom, 2022-01-22:
Using a 1626 at 2.7V - 8MHz external clock - core version 2.5.6

Mixing use of analogRead(pin1) and analogReadDiff(pin2, pin3, 12, 16) causes bad values to be returned for either or both.

Fixed by changing analogRead(pin1) to analogReadEnh(pin1, 10, 1) --- tried analogReadEnh(pin1, 10, 0) but did not work

Was reading the results as quickly as possible so it is possible there is some PGA init/de-init time delay that is not being respected or "ready" flag that isn't being checked.  

So basically keep the PGA enabled I guess.  Don't know if this is a bug.  Perhaps a note in the ADC guide would cover it.

SpenceKonde, 2022-01-23:
Had you turned off low latency mode? I wonder if we're getting errata'ed here.... 

What voltages and approximately what impedance was connected to te pins? Did you change the ADC clock or any other ADC-related settings. 

There seems to be some errata related to the ADC initialization delay.

SpenceKonde, 2022-01-23:
Also what reference were you using?

EclipseBoom, 2022-01-24:
30k impedence on the differential reading ~200k on the single-ended, no change to latency or sample duration.  VDD reference.

Basically just using the defaults with the resolution set to 12 and gain to 16 on the differential and the single-ended at 10/1.

The glitchy readings cleaned up by activating the PGA at a gain of 1.  So it stays initialized I guess and just changes settings.

Also, I noticed the ADC conversion rate with the 12/16 differential reading was significantly higher than listed in the ADC guide.  I didn't write it down but it seems like it was about 4.2uS per conversion vs 10uS in the guide.  This was in "called" conversions not free running or burst mode.

SpenceKonde, 2022-01-25:
Can you try a few experiments for me? I'm not seeing the same behavior, I don't think. 

With the combination that gives bogus results, try:
1: taking several consecutive readings with analogRead() (and analogReadDiff() for good measure) printing each value 
2: If the values given get better after a few reads, then that mostly answers our question. I will note that per the datasheet the ADC is optimized for 10k ohm impedance - but I would think that since I have slowed it down significantly with SAMPDUR, that shouldn't be the problem. 
Passing a larger value to analogSampleDuration (default is 14, which gives 15x the normal sampling duration) would help, but like I said, the datasheet **implies** that it's usable for 10k impedance with at the rated clock speed and default settings, though they carefully avoid anything that could give any quantitative guidance on how to choose appropriate settings.
3.print out the settings of the ADC with
```
Serial.printHex((uint8_t *)&ADC0,14,',');
```
(outputs the 14 non-reserved configuration registers, comma separated, in hexadecimal) 
before and after each type of measurement, and also let me know your F_CPU. 
4. Time the measurements more accurately:
```
uint16_t start,end;
int32_t reading;
start=TCB1;.CNT;
reading=analogRead(pin);
end=TCB1.CNT;
Serial.print(reading);
Serial.print(" took ");
Serial.print(end - start);
Serial.println(" timer ticks")'
```
(assuming you're using the default timer, TCB1 for millis/micros. Each timer tick is 2 system clock cycles if it's a TCB being used for millis (except at 1 MHz). Using the micros to time internal operations can be deceptive because it's surprisingly slow to return (it grabs the number as soon as it can but then it has to calculate a number of microseconds... and that involves multiplying an uint32 (the number of elapsed millis) by a 16-bit constant (1000), and using ersatz division (at most F_CPUs) to to divide the timer counts by (F_CPU/2000000)... 
|F_CPU      | micros() return time |
|    24 MHz |   5 us | 
|    20 MHz |   6 us | 
|    16 MHz |   6 us | 
|    12 MHz | 10 us | 
|    10 MHz | 10 us |
|     8 MHz |  11 us | 
|     5 MHz |  23 us |
|     4 MHz |  21 us | 
Notice how it doesn't really get faster over 50% speed increases - that's because the powers of 2 can do exact division with a few rightshifts, while the rest get rightshifted until the tick value is between 750 and 1500, and then ersatz division is applied. In the case of the Frustrating Fives, that's  a 6-term approximation implemented in hand-optimized assembly, and the Terrible Twelves need 9 terms, also hand optimized. in order to meet the criteria I used to assess them:  That a scaled (after the rightshifts and before the ersatz division) count N shall never result in a micros return value more than N+1 would, nor more than 2 consecutive scaled ticks return the same value nor shall more than 1 micros return value be skipped. 

The 12's and 5's - the most important clock speeds, turned out to be the hardest and the third hardest respectively (7/14/28 is second hardest).. 25, 27, 15/30, and 9/18/36 are all easier!

EclipseBoom, 2022-01-25:
I will do some experiments.  I have to deliver this thing so now that I've got it working as well as I can it will be lower priority to delve into the guts.

I did the conversion timing by doing 10,000 conversions in a for loop and dividing at the end so the millis()/micros() calculations should be minimized.  No printing in the loop just the conversion calls.

I'm running the Tiny1626 at 8Mhz with a 2.7V system voltage.  External crystal oscillator so timing is good within a few ppm.

I observed the "better" / "worse" performance by looking at the data plotter tool in the IDE on a static value (I have a resistor divider of around 200k) to monitor the system voltage.  That is the single-ended source.  The differential source is a wheatstone bridge of 30k resistors in a magnetic field sensor.  During the testing both should have given a single value but of course there is some noise esp. with the differential reading at 16x PGA gain.

Default system timer setting from IDE menu and no modifications to the default for ADC settings.

`  uint16_t vpwr;
  // pwr connection
  vpwr = analogReadEnh(VPWR_PIN, 10, 1);  // to be like standard analogRead()
`

`  int32_t spin_val;
  spin_val = analogReadDiff(VSPIN_P_PIN, VSPIN_N_PIN, 12, 16);  // update spin value 
`

SpenceKonde, 2022-01-30:
Let me know how the above tests go. Any resolution is not going in 2.5.10.

 No work that requires me to test with hardware is going into this or megatinycore until ATTinyCore 2.0.0 is released. I have been promising them that update for over 6 months. Only fixing things that can be fixed with the information in the bug report and validated with CI.

SpenceKonde, 2022-02-17:
I was heavily mixing analogReads and analogReadEnh and saw no anomalies not long ago . I think this is a case of the impedance being too high for the sample duration and ADC clock speed,. An early version of this core set a bogus constant value for the register controlling CLK_ADC. Specifically, it maxed the clock out at F_CP. Util I realized it was m y software bug, I was profoundly disappointed with these parts. It worked we;; enough that you can almost belive the ADC is just increidbly flaky but that' 

If ytou're always using analogReadEnh with the PGA, you'll definitely do better under adverse conditions like measurig relatively high impedance sources, measuring sources with radicall different voltages on them in succession, etc than you otherwise would (That PGA with 1x gain definitely seems to help sthe quality of measurements considerably.

SpenceKonde, 2022-04-07:
Please reopen this if you have any information on this. Example code that could reproduce this without whatever weirdo sensor your using, would be ideal. Because I don't have that wacky sensor and never heard of it unti this issue.. I am otherwise unable to provide any assistance.

closed 2022-04-07 by SpenceKonde