SpenceKonde/megaTinyCore #625: With 2.5.9 loop seems to restart after Interrupt - did work as expected with 2.4.2

DietmarM, 2022-01-23:
My code worked as expected with 2.4.2 but after I had installed 2.5.9 the loop seems to be restared after an IRQ from the PWM in arrived and does not get finished after. The behaviour changed at least with 2.5.1 which I also tested.

Reading a RC PWM on PIN_A2 which comes each 20ms. Running on 10MHz

Any Idea also what the Index error means? Is it connected to the stopped loop?

I stripped the code to this example:

//Software PWM only works with 2.42 and not with 2.5.9
//Arduino 1.8.13
//Also see error in Arduino:
//Index error: could not find referenced tool name=avrdude version=6.3.0-arduino17or18 packager=arduino

volatile static uint8_t RXtype = 0; //0= software PWM 900-2100; 

//PWM Eingang vom Empfänger
volatile const byte RX_PIN = PIN_A2; //PIN_A2 
volatile unsigned long RX_rising_start = 0;
volatile long RX_length = 0;
volatile long RX_OVF = 0;
volatile unsigned long RX_last_falling = 0;
volatile unsigned long RX_last_rising = 0;
volatile unsigned long PWM_delta_last_Puls = 0;
volatile unsigned int RX_CH_act = 0;
volatile bool RX_valid_pulse = false;
volatile bool RX_valid_sequence = false;

volatile bool RX_new_value = false;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200); // Setup serial debug connection

  if (RXtype <= 1) {
    //PWM Eingang von RX
    pinMode(RX_PIN, INPUT);  // PWM Eingang
    attachInterrupt(digitalPinToInterrupt(RX_PIN), RXchange, CHANGE);
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print(analogReadEnh(ADC_VDDDIV10) / 2);
  Serial.print("V "); // print value

  Serial.print(RX_CH_act + 1);
  Serial.print("CHs "); // print value
  //Serial.print(PWM_delta_last_Puls);
  Serial.print(RX_rising_start);
  Serial.print("t "); // print value
  Serial.print(RX_length);
  Serial.print("RX_length "); // nether gets printed with 2.5.9 but with 2.4.2 when PWM signal is present with 20Hz

  delay(100);
}

void RXchange() {
  if (RXtype == 0)   //Software PWM 
  {
    if (digitalRead(RX_PIN) == HIGH) {
      RX_rising_start = micros();
    }
    else {
      unsigned long PWM_falling = micros();
      unsigned long PWM_Puls_Laenge =  PWM_falling - RX_rising_start;
      PWM_delta_last_Puls =  PWM_falling - RX_last_falling;
      if (PWM_Puls_Laenge > 700) {
        RX_length =  PWM_Puls_Laenge;
      }
      RX_new_value = true;
      RX_last_falling = PWM_falling;
    }
  }
}

SpenceKonde, 2022-01-23:
The index error is a bug in the board manager json entry for 2.5.8, which will be fixed a few days after I do the next release (I want as many people as possible to be on version 2.5.9 or later or 2.5.7 or earlier when I do it - versions of the IDE 1.8.9 or earlier, if they have 2.5.8 installed when I fix that,  will have to manually delete the core, otherwise they cannot update, remove, or uninstall it. There were other ways that could be used to get rid of the error, but those were found to be even worse (they cause null pointer exceptions if you have installed the offending version). 

Regarding the interrupt issue, which mode do you have selected in the attach interrupt mode menu? if you have manual mode engaged, that's expected since you don't enable interrupts on and of the pins. And you can also select the old implementation to go back to the old one and get exactly the same attachInterrupt behavior as before. 

Also, IMO, attachInterrupt is not really up to the task of accurately measuring pulse lengths on the order of tens of microseconds while running at 10us. Any construction in which you pass a function pointer to something to set it as the interrupt is DOA because that forces a gigantic prologue and epilogue. 16 registers or something have to be saved and restored in addition to the usual, so that's 48 clocks plus another dozen just with bare minimum overhead, then thee's another 20-50 depending on the pin to find which pin it is, realize it's your pin, and call the function, and your function, because you're using all those large datatypes, has to save and restore 10 more registers (30 clocks), and you're using digitalRead instead of digitalReadFast; digital read is > 50 clocks I think.... and you call micros (another notoriously slow function, even after I optimized the hell out of of it; 10 MHz is running hand-tuned assembly and it still takes 10us to execute. I'm guessing the overall execution time of that interrupt is something on the order of 35-40 us. And you're trying to measure pulses at 20 kHz, (50us per cycle, so 25 per pulse if 50% duty cycle?) You are not even meeting the nyquest criteria.... 

Doing it by ditching attachInterrupt and writing the ISR directly would save  5-7 us. switching to digitalReadFast() wold save another 5-ish, but it['s still too slow. 

This is really the kind of situation that you want to deal with by using the unused type B timer in input capture pulse width measurement mode, you can measure pulses up to 6553.5 us with 0.1us resolution, or pulses twice that long with half the resolution, and there's a trick you can use to double it if you have to.

Unfortunately I don't have a tutorial ready for that. 

 pulseIn() is also accurate to 1us, and a carefully written polling loop is accurate to within 0.5us when running at 10 MHz.... 

but yeah, attachInterrupt definitely should work (unless it's set to manual), and I thought I had fixed the bugs with the new implementation, I am not sure why it's not and can't diagnose it without knowing what interrupt mode you had selected.

DietmarM, 2022-01-23:
Thank you so much for your input. I'm still learning to use the attiny more and more.
Sorry - I can't find the interrupt mode menu: where can I find it?
This is how I defined it in the code:  
attachInterrupt(digitalPinToInterrupt(RX_PIN), RXchange, CHANGE);
Anything missing to work in 2.4.2 as 2.5.9?

I run the µC at 10MHz because the voltage can finaly go down below 3V. I use TCA to get 2 variable frequency PWM as output with high res, TCB0 for one PWM input and I would like to have one more PWM input for what the "software with ISR" would be enough (~ +/- 5µs what I observed) - if it not crashes the loop.
Every 20ms is only called with 50Hz - works nice in 2.4.2 version.

SpenceKonde, 2022-01-24:
Tools -> attachInterrupt Version. It is not present in 2.4.x - it was one of the big features. But you say it works in 2.4,2. It is almost certainly a problem related to that new menu. The new menu has three options - all ports with the new method (this is supposed to work, and is the default), manual (you need to call a function to enable it for a given port  see the interrupt reference for details, for it to work; and the "old version" which will fully revert that to the 2.4.x version (I expected there to be problems, but I thought I'd fixed them all at this point)

The idea behind the change was "okay, attachInterrupt was always bad, it's even worse on modern AVRs, we shouldn't use it because directly defined interrupts have a fraction of the overhead (both flash and speed), but the old implementation would grab every pin interrupt available if it was referenced ones in any code, including in a library, thus making it impossible to define (performant, small, efficient) interrupts yourself - your only option was using attachInterrupt yourself, and that sucked". 

I don't see you using the timers, so I'm sort of confused about what you're hoping to do - generate PWM? Or use it as an input? 

And also uhhh.... 50 Hz is very different from 20 kHz.... 20 kHz is 50us per cycle and 25 per pulse (250 clocks) 50 Hz is 20ms  per cycle, 10ms per pulse -> 100000 system clock cycles. the two numbers you've given for frequency are different from each other by a factor of 400... I think it's clear that there was a miscommonication w/regards to one of those numbers >.>

SpenceKonde, 2022-01-24:
Oh - I suspect that you're uising PIN_A2 when you mean PIN_PA2, and it's is only working correctly here by coincidence. PIN_A2 is the pin corresponding to for ADC channel . ). 

PIN_PA2 is the pin referred to as PA2 in the datasheet, pin 2 in PORTA, which is what I think you meant. 

Those happen to be the same pin on the tinyAVR 2-series. They aren't on most other parts.

DietmarM, 2022-01-24:
Thanks again! I can confirm it works in 2.5.9 with interrupt setting to Old Version. The two other options don't work for me.

I reduced the code to reproduce the issue and find out what happens. This is 100% code (above) I can reproduce the issue with.
Thanks for explaining/hinting to my wrong implementaion PIN_A2/PIN_PA2 - but as you may be expected no difference on the 1624.

SpenceKonde, 2022-01-24:
Will investigate- I thought I'd gotten all the bugs put of the new
implementation - so it hangs as soon as the interrupt would fire?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

DietmarM, 2022-01-24:
When I start the µC without PWM connected it sends on the serial all (including "RX_length ")
As soon PWM comes in it restarts in the middle of the print statements.
PWM disconnected and  it sends on the serial all again (including "RX_length ")

DietmarM, 2022-01-25:
Serial.print(analogReadEnh(ADC_VDDDIV10) / 2);
If it is connected to the ReadEnh #624  ?

closed 2022-01-25 by SpenceKonde

SpenceKonde, 2022-01-25:
Ugh, what a stupid bug. 
To work around the fact that the io headers all include all three PORT registers, even on parts that have only one or two ports, I have my own defines in core_devices.h to tell me how large I need to make the arrays of function pointers, There was no define for PORTA because it wouldn't let me save any flash - all devices need an 8-entry PORTA, even the 8-pin parts with only 6 I/O pins,, since they have a "hole" in the middle, and the bit position is used to index the array. I'd just forgotten that when I was last poking at the ISR definitions there to fix one of the other bugs. 

Note that the manual mode was not impacted by this bug.

commit 2f33234c (2022-01-25): Fix #625

Attaching interrupts to pins on PORTA with the default (all ports enabled) option would trigger BAD_ISR -> dirty reset -> reset  sequence... because I put the same test for a PORTx_PINS define that I do for the other ports. But I don't define PORTA pins because there is no part that would have been able to benefit from greater efficiency. Unfortunately I had forgotten that when I  tested for it to decide whether to define the ISR.

This is corrected and will be in the next release.