SpenceKonde/megaTinyCore #650: Timing problems while using ATTINY3227 with internal 1MHz

Uwne, 2022-02-14:
Hi,

for a simple project I want to use the internal 1MHz clock.
After flashing the "blink without delay" code my led blinks @500ms, not @1000ms.
If I use more than 1MHz system clock it works fine.

Any idea?

BR
Uwe

SpenceKonde, 2022-02-14:
Sounds like a bug in the core, I thought I'd fixed all of those 1 MHz-specific issues, but it sounds like not. At all other speeds when a type B timer is used for millis, the timer clock sourece is set to CLKPER / 2, and the peroiod of the millis timer (default TCB1) is set to F_CPU/2000

SpenceKonde, 2022-02-14:
I will add that 1 MHz speed doesn't get exercised much - most people are running these things at the full 20 MHz as far as I know.

Uwne, 2022-02-14:
Thank you very much for your quick response. Because of the power consumption I need to use the internal 1MHz.

SpenceKonde, 2022-02-15:
The real power savings comes from spending more time in sleep modes where the clock is turned off, and it doesn't matter what speed the clock is set to. Assuming you need to do a constant number of clock cycles of work each time the chip wakes up, between which times you will be in sleep with the clock turned off, you're better off spending 1/20th as long awake at 20MHz, - Per section 33.5, active power consumption, with constant supply voltage, the power consumption increases in less than direct proportion to the clock speed (active mode 3.2 mA at 5 MHz, 5.5 mA at 10 MHz and 10.2mA at 20. Doing the same amount of work at 1 MHz will take more power than doing it much faster at 20 MHz and spending the rest of the time sleeping 

Running at lower clock speeds does matter if you need it to stay in active mode for a duration measured in units of time, rather than clock cycles of work (ex "wake up and monitor a pin for 10 seconds before returning to sleep")

Uwne, 2022-02-15:
Yes this is exactly what I want to do. The next step is to implement sleep mode, only wake up by PCINT,
do work and go back to sleep. I also need to update a alive LED from time to time.

closed 2022-02-24 by SpenceKonde

SpenceKonde, 2022-02-24:
Fix for this is checked in here, porting to DxCopr as bug was also present on DxCore.

commit c4886d8f (2022-02-24): Fix #650  - timing problems at 1 MHz and 2 MHz

Implement the behavir so it works as intended, ie, to make TCB exremely clock efficient as  millis source,  particularly at very low clock speeds

It looks like throughout the core I'd had different ideas f what this shuld do at different times.
I think this is what it ought to be though,,,,

All TCBn millis timers wiill use the /2 clock source *except* for those running at 2 MHz or less. to F_CPU/2000

Except for the two low speeds. 1 MHz counts with unprescaled clock, so 1000 of them are a millisecond. No math is needed to convert units into us [... cut]