SpenceKonde/megaTinyCore #655: analogWriteFast using the fast digital io functions

timpur, 2022-02-18:
is it possible to also have an analogWriteFast to use the fast io functions to save space ?

SpenceKonde, 2022-02-18:
Well, the problem is that the fast digital IO functions require that the pin be known at compiletime. 

A call to digitalWriteFast is set to always be inlined, and since that compiles down to a single instruction, it's very space efficient. 

How much could analogWrite be collapsed down to with the pin being known? Probably less than you would hope (and remember that digitalWriteFast() doesn't do anything outwardly visible if there's currently PWM on the pin, only the real digitalWrite() calls turnOffPWM() (which accounts for a large fraction of the bloat that digitalWrite pulls in. 

With a variable duty cycle and constant pin... te best ou could do is what... SBI for pin direction,lds, andi, sts for turning off the pullup, (the digitalWriteFast) then you need to figure out if duty cycle is 255 or zero... it's in a uint8_t so you'd copy to a new variable and subtract 1 from the original one (this would tend to produce better code than the alternative I think, test if it's >= 254, and in that case. Then LDS, ANDI STS to clear the relevant bit of the timer compare register then test if the original is equal to 0 to choose between writing the pin high or low. otherwise, it's an lds, ori, sts to turn on the channel and another sts to write the new duty cycle. So the best possible analogWriteFast, assuming the compile does nothing too stupid might be 1+5 + 4 (test if 0 or 1) +  5 (digital-write-with-variable-value) + 5 (clear tehe timer bit) + 1 return ; that covers the codepath taken when it's 0 or 255, the rest of would be 7 (5 for turning on the channel, and 2 more for setting the duty cyle), then 1 more for return. 27 words of 54 bytes per call with an execution time in the neighbothood of 50 clocks (only one branch runs depending on duty cycle, but LDS is 3  clocks), and that would have to get inlined every time you called it.. 

And that is all assuming it's driven by TCA. If it was a 1-series part's TCD pin, it would be significantly bulkier. 

So I'm not really sure that anything that one could do for an analogWriteFast would save as much as you hope it would, 

What chip is this? One thing I've been thinking about it adding more menus for flash-saving options, most obviously one that would disable TCD PWM on 20 and 24 pin tinyAVRs where it is used, since that definitely bloats analogWrite and digitalWrite (though turnOffPWM)

And if it has no pwm controlled by YCD0, you can save some flash right now by switching to TCA0 or TCB0 for millist timekeeping, or save even more disabling millis entirely (you still get delay and delayMicroseconds, though the're implementd as busy-waits, so unlike classic delay,() if an interrupt fires during the delay, the countdown stops (with millis enabled it does not)

Yeah I just don't see how an analogWritEFast is going to get you much. I've toyed with the idea of a function that doesn't turn the PWM on or off, just gets inlined and sets the duty cycle to what you specified. - but that's a speed opotimization, it won't save much if any space unless you menially enabled and disabled PWM when needed. And at that point, you might as well implement your own function to change the duty cycles too.

timpur, 2022-02-19:
Thanks for the quick reply

Im using an AtTiny402.... though im not a pro at writing hardware drivers ive just been playing around.

Im sure you know better, just a question i had to ask :P Thanks

closed 2022-02-19 by timpur