SpenceKonde/megaTinyCore #656: The "NVM_STATUS 'was not declared in this scope" error is still present when compiling on Linux.

michpro, 2022-02-18:
When using Arduino on Linux, the error "'NVM_STATUS' was not declared in this scope" still occurs when trying to compile a program that uses macro "eeprom_busy_wait()" from <avr/eeprom.h>. The problem does not occur with the same file compiled under Windows. Both have "megaTinyCore" version 2.5.10 installed.

SpenceKonde, 2022-02-18:
Is that a manual installation megaTinyCore there, and board manager on windows? You need either a board manager installation or manual compiler toolchain update to not receive that error.

michpro, 2022-02-18:
On both systems megaTinyCore was installed by the board manager. In both cases Arduino itself is installed in the "portable" version from the unpacked archive.

SpenceKonde, 2022-02-18:
Please enable verbose compilation in preferences and get me the full output. If you can't get it to attach to github, put it into a tarball and email it to spencekonde@gmail.com.

SpenceKonde, 2022-02-18:
There is something different about your configuration, as we have enough linux users that it is implausible that none of them use eeprom.

SpenceKonde, 2022-02-18:
That's expected. 2.5.9 was released in a panic, and I forgot to bump the version number, not related to the  issue at hand.

SpenceKonde, 2022-02-18:
You're saying that compiles for windows? I get the same sort of error. 

But t<avr/eeprom.h> is not within my sphere of responsibility to maintain. That's microchip's job, and they aren't particularly diligent about it. 

That's why the EEPROM.h library included with the core (providing the usual Arduino EEPROM.h interface) doesn't actually use avr/eeprom.h, I'd forgotten we still have zero usage of t symbols from avr/eeprom,h....  (I'd thought it was fixed, but I must have been mistaken, it's still broken in all versions of the toolchain I have around.)

michpro, 2022-02-18:
Yes, the code builds without problems on Windows (on two different Win10 and Win11 machines). I know <avr/eeprom.h> is not yours, but I find it strange that it works on Windows and not on Linux: /. I need the functions from <avr/eeprom.h> for my bootloader and I just need to do a workaround to redefine the macro "eeprom_busy_wait ()"

michpro, 2022-02-18:
I think I found the source of the problem, ... on windows in the eeprom.h file I have:
```
#if defined (__DOXYGEN__)
# define eeprom_is_ready ()
#elif defined (NVM_STATUS)
# define eeprom_is_ready () bit_is_clear (NVM_STATUS, NVM_NVMBUSY_bp)
#elif defined (NVMCTRL_STATUS)
# define eeprom_is_ready () bit_is_clear (NVMCTRL_STATUS, NVMCTRL_EEBUSY_bp)
#elif defined (DEECR)
# define eeprom_is_ready () bit_is_clear (DEECR, BSY)
#elif defined (EEPE)
# define eeprom_is_ready () bit_is_clear (EECR, EEPE)
#else
# define eeprom_is_ready () bit_is_clear (EECR, EEWE)
#endif
```

and that's ok, and on linux it's:

```
#if defined (__DOXYGEN__)
# define eeprom_is_ready ()
#elif defined (__AVR_XMEGA__) && __AVR_XMEGA__
# define eeprom_is_ready () bit_is_clear (NVM_STATUS, NVM_NVMBUSY_bp)
#elif defined (DEECR)
# define eeprom_is_ready () bit_is_clear (DEECR, BSY)
#elif defined (EEPE)
# define eeprom_is_ready () bit_is_clear (EECR, EEPE)
#else
# define eeprom_is_ready () bit_is_clear (EECR, EEWE)
#endif
```

old version causing the problem.

Do you prepare packages separately for each system?

SpenceKonde, 2022-02-18:
Yeah, I have to prepare them separately for windows vs everything else - otherwise the line endings are fucked and nothing works. 

The method used to prepare them is horrifying since I most certainly do not know how to set up a build environment that can cross-compile the toolchain for or what, 7 platforms? the Azduino packages are frankenstein'ed together from the last Arduino package and the last 

A build is run on a linux system in the cloud specifying the latest ATmega, ATtiny, and Dx-series ATpacks, on a rather large linux system (to get the build time down to something sane - it's an EC2 compute instance with like 8 threads or something, would cost a fortune to keep deployed). That generates an x64 linux version and all the pre-compiled shit. That is downloaded and then the cloud system is shut down. 
That gets extracted to windows, alongside the last Arduino toolchain. All the includes and precompiled AVR .a and .o files are copied onto the Arduino package, replacing what was there (but leaving the last arduino package's working binary tools in place) 
The same archives (newly build x64 linux, and all the arduino7 packages for non-windows platforms) are copied to WSL ubuntu. 
A series of commands is copy+pasted from text files to copy the files and and repack them . 
The wndows archive is copied to the same location in wsl, and another command is copied and pasted from a text file, and the output passed through a regex also copy-pasted from a text file, to transform it into the information that is needed for the board manager json,  except for the URL for x64 linux being wrong, because 95+% of the traffic for that is from github actions CI. The other platforms are uploaded to Amazon S3 manually. The x64 linux version is uploaded to the github repo so that I don't go broke and so github doesn't waste tons of bandwith to outside their network. Finally, the x64 linux version is uploaded to github, the final URL added to board manager json, and the board manager json file updated. 

You can see why I only do this when there's a compelling update. It's been quite a while (I think the last time was to pick up the 322x support, and the next will likely be when the AVR64DD28 and AVR64DD32 parts show up on Microchip Direct as available to preorder.

michpro, 2022-02-18:
But the most important thing is that you already know what and how to fix it :)

SpenceKonde, 2022-02-18:
Yeah - making sure it's fixed in Azduino5 won't be a problem.