SpenceKonde/megaTinyCore #658: DIRSET and DIRCLR Order impacts sleep power consumption.

geoffgscott, 2022-02-19:
This is a weird one and I might be missing something obvious but from the datasheet it looks like neither DIRSET nor DIRCLR should impact any other pin. However I am see 5uA more power consumption during sleep when I set the output first and then the input. This may not be library related but I wanted to note it for people looking a minimizing power consumption. 

```c++
#define SET_REG(pin, reg) (reg = reg | (1 << pin))
```

```c++
// Sleep power consumption is 5uA higher
SET_REG(2, PORTB.DIRSET);  // Output
SET_REG(3, PORTB.DIRCLR);  // Input

// Enable pullup and change interrupt
PORTB.PIN3CTRL = 0b00001001;
```
```c++
// Sleep power consumption is normal
SET_REG(3, PORTB.DIRCLR);  // Input
SET_REG(2, PORTB.DIRSET);  // Output

// Enable pullup and change interrupt
PORTB.PIN3CTRL = 0b00001001;
```

Following the above code blocks I am setting all the other pins to outputs.
```c++
 // Set all pins to output to reduce sleep
  SET_REG(1, PORTB.DIRSET);  // Output
  SET_REG(0, PORTB.DIRSET);  // Output

  SET_REG(0, PORTA.DIRSET);  // Output
  SET_REG(1, PORTA.DIRSET);  // Output
  SET_REG(2, PORTA.DIRSET);  // Output
  SET_REG(3, PORTA.DIRSET);  // Output
  SET_REG(4, PORTA.DIRSET);  // Output
  SET_REG(5, PORTA.DIRSET);  // Output
  SET_REG(6, PORTA.DIRSET);  // Output
  SET_REG(7, PORTA.DIRSET);  // Output

  // Disable ADC for minimum sleep current
  ADC0.CTRLA &= ~ADC_ENABLE_bm;
```

SpenceKonde, 2022-02-20:
This is expected behavior because your SET_REG macro manipulates the registers inappropriately for the registers in question. 

In fact, your whole scheme for interacting with the port registers there appears to be built upon incorrect understanding of how those registers work. 

The PORTx.DIR/OUT____ registers will perform the specified operation on the bits that are set 1 when written to. 

When READ FROM, they will always read back the **value of that register**

So 

PORTA.DIRSET = PORTA.DIRSET | (something) 

will do the same thing as:

PORTA.DIRSET = (something)

(but it will take an additional four clocks of additional execution time and 6 bytes of flash as it unnecessarily reads the register before writing it. And it's also non-atomic, so an interrupt that changed the port direction register and unset any bit would get those bits re-set. 

But 

PORTA.DIRCLR = PORTA.DIRCLR | (sometthing)

will write the current value of PORTA.DIR | (something) to PORTA.DIRCLR. That is, it will clear the direction bit for EVERY BIT THAT IS CURRENTLY SET, and the (something) doesn't matter, because either that bit was already set (and so will be cleared per above logic), or it wasn't set, in which case it doesn't matter whether you clear it or not. In addition to being non-atomic and hence not interrupt safe. 

The PORTx.__SET/CLR/TGL registers should **only** ever be used with simple assignment of a value. That value must not be a the result of bitwise operations involving the value of register. Otherwise you defeat the entire purpose of those registers and get unexpected and unwanted behavior. 

The purpose of those registers is twofold - and your code misuses them in a way that defeats both of them, in addition to mishandling DIRCLR such that you clear all currently set bits: :
1. They allow atomic setting and clearing of port registers when the register and/or bit is not known at compile-time. By reading the value of the register, you make it non-atomic.
2. They permit atomic setting and clearing of more than one bit at a time, with a single write. By doing it one bit at a time you avoid this benefit. 

Your chain of writes to PORTA.DIRSET results in 8 sequences of the following being generated (this is not valid assembly, but rather a rendition more readable to people who don't know AVR assembly, but it maps directly to it)
```
LDS reg, PORTA.OUTSET
ORI reg, value; // where value is 1<< n; SET_REG is a macro, and pin is constant and known at compiletime, and constant folding occurs.
STS PORTA.OUTSET, reg;
```
Each one of those takes 10 bytes of flash (LDS and STS are 2 word instructions) and takes 6 clocks to execute. 
The compiler cannot combine them, because it cannot optimize away access to volatile variables, which the peripheral special function registers are. 

Hence, your configuration of for PORTA alone uses 80 bytes of flash and takes 48 clocks to execute. 
Better ways to do the same thing include:
```
PORTA.DIRSET = 0xFF;// This compiles to LDI, STS. 6 bytes, 3 clocks. 
PORTA.DIR = 0xFF; // You are setting every bit to a known value. No need to use a SET/CLR register. 6 bytes 3 clocks
VPORTA.DIR = 0xFF; // Since the port is known at compile time, slightly higher efficiency can be achieved by using the VPORT. 
// That results in LDI, OUT - 4 bytes, 2 clocks. 
```

So your code could be simplified and corrected by:
1. Remove the BIT_SET macro. Never use any macro like that ever again - that sort of macro promotes inefficient and bad code. That is `cbi()` / `sbi()` syndrome. Analogous macros like the ones you have. named sbi() and cbi() existed on ATTinyCore, and misuse of them - either using more than 2 in succession when assignment would have been more appropriate, or use on registers to which they were not applicable* resulted hundreds of bytes of overhead on parts that could not spare the flash, some of which was only recently recognized and corrected. To set a bit in a register, just use the `REGISTER |= (1 << n)` or `REGISTER &= ~(1 << n)  
2. Since, at compiletime, you know which registers you want to set (the value does not need to be known at compiletime for straight assignment to a VPORT), you can just do this: 
```
VPORTB.DIR = 0b00000111; // set PB0, PB1, PB2 output, PB3 input. The 4 high bits are "don't care" assuming you're on a 14-pin part, since those pins don't exist. 
VPORTA.DIR = 0b11111111; // Set all pins on PORTA to outputs.
ADC0.CTRLA &= ~ADC_ENABLE_bm;
````

The VPORTs are cut-down mirrors of the PORTs, having only 4 registers each - but they reside in the low I/O space. When writing a value with only a single bit set to them with |= or a value with all but one bit set with &=, they compile to the fantastically efficient SBI or CBI instruction, which atomically flips the specified by in a single clock cycle.
When writing a value to them with simple assignment (`=`), the code is 2 bytes smaller and runs 1 clock faster than the corresponding PORT registers, as long as the register is constant and compile-time known. 

Note:  Within methods of a classs member (ie, class member functions) the properties (ie member variables) class member, even if  set in the constructor and declared const, are not "compiletime known" - the optimizer cannot make any assumptions about them, even if there is only ever a single member of that class instantiated.  

[... cut ...] low I/O. Most others did not. Most have the USI registers in low I/O space. Some do not. The classic AVRs only had 3 GPIORs, but only GPIO0 was always located in low I/O - the others usually were, but sometimes were stuck in the high I/O memory, being readable or writable in a single clock, but without bit-level access. But much of the code in ATTinyCore when I forked it was written with heavy use of sbi/cbi macros that were assumed to be compiling to, well, sbi and cbi, but often they were not.

closed 2022-02-20 by SpenceKonde

geoffgscott, 2022-02-20:
This is made even more embarrassing because I remember going through those registers in the datasheet and thinking how nice it is to no longer have to unset bits like on the old series. 😳 

I _really_ appreciate the detailed response, it makes the register strategy on these devices so much clearer. Also a really good point on the macros; It was a strategy I used frequently on the ATTiny85s but I had not though through the compiler downsides. 

Cheers!