SpenceKonde/megaTinyCore #659: How to use/prepare ATtiny1627 Curiosity Nano with megaTinyCore 2.5.10?

pcfreak1201, 2022-02-20:
I would like to use my new Curiosity-Nano-Kit (DM080104) with Arduino 1.8.19 (on my Linux).
Virtual drive occurs and ttyACM is available, but I can't do burn bootloader or upload code with or without internal programmer nEDBG.
I've unlocked the chip with Microchip Studio, but how to go on?

[image]

SpenceKonde, 2022-02-20:
What is the error message when you attempt to "burn bootloader" via nEDBG? 

Upload using Programmer is not supported for Optiboot board definitions (the compile process generates a binary that will only work if there is a bootloader - this is inherrant to how the bootloader support works on these parts - binaries compiled for use with a bootloader cannot be used without one and ones compiled for use without a bootloader cannot be used with one.

and if you can't burn bootloader, it's no surprise that you can't upload sketches (either through the bootloader which you aren't able to install, or for a non-optiboot board definition through nEDBG)

pcfreak1201, 2022-02-21:
```
/home/stefan/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18/bin/avrdude -C/home/stefan/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/avrdude.conf -v -pattiny1627 -ccuriosity_updi -Pusb -e -Ufuse0:w:0x00:m -Ufuse1:w:0x00:m -Ufuse2:w:0x02:m -Ufuse5:w:0b1111{bootloader.resetpinbits}10:m -Ufuse6:w:0x07:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m -Uflash:w:/home/stefan/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/bootloaders/hex/optiboot_Curiosity1627.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/stefan/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/avrdude.conf"
         User configuration file is "/home/stefan/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : curiosity_updi
avrdude: usbdev_open(): Found nEDBG CMSIS-DAP, serno: MCHP3258...
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: usbdev_send(): wrote -5 out of 912 bytes, err = Input/output error
avrdude: jtag3_edbg_prepare(): failed to send command to serial port

avrdude done.  Thank you.

Fehler beim Brennen des Bootloaders. (Error while burning bootloader)
```

SpenceKonde, 2022-02-21:
This is a known issue with AVRdude on linux. It is not within my power to correct, AVRdude needs to be built with corrected support for the Microchip USB programming tools on linux, and then a package containing fixed binaries released, and then I can set the core to use it,

closed 2022-02-21 by SpenceKonde

pcfreak1201, 2022-02-22:
OK, looks like Hans (@MCUDude) had already a "working" ATtiny1627 Curiosity Nano. Hoping for a new version of avrdude :-)