SpenceKonde/megaTinyCore #668: ADC high power consumption in POWER DOWN

Newbie414, 2022-03-21:
Hi,

I'm working on a coin battery-operated project. I compiled every possible low-power example I found. And in POWER DOWN mode I've measured on DMM ~140uA@5V/3V3.

So I hook up the power rail to the oscilloscope to verify the MCU sleep cycle. And consumption cycles changes were as expected but in sleep, consumption was ~140uA@5V. 

Long story short ADC0.CTRLA&=~ADC_ENABLE_bm fixed the problem. And power consumption is 0.1uA@5V@27°C.

So maybe this could be addressed in DOCS or if its not expected behavior fixed.

MCU: attiny3224
BOD: DISABLED
CLK: INTERNAL 1MHz

```c
#include <avr/sleep.h>

void setup() {
  pinMode(0, OUTPUT);
  pinMode(1, OUTPUT);
  pinMode(2, OUTPUT); 
  pinMode(3, OUTPUT); 
  pinMode(4, OUTPUT); 
  pinMode(5, OUTPUT); 
  pinMode(6, OUTPUT); 
  pinMode(7, OUTPUT); 
  pinMode(8, OUTPUT); 
  pinMode(9, OUTPUT); 
  pinMode(10, OUTPUT); 

  digitalWrite(0,0);
  digitalWrite(1,0);
  digitalWrite(2,0);
  digitalWrite(3,0);
  digitalWrite(4,0);
  digitalWrite(5,0);
  digitalWrite(6,0);
  digitalWrite(7,0);
  digitalWrite(8,0);
  digitalWrite(9,0);
  digitalWrite(10,0);

  ADC0.CTRLA&=~ADC_ENABLE_bm;

  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
  sleep_cpu();

}

void loop () {
  sleep_cpu();
}
```

SpenceKonde, 2022-03-21:
I thought this was covered somewhere in the documentation.... apparently not clearly enough.....

Newbie414, 2022-03-21:
[PowerSave.md](https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/PowerSave.md) mentions clock, BOD, SLEEP, and floating pins but nothing about ADC. I even suspect this might be bug because some examples on net claim sub uA consumption without explicit disabling ADC. Maybe in older core disabling ADC was automatic.

SpenceKonde, 2022-03-21:
It's only the case with the 2-series parts (and the classic AVRs) but is not the case for other modern AVRs. it may be dependent on the setting of the LOWLAT bit. 

The 2-series post-dates the writing of that powersave documentation so it is not surprising that it is not discussed in that document. The legacy documents, which are planned to be replaced with ones with names like Ref_topic.md) should not be treated as authoritative. \
That one, specifically, cannot be rewritten until I better understand sleep and power save, and that won't happen until I have used it in the course of writing the sleep and powermanagement libraries

That is the third priority core development  (after releasing the CRITICAL BUGFIX 2.5.11 and 1.4.10 releases of this and DxCore (which I promised to clients by the end of lastweek and still isn't done), and then the 2.0.0 version o ATTinyCore which they have been patiently waiting 18 months for and which is nearly done.

dapeake2, 2022-03-23:
I am experiencing a similar problem with the Attiny826. I ran Newbie414's code and I did not see a reduction in current. My circuit is basic: a Attiny826, an 18650 battery, .47uf cap VCC~GND, and a Keithley picoammeter. Any pointers would be greatly appreciated.

Thanks,

Dan
San Jose

dapeake2, 2022-03-23:
I tried setting  LOWLAT to 0 `ADC0.CTRLA&=~0b100001`;  , and current consumption did not change.

SpenceKonde, 2022-03-23:
The fact that you people here see what is claimed to be the same bug yet starkly different behavior here is disturbing,

dapeake2, 2022-03-24:
I believe I was chasing more than one issue. I got the current down to 150na on a second carrier board. 

Newbie414 sets the pins to outputs and then pulls them low, where I was setting them to INPUT_PULLUP. When I followed Newbie414, (proper pin statements and ADC disabling) I observed 150nA 

When I said his code did not reduce current consumption on my CPU I had overlooked the 7 additional pins that need to be quieted on the 826. apologies to all.

I am also seeing identical carrier boards exhibit different current behaviors: running the same code - Board 1, I got down to 24uA, where I achieved 150nA on Board 2 (then I blew it up somehow - programs fine, but draws 75mA now.) I suspect my original problem with carrier 1 is overheating during soldering or perhaps ESD. 

Dan

dapeake2, 2022-03-24:
BTW: I created a 3rd carrier and A/B tested the following ADC0.CTRLA statements: current draw, 128nA, is the same for both statements.
```
ADC0.CTRLA&=~ADC_ENABLE_bm;

ADC0.CTRLA&=~0b100001;
```

Newbie414, 2022-03-24:
@dapeake2 maybe you should check leaking current on CAP due to overheating also every pin pull is connected on silicon to the power rail. So pullup on RESET/UPDI can source a few uA to VDD and lower measured current if measuring directly on VDD. Also parasitic conductivity from dust, moisture can sink some current. I believe in my application consumption is the same for INPUT_PULLUP. As I use INPUT_PULLUP on every pin except 0, 1 which are drivers for 2N7002 FETs. I measured the 6uA average in my application with cyclic wakeup. But I could recheck that.

dapeake2, 2022-03-24:
Thanks for the input!  I had actually removed the cap - which never overheated because it was an electrolytic "can" put on after the SMT soldering. The current readings did not differ after removal.

_"So pullup on RESET/UPDI can source a few uA to VDD and lower measured current if measuring directly on VDD."_

I want to make sure I understand. Are you saying that to achieve optimal current reduction the RESET/UPDI needs to be pulled low? And, if I use INPUT_PULLUP on RESET/UPDI it will result in a small uA load? The datasheet indicates the RESET/UPDI line will draw 100uA when pulled low. Note: my IDE burn settings are configured for "UPDI (no reset pin)".

_"As I use INPUT_PULLUP on every pin except 0, 1"_

The test code you posted above has all OUTPUT / SINK. I am assuming your project is INPUT_PULLUP. Are you saying you have observed no difference? I was going to A/B test that today as soon as I build another carrier board.

Thanks,

Dan

SpenceKonde, 2022-03-24:
I think he's refering to the potential for an external pullup on the RESET/UPDI to pass a small amount of current backwards, lowering the apparent load, depending on how the measurement is made? 

I would imagine one would measure the current being drawn in total by the board with the chip and capacitor(s) and any pullup on reset in place, specifically soas not include those factors, rather than by lifting the Vdd pin and measuring current directly on that or something? 

I'm not sure what the magnitude of surface contamination effects are where the surface contaminant is typical no-clean flux, how much of a difference various cleaning methods make; I suppose it probably depends on the brand and type of solder paste or solder/flux. I can smell a HUGE difference between what is in various brands of no-clean solder paste. I have no idea what the volatiles are that I associate with PPD S600-intermediate temperature lead free vs Relife RL401 leaded solder vs Chipquik. but they are each a distinct nasty smell....

dapeake2, 2022-03-25:
" lifting the Vdd pin and measuring current directly on that or something?"

Makes sense, as that is the equivalent of what I am doing ( isolating everything to get baselines ). 

"Chipquik. but they are each a distinct nasty smell...."

Yes, headache inducing. Chipquick residue looks like trouble but has never caused issues, unless over-applied. Different topic, but properly soldering the VQFN20s is a chore if you buy cheap carriers - nothing auto-aligns!

Thanks Spence!

SpenceKonde, 2022-05-01:
Closiong This is at it'[s core a duplicate of #158

closed 2022-05-01 by SpenceKonde