SpenceKonde/megaTinyCore #671: Logic library does not permit manually defined CCL interrupts

SpenceKonde, 2022-03-22:
All forms of attachInterrupt (whereby a function pointer is provided to the function that will be called by the ISR are inherrently evil. The calling of functions from within an ISR is specifically cautioned against.

The CCL interrupt should not be compiled in and included in the binary if not needed. 

Note that the simple existence of such an ISR even when not used, without appropriate countermeasures, also prevents users from using the library if they require an efficient and well written ISR (which is impossible if it is 'attached' via an attachInterrupt() like functions), as they will encounter a duplicate vector error. attachInterrupt should be used only by novices for simple tasks, or when it is absolutely necessary to arbitrarily change what the interrupt does at runtime. We must ensure that innocent skilled users are not prohibited from using this library

SpenceKonde, 2022-05-01:
Path forward is simple, just need to do it,.

MCUdude, 2022-06-18:
I've come up with a solution that optimizes away the ISR when not in use, thus saving ~200 bytes for every project that includes Logic.h.

[Here's the commit](https://github.com/MCUdude/MegaCoreX/commit/41b09f2106ba4d554a8e9ca0d8cabe9e4c0c40fb). The trick is to move attachInterrupt, detachInterrupt, and the ISR itself into a dedicated file. When `dot_a_linkage=true` is added to library.properties, the compiled .a file is optimized away. You should probably apply the same fix to the Comparator and ZCD libraries as well.

But why not tag and discuss with me before adding bulky, ad-hoc C code that doesn't fit very well in the first place? 
It is very much a valid point that ISRs shouldn't be pulled into the compiled binary when not in use, and to be honest, I didn't realize this before I stumbled across this issue. I'm happy to help out with future library-related issues, but you can't expect me to show up if I don't get mentioned and pulled into the discussion.

SpenceKonde, 2022-06-19:
Argh, I thought I had @mentioned you.

I'm surprised .a linkage needs to be true. Doesn't that mean we need compile a .a file for every supported part? I thought that's what that option did, and my ugly hack didn't need 

I though it would work as long as no function i the file with the ISR was referenced. (and the reason for C code was i didnt realize class methods didn't all need to be in one file - see I still don't know C++ worth shit. I prefer c with inline assembler on arm over C++ any day.

SpenceKonde, 2022-07-23:
Changes checked into Logic and Comparator.

closed 2022-07-23 by SpenceKonde