SpenceKonde/megaTinyCore #678: TypeError: ld16() when uploading program using SerialUPDI

technoblogy, 2022-03-30:
My program gives this error from SerialUPDI when uploading under megaTinyCore 2.5.10 to an ATtiny402:

    TypeError: ld16() takes 2 positional arguments but 3 were given

Any idea what could be causing this? The error disappears if I make trivial changes to the program. I'm struggling to create an MCVE, but will persevere if it would be useful.

technoblogy, 2022-04-01:
I'm closing this issue because I tried the exact same thing this morning and I'm not getting the bug. I'll reopen if I can demonstrate it consistently.

technoblogy, 2022-04-01:
The bug occurs during the Verify phase, and occurs with the Clock option set to "10MHz internal" but does not occur with the Clock option set to "20MHz internal".

Here's the SerialUPDI output:

````
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.3 - Jan 2022
Using serial port /dev/cu.usbserial-A104OVGT at 57600 baud.
Target: attiny402
Set fuses: ['2:0x02', '6:0x04', '8:0x00']
Action: write
File: /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_842724/TFTGraphicsLibrary3.ino.hex
Pinging device...
Ping response: 1E9227
Setting fuse 0x2=0x2
Writing literal values...
Verifying literal values...
Action took 0.19s
Setting fuse 0x6=0x4
Writing literal values...
Verifying literal values...
Action took 0.19s
Setting fuse 0x8=0x0
Writing literal values...
Verifying literal values...
Action took 0.19s
Finished writing fuses.
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.06s
Writing from hex file...
Writing flash...
[==================================================] 57/57
Action took 4.46s
Verifying...
[=====================================             ] 6/8Traceback (most recent call last):
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/prog.py", line 285, in <module>
    main()
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/prog.py", line 128, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/prog.py", line 259, in pymcuprog_basic
    run_pymcu_action(pymcu._action_verify, backend,
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/prog.py", line 138, in run_pymcu_action
    status = func(backend, *args, args_pymcu)
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/pymcuprog/pymcuprog_main.py", line 267, in _action_verify
    verify_status = verify_flash_from_hex(args.filename, backend, max_read_chunk=args.max_read_chunk)
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/pymcuprog/hexfileutils.py", line 120, in verify_flash_from_hex
    verify_status = backend.verify_memory(segment_data, 'flash', segments[i][0], max_read_chunk=max_read_chunk)
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/pymcuprog/backend.py", line 545, in verify_memory
    return self.programmer.verify_memory(data=data, memory_name=memory_name, offset=offset_byte, max_read_chunk=max_read_chunk)
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/pymcuprog/programmer.py", line 208, in verify_memory
    data_verify = self.read_memory(memory_name, offset, len(data), max_read_chunk=max_read_chunk)[0].data
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/pymcuprog/programmer.py", line 290, in read_memory
    data = self.device_model.read(meminfo, offset_adjusted, numbytes_adjusted, max_read_chunk=max_read_chunk)
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/pymcuprog/nvmserialupdi.py", line 219, in read
    data += self.avr.read_data_words(offset, read_chunk_size>> 1)
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/pymcuprog/serialupdi/application.py", line 147, in read_data_words
    return self.readwrite.read_data_words(address, words)
  File "/Users/david/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/pymcuprog/serialupdi/readwrite.py", line 102, in read_data_words
    return self.datalink.ld16(self,address)
TypeError: ld16() takes 2 positional arguments but 3 were given
TypeError: ld16() takes 2 positional arguments but 3 were given

[===========================================       ] 7/8
````

SpenceKonde, 2022-04-01:
This is already fixed in 2.5.11.

SpenceKonde, 2022-04-01:
You can see that 2.5.11 has changed line 102 from ld16(self,address) to ld16(address). 

2.5.10 has several other critical defects and should not be used.

closed 2022-04-01 by SpenceKonde

technoblogy, 2022-04-01:
I confirm that it's fixed in 2.5.11. Thanks!

SpenceKonde, 2022-10-11:
I swear I fixed a bug just like this. Did it get reverted somehow? It was
dependant on the size of the code it wad trying to upload

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

technoblogy, 2022-10-12:
@SpenceKonde Not sure why you commented yesterday; this was relating to a problem I reported as fixed in April.