SpenceKonde/megaTinyCore #680: DxCore tools used when compiling for ATtiny1624/6

HeadBoffin, 2022-04-05:
Just a sense check here - as I can get ATtiny1624's & 6's (thank you Mouser & thank you for limiting purchases so we can all get some) I've a OneWire & I2C based project that **only** compiles for Series 2 if the DxCore is installed as well as this (wonderful) BSP.

Is this expected / correct / documented?

If this is unexpected, I'll capture a failed compile and post it here.

SpenceKonde, 2022-04-07:
Please capture tre failed compile!!!!!!!!!

SpenceKonde, 2022-04-07:
It is incorrect becavior, totally unexpected and this is the first I have heard aboiut it!

SpenceKonde, 2022-04-13:
Actually - was either core installled manually? Additional steps are required for manual because of the strange way that board manager vs manial installs happen.

SpenceKonde, 2022-05-01:
almost three weeks without required information. Cannot resolve without verbose compile log showing failure

closed 2022-05-01 by SpenceKonde