SpenceKonde/megaTinyCore #683: 2-Series temperature sensor

SpenceKonde, 2022-04-09:
### Discussed in https://github.com/SpenceKonde/megaTinyCore/discussions/536

<div type='discussions-op-text'>

<sup>Originally posted by **mechatroniks-git** October  9, 2021</sup>
I took a stab at reading the internal temp sensor, and it appears to work, sharing here. 

```
uint16_t readTemp() {
  volatile uint16_t adc_reading;
  uint8_t sigrow_gain = SIGROW.TEMPSENSE0;
  int8_t sigrow_offset = SIGROW.TEMPSENSE1;
  uint32_t temp;

  //ATtiny1624/1626/1627 datasheet section 30.3.3.7 Temperature Measurement
  //1. Configure the voltage reference to internal 1.024V by writing to the Reference Selection (REFSEL) bit field the ADCn.CTRLC register.
  analogReference(INTERNAL1V024);

  //2.  Select the temperature sensor as input in the Positive Input Multiplexer (ADCn.MUXPOS) register.
  ADC0.MUXPOS = ADC_MUXPOS_TEMPSENSE_gc;

  //3. Configure the ADC Sample Duration by writing a value ≥ 32 µs × CLK_ADC to the Sample Duration (SAMPDUR) bit field in the Control E (ADCn.CTRLE) register
  analogSampleDuration(50); 

  //4.  Acquire the temperature sensor output voltage by running a 12-bit Single-Ended conversion.
  adc_reading = analogRead(ADC_TEMPERATURE);

  //5.  Process the measurement result, as described in datasheet, except the offset is added, not subtracted
  temp = (uint32_t)adc_reading + sigrow_offset;
  temp *= sigrow_gain;
  temp += 0x80;
  temp >>= 8;

  //Serial.printf("K=%d\n", temp);
  //Serial.printf("C=%d\n", temp - 273);
  //Serial.printf("F=%d\n", (((temp - 273) * 9) / 5) + 32);

  return (uint16_t)(temp & 0xFFFF);
}
```

Also, the example was showing 334mV for the supply voltage, so I scale by 10 to get 3340mV, closer to actual. 

```
uint16_t readSupplyVoltage() { //returns value in millivolts to avoid floating point
  analogReference(INTERNAL1V024);
  int32_t vddmeasure = analogReadEnh(ADC_VDDDIV10, 12); // Take it at 12 bits & discard
  vddmeasure = analogReadEnh(ADC_VDDDIV10, 12); // Take it at 12 bits
  return ((vddmeasure * 5 ) >> 1 ); Multiply by 10 and divide by 4
}
```</div>

jvasileff, 2022-04-12:
I believe this was fixed in d196db6055d4989a2b.

SpenceKonde, 2022-05-01:
that only fixes half the problem. we still dont know if we subtract or add the offset

SpenceKonde, 2022-05-01:
was going to test to=day but cvouldnt find any working Serial UPDI programmers. Those things suffer from wicket bitrot

SpenceKonde, 2022-05-01:
btw tomorrrow morning I'm gonna add a shittload odf new stock to the 32k 2 series parts!!

closed 2022-05-21 by SpenceKonde

jvasileff, 2022-05-21:
Somehow our results differ.

For my chips, d3e22ece4 breaks rather than fixes the temperature measurements. I re-ran my earlier tests to be sure, with the following results from `readTempVcc.ino`:

Chip | `reading += sigrowOffset;` | `reading -= sigrowOffset;`
-- | -- | --
1626 Sample 1 | 317 K | 301 K
1626 Sample 2 | 264 K | 299 K
1626 Sample 3 | 279 K | 301 K
3226 | 331 K | 301 K

It's clear that for my chips, the datasheet's `-=` is correct.

SpenceKonde, 2022-05-22:
Interesting - I went and tested 5 more and found two things - yes, the -= seems to produce results that are consistent chip-to-chip. But it ain't anywhere near +/-5 K accuracy! Consistently reads 10-15 K higher than actual temperature o_o 

Yet yours do look like they're showing plausible temperatures, which strikes me as peculiar since mine dont. 
300 K = ~26 C = ~80F. Not an unreasonable temperature in the summer. 

Mine all read 314-318 = ~40 C or 105 F in a room that's 85F.

The chip that I tested with initially happened to have a smallish negative offset, so adding instead of subtracting flipped the sign of the offset and made it look more correct. I find it rather disturbing that we both get the same value from every chip we test, but your value is consistently accurate and mine is consistently inaccurate.

jvasileff, 2022-05-22:
> Mine all read 314-318 = ~40 C or 105 F in a room that's 85F.

My room temperature was 70 or 72F, so my readings were about 10F over room temp, shortly after power was applied, and on a plastic breadboard.

commit d3e22ece (2022-05-21): Fix #683

Empirically this was found to give  correct answers,