SpenceKonde/megaTinyCore #685: Wakeup fails to work when combined with HIGH GPIO while sleeping

ayushsharma82, 2022-04-13:
Hi,

I stumbled across this issue while making a gauge sensor with Attiny402. The Attiny402 is put into power-down sleep mode and has a pin change interrupt ( RISING ) attached to PA6 GPIO and PA3 is attached to HX711's SCK pin. ( to put HX711 into power saving mode, we have to keep the SCK pin HIGH ). 

How to replicate the issue:
1) Attach a pin change interrupt ( RISING ) to PA6
2) Keep the PA3 attached to SCK of HX711 HIGH before sleep
3) Put MCU into sleep using sleep_cpu() in power-down mode.
( --- Till here everything will work fine, i.e. TWI address match wakeup and WDT timers --- Problem occurs only when we send a pulse to PA6 interrupt )
4) Send a pulse to PA6 for wakeup
5) BAAM. MCU is unresponsive.

Any idea about what could be causing this weird issue?

**Setup:**
Board: ATtiny402
Clock: 4Mhz
VCC: 3.3v

ayushsharma82, 2022-04-13:
*Observations:* This only happens when I keep GPIO high attached to SCK of HX711. Every other scenario works fine. Totally weird.

SpenceKonde, 2022-04-14:
You're asking us to debug complicated code that we don't know anything about. 

What happens if you replace the SCK pin of the HX711 with a  a resistor that draws similar load? 

What is the schematic like? Anything wierd? You mention using I2C...

There sound like there are too many moving parts to diagnose without yourposting schematic and code.

ayushsharma82, 2022-04-14:
I understand It can be hard to understand the issue with text only, I'll provide a sample code and schematic soon.

ayushsharma82, 2022-04-14:
@SpenceKonde I was making the sample code and went through some scenarios, it looks like the problem was with my code & the library used ( [HX711](https://github.com/bogde/HX711) ). My speculation was wrong, for anyone having the same problem: It's not related to the wakeups or sleep, by default HX711 lib internally uses a while loop in its read function which locks up the microcontroller until the HX711 board is connected/working. 

Upon waking up my code was calling scale.tare() function which locked up the microcontroller and gave an expression that it was not waking up, in reality it was stuck in a while loop internally. Closing the issue.

closed 2022-04-14 by ayushsharma82

SpenceKonde, 2022-04-15:
Well that would do it = thjose HX711's are kind of flaky