SpenceKonde/megaTinyCore #686: [Attiny426] Change internal clock frequency

JorgeTorres01, 2022-04-13:
Hi, 

I need to define the PWM frequency of 1kHz. 

If I define the TCA0.SPLIT.CTRLA  with TCA_SPLIT_CLKSEL_DIV64_gc the MCU generates a signal approx. 1,22 kHz (with TCA0.SPLIT.LPER = 0xff; TCA0.SPLIT.HPER = 0xff;)

If I define TCA0.SPLIT.CTRLA  with TCA_SPLIT_CLKSEL_DIV256_gc the MCU generates a signal approx. 0,305 kHz (with TCA0.SPLIT.LPER = 0xff; TCA0.SPLIT.HPER = 0xff;). But the 1kHz achieved with TCA0.SPLIT.LPER = 0x4D; TCA0.SPLIT.HPER = 0x4D (77, DEC);

This way, I have a reduced duty cycle resolution.
Is it possible to reduce the clock frequency from 20 to 16 MHz?

Thank you,
Jorge

JorgeTorres01, 2022-04-13:
In Arduino IDE it is possible to change the CPU frequency.

How can it be changed in code?

The check with F_CPU but it does not work. 

[image]

SpenceKonde, 2022-04-14:
there's a fusebit to set 16-derived vs 20 derived It is set to match selection whenever a sketch is uploiaded over UDPI. It cannnot be set with optoboot except by rebuning boootloader

This appliies to tinyAVR 0/1/2, megaAVR 0, and as far as we know upcoming EA-series:

The compliance of the oscillator is nothing sort of astonishing. Many times larger than the spec implies: the actual dffference between min and max  clock speed (assuming the clock functioned atthe thigh end which it doesn't always at the high speed setting) See https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/Ref_Tunin.md

What are your actual requirements? What frequency How many vchannels? Whar eelse is goingfon .

Posting the full code and schematic wol help us help you

JorgeTorres01, 2022-04-14:
Thank you [SpenceKonde](https://github.com/SpenceKonde)

I have this code on Arduino IDE with an internal clock of 16 MHz. 
I change the PER to 0xF8 to achieve a frequency of 1 kHz on PWM signals. And this works correctly

```
uint8_t WO0 = 9;
uint8_t WO1 = 8;
uint8_t WO4 = 0;
uint8_t WO5 = 1;

void setup () {
  TCA0.SPLIT.LPER = 0xF8;
  TCA0.SPLIT.HPER = 0xF8;

  TCA0.SPLIT.CTRLA = ~((~TCA0.SPLIT.CTRLA) | TCA_SPLIT_CLKSEL_DIV1_gc | TCA_SPLIT_CLKSEL_DIV2_gc | TCA_SPLIT_CLKSEL_DIV4_gc | TCA_SPLIT_CLKSEL_DIV8_gc | TCA_SPLIT_CLKSEL_DIV16_gc | TCA_SPLIT_CLKSEL_DIV64_gc | TCA_SPLIT_CLKSEL_DIV256_gc | TCA_SPLIT_CLKSEL_DIV1024_gc); 
  TCA0.SPLIT.CTRLA |= TCA_SPLIT_CLKSEL_DIV64_gc;

  pinMode(WO0, OUTPUT);
  pinMode(WO1, OUTPUT);
  pinMode(WO4, OUTPUT);
  pinMode(WO5, OUTPUT);

  analogWrite(WO0, 20);
  analogWrite(WO1, 70);
  analogWrite(WO4, 140);
  analogWrite(WO5, 200);
}

void loop () {
}
```
I'm trying to use PlatformIO instead of Arduino IDE because it's easier to organize and open functions...
When I was writing this post, I saw that the PlatformIO uses a different platform (https://github.com/platformio/platform-atmelmegaavr.git)

Is it possible to use megaTinyCore on PlatformIO for attiny426?

SpenceKonde, 2022-04-15:
I don't know anything about platform IO, other than that a alot of my users are on it. There may be an incomplete document in this repo about it because I don't know any of the relevant answers

iirc you have to manually specify the #defines in your configuration file. I suspect it defaults to 20

SpenceKonde, 2022-04-15:
And this is by the way one of the most fustrating things about the widespread use of platform IO - I don't have a method to make sure that a valid combination of options is seleected

SpenceKonde, 2022-04-30:
Closing. If anyone wants to help clarify the matter in the PlatformIO.md file, I gladly merge PRs

closed 2022-04-30 by SpenceKonde