SpenceKonde/megaTinyCore #689: Attiny 3216

Konstantinnios, 2022-04-17:
Hallo,
I am trying to migrate from 328P to Attiny 3216. For now I am having difficulty with the functions for writing and reading information from an external EEPROM with SPI communication.
So far I have used:

void write_eeprom (long address, byte data2) {
  digitalWrite (SLAVESELECT, LOW);
  spi_transfer (WREN); // write enable
  digitalWrite (SLAVESELECT, HIGH);
  digitalWrite (SLAVESELECT, LOW);
  spi_transfer (WRITE); // write instruction
  spi_transfer (int) (address >> 8)); // send MSByte address first
  spi_transfer ((int) (address)); // send LSByte address
  spi_transfer (data2); // write data byte
  digitalWrite (SLAVESELECT, HIGH); // release chip
  // wait for eeprom to finish writing
  delay (5);
}
// =========================================== =============
byte read_eeprom (int EEPROM_address) {// READ EEPROM
  int data;
  digitalWrite (SLAVESELECT, LOW);
  spi_transfer (READ); // transmit read opcode
  spi_transfer (char (EEPROM_address >> 8)); // send MSByte address first
  spi_transfer (char (EEPROM_address)); // send LSByte address
  data = spi_transfer (0xFF); // get data byte
  digitalWrite (SLAVESELECT, HIGH); // release chip, signal end transfer
  return data;
}

I would be grateful if you could share read/write example/function.

Modac, 2022-04-17:
What is the spi_transfer function?

Modac, 2022-04-17:
It seems you include the SPI library but never actually use it. Instead the SPI registers are used directly. 
So you have two options:
1. Migrate to new registers, this will probably require less actual change in your program but you will have to read through the datasheet SPI section. Your program will also not be compatible with any other chip exept for the new attinys.
2. Migrate to Arduino SPI library. This is in my opinion more sensible as your program will be compatible with all chips with an Arduino SPI ligrary defined and it will probably not require much more changing. Maybe [this](https://github.com/dndubins/EEPROMsimple) repo can give you some infos how to use the SPI library for EEPROMs

Konstantinnios, 2022-04-17:
Thanks for the quick response.
I tried the code below. Works on 328P, but not on Attiny 3216. I checked the connections, pins, I / O - all OK.

#define WRITE 2
#define READ  3
#define WREN  6

#include <SPI.h>

unsigned int data;
unsigned int address = 0;
byte outval = '9';
byte invalue;

void setup()
{
  Serial.begin(9600);
  Serial.println("--------------");
  Serial.println(SS); // chip select
  Serial.println(MOSI); // master out, slave in
  Serial.println(MISO); // master in, slave out
  Serial.println(SCK); // clock
  Serial.println("--------------");
  
  // set up to match device datasheet
  SPI.setBitOrder(MSBFIRST);
  SPI.setDataMode(SPI_MODE0);
  SPI.setClockDivider(SPI_CLOCK_DIV2); // max clock is 20MHz, so can set high speed
  SPI.begin(); // sets up pin modes etc.
  
  // Enable writing
  digitalWrite(SS, LOW);
  SPI.transfer(WREN);
  digitalWrite(SS, HIGH);
  
  // Write One Value to One Address
  digitalWrite(SS, LOW);
  SPI.transfer(WRITE); // write instruction
  SPI.transfer((address >> 16) & 255);
  SPI.transfer((address >> 8) & 255);
  SPI.transfer(address & 255);
  SPI.transfer(outval);
  digitalWrite(SS, HIGH);
  delay(5);
  
  delay(1000);
  
  // Read One Value from One Address
  digitalWrite(SS, LOW); 
  SPI.transfer(READ); // read instruction
  SPI.transfer((address >> 16) & 255);
  SPI.transfer((address >> 8) & 255);
  SPI.transfer(address & 255);   // 3rd byte of address
  invalue = SPI.transfer(0);        // Clock out the data
  digitalWrite(SS, HIGH);
  
  Serial.print("Read Data = ");
  Serial.println(invalue);
  Serial.print("Read Data2 = ");
  char Test = ((char) invalue);
  Serial.println(Test);

}

void loop()
{
   
}

Serial monitor return:
18:47:56.221 -> --------------
18:47:56.221 -> 0
18:47:56.268 -> 14
18:47:56.268 -> 15
18:47:56.268 -> 16
18:47:56.268 -> --------------
18:47:57.252 -> Read Data = 0
18:47:57.252 -> Read Data2 = 

I wonder where I'm wrong ...

Modac, 2022-04-17:
Just for clarification, did you consider that the pins being output in the serial monitor are the Arduino pin numbers and not the physical pin numbers?

Konstantinnios, 2022-04-17:
[image]

SS: 0 (pin 2)
MOSI: 14 (pin 17)
MISO: 15 (pin 18)
SCK: 16 (pin 19)

However, there is no pin "0" :)

Modac, 2022-04-17:
Ok, so you hooked the EEPROM to the physical pins 17,18,19 and 2, right?

Two things I would try:
1. Set the pinMode of the SS pin to OUTPUT before the first digitalWrite to SS
2. Use another pin for SS (chip select) but still set the pinMode

Konstantinnios, 2022-04-18:
After defining pinMode (0, OUTPUT); things worked out. Thank you. I will not close the topic until I integrate the functions in the project, and if anyone has questions on the topic we can discuss them.

Modac, 2022-04-18:
Perfect, glad it worked.
I would suggest converting this issue to discussion and changing the title to something more descriptive like Attiny3216 SPI EEPROM, because this is not an issue with the core.

SpenceKonde, 2022-04-18:
On modern AVRs with the implementation of Wire we use, there is nothing special about the SS pin. So the SS pin is whatevver pin you connect and tell the code to drive low for comunication - isn't that how it works on the 328p, except there's this SS pin that must be kept output or it can swtich SPI to slave mode?

SpenceKonde, 2022-04-18:
The modern AVRs added a SS Disable bit which we always set. If i's not set then you get the old behavior of the pin being used as an input puitting SPI into slave mode.

SpenceKonde, 2022-04-19:
But if there is code in widespread use in libraries that uses SS while in master mode, we should probably 

```
#define SS NOT_A_PIN
```
so it will be passable to digital I/O functions without errors.

Konstantinnios, 2022-04-19:
As for the SS, that's right. In my project I use 2 SPI devices, SoftwareSerial and hardware UART communication. When I corrected the code for communication with EEPROM and applied it to the rest of the code, serious problems started. I was losing communication with some or all of my devices. Everything was fine after I choos "old version" from attachInterrupt version" from Arduino IDE. I guess it would be good to work on this problem and it is no coincidence that you can choose "old version, (may fix bugs)" :] I noticed that the older version of the library supported this MCU lacks this option. I know I'm writing like a layman, but I haven't delved into it. Thank you SpenceKonde. Keep up the good work.

SpenceKonde, 2022-04-19:
Oooooh, that's a totally different problem. Are you using 2.5.11? That
should have the fix for it - prior to that dependingon what happenened in
the isr it may or may not work correctly. I put a ton of effort into trying
to make attach interrupt suck less, and it has been quite a task because my
testing always passes.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

Konstantinnios, 2022-04-19:
2.5.9 I will update it with 2.5.11 and after testing I will write again.

Konstantinnios, 2022-04-20:
With version 2.5.11, interrupts work perfectly with the "On all pins, like usual" option.

SpenceKonde, 2022-04-20:
Huzzah!

Yeah, in 2.5.0 I added a new implementation of attachInterrupt to reduce flash size (by less than I'd hoped) and improve performance (also by less than I'd hoped) and make it possible to have attachInterrupt() enabled only on ports that it's used with (the best I was able to do was the manual mode, where you have to reference a certain function to enable attachInterrupt on that port). The last one was actually the main motivation, because there are libraries that use attachInterrupt(), and some of them are widely used, (SoftwareSerial, for example). If attached interrupts performed as well as a normal interrupt, so what? But they don't - they're way the hell slower. 

But the response time of attachInterrupt is garbage compared to what a manually defined interrupt could achieve. Simply because it calls a function (and since that function is supplied as a function pointer to use with icall it can't be optimized away like function calls normally can be), it has to assume that all 16 working registers the function is allowed to use for local variables are used by this function pointer. And being an ISR, it wouldn't know it has to push half the working registers onto the damned stack, and then pop them off at the end of the ISR (that's 16 clocks of pushing and 32 clocks of popping, in addition to the in push out pop of the SREG), even if the attached function itself doesn't need all those registers, and that's without even considering the logic to figure out which function should be called, which is another 50+ clock cycles. 

And each interrupt can exist in only one place. So either you couldn't use attachInterrupt() anywhere, or your interrupts would all be slow because they'd be forced to go through the attach handler, which would be duplicated with the change of only one instruction word for each port (up to 3x on tinyAVR, up to 7x on DxCore). 

On classic AVRs there are two kinds of pin interrupts - INTn-like and PCINT-like interrupts, and attachInterrupt only works on the first kind, leaving the PCINT handlers undefined and generally available for the application, so while attached interrupts are slow there too, nobody used to notice or complain. But here it became very confining, because people often wake from sleep with a manually defined pin interrupt (which might be just the RETI or SBI, RET) or expect response times comparable to what the datasheet describes.

The optimizer doesn't (because it's not smart enough) or can't (because the rules of the language formally prohibit such optimizations) do anything to help with this.

closed 2022-04-20 by SpenceKonde