SpenceKonde/megaTinyCore #690: selecting 1mhz clock broken after tuning?

ObviousInRetrospect, 2022-04-17:
I hit a pair of tiny1614s that didn't have a UART acceptable 16mhz osc. Which broke the 1mhz clock. So I tuned it.  I now hit the following when attempting to compile my real sketch with a 1mhz clock.  The 4mhz clock compiles (and was fixed by the tuning).

```
Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/wiring.c: In function 'tune_internal':
Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/wiring.c:1371:9: error: 'TUNED_CALIBRATION_OFFSET' undeclared (first use in this function)
     if (TUNED_CALIBRATION_OFFSET == -1) {
         ^~~~~~~~~~~~~~~~~~~~~~~~
Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/wiring.c:1371:9: note: each undeclared identifier is reported only once for each function it appears in
In file included from Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/avr/include/avr/io.h:784:0,
                 from Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/wiring_private.h:26,
                 from Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/wiring.c:23:
Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/tune_guesses.h:95:55: error: 'CLKCTRL_PDIV_20X_gc' undeclared (first use in this function); did you mean 'CLKCTRL_PDIV_24X_gc'?
     #define TUNE_PRESCALE (CLKCTRL_PEN_bm | (osccfg ? CLKCTRL_PDIV_20X_gc : CLKCTRL_PDIV_16X_gc)
                                                       ^
Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/wiring.c:1398:41: note: in expansion of macro 'TUNE_PRESCALE'
     _PROTECTED_WRITE(CLKCTRL_MCLKCTRLB, TUNE_PRESCALE);
                                         ^~~~~~~~~~~~~
Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/wiring.c:1398:55: error: expected ':' or ')' before ';' token
     _PROTECTED_WRITE(CLKCTRL_MCLKCTRLB, TUNE_PRESCALE);
```

ObviousInRetrospect, 2022-04-20:
I'll admit I'm also somewhat baffled by the mechanism that allows the compilation to know tuning was run on the chip and start failing.

microPaul, 2022-04-21:
This may not be the same thing, but I'll mention it, in case it's helpful to your situation.  When I was working with the Crystal Frequency Calibration register for the 32khz xtal oscillator on the AVR128DB32, I had assumed that the calibration value was a two-compliment value.  But it's not (as confirmed by Microchip tech support).  The calibration for that was that the magnitude of the calibration value was in the lower value bits and the sign of the adjustment was in Bit 7.  So if you are putting -1 (0xFF) into the calibration register you may be asking for an adjustment of 127 units in the negative direction.  If the allowable range of magnitude values is less than 127, that may be why you are getting the error message, although it would seem surprising to me if the compiler is smart enough to know of that issue.

SpenceKonde, 2022-05-01:
Its not, its missing #defines in my code

SpenceKonde, 2022-05-01:
And also much thanks for reporting this one. I will add it to CI.

SpenceKonde, 2022-05-01:
BTW next intermediate release will auto cakibrate if no external clock is avaiklabvle  as lojng as it's aroujnd room temp iot will be very accurate,.

ObviousInRetrospect, 2022-05-02:
Cool, is that against the watch crystal? (auto-calibration with the rtc as a reference seems like it should be a thing).

The dd14 parts can't come soon enough - after switching to the db32's for some stuff the tiny-1 series is a power hog and the clock is horrible by comparison.

One area in the core that feels like it could be improved is a clean way to switch between 2-3 clock speeds (at runtime) while keeping millis(), delay(), and the UARTs working correctly. If nothing else, some of the bugs like uart wakeup can be worked around by sleep_mode_idle at a very low clock speed and then ramp the clock back up to normal on wakeup. I often have code that can get by at 1mhz 99% of the time but might want to burst to 4mhz or 8mhz rarely. Maybe fodder for the power saving library.

SpenceKonde, 2022-05-02:
I've thought about that but well, it would require 400 changes across 43 files, and prevent compile time optimization of timing related stuff..... Maybe 300 of those changes are in code,, and each change would add 4-8 bytes to the size of every binary even fi you didn't want it. The morning after I released it i'd be awoken by an angry mob outside my apartment complaining about the bloat and performance impact

SpenceKonde, 2022-05-02:
And micros() couldn't get the division optimized away either. that''d be another few hundred by my reckoning.

SpenceKonde, 2022-05-02:
And no I have a way that''s good enough without using a watch crystal when running at room temperature. It uses the oscillator error terms from the sigrow (at 25C) to calculate the actual speed of the HF oscillator. Then you time the internal 32 KHz low power oscillator against that. That oscillator is inaccurate, but it is stable. Loop over all possible cal values, timing the HF oscillator against the speed tyou determined the low speed oscillator is running at. Store the best cal values for targeted speeds as the current tuning does. Badabing. 

Somewhere in the documentation there's a jolly story about absentminded Prof Konde giving a lesson rambling about how if only there was a second timebase we could self-calibrate while Mr. Clock keeps interrupting, then chucks me out the open window,  announces his name to the class and that he "keeps it real, unlike a certain instructor" and describes for the class the general principle. I get my revenge at  the end of that little vignette though, because he forgot that the window hhe threw me out of was for the windowed WDT and triied to dismiss class early to hit the beach, arousing the watchdog which tore him to ribbons and reset he class to the beginning ;-)  Sometimes I need to have a little fun with these docs.  I think it's in the tuning reference.

SpenceKonde, 2022-05-21:
It does appear that there may be a bug with the guesses used when turned frequency options are selected but the part has not been tuned.

SpenceKonde, 2022-05-24:
Uh.... wait... how on earth are you getting the option to compile with tuning for prescaled clock speeds? That was never supported...

SpenceKonde, 2022-05-24:
Bug found with 1 MHz if tuning is enabled. And apparently all the machinery for tuning support at prescaled speeds is in 2.5.11 so will add them to the menus as well as the fix.

ObviousInRetrospect, 2022-05-25:
For what it is worth since the tuning was broken at 1mhz I found a workaround that is actually much less effort and might be worth documenting:

Taking a step back, I was using tuning because I hit a pair of chips that were sufficiently mistuned as to corrupt their serial output at on any of the 16mhz derived speeds (including 1mhz). (I've used dozens of this part, this was the first time I hit chips that weren't good enough for serial by default). I realized that a simple sketch that stepped through each of the 64 tuning values, enabled it, and printed a message (including the value) to the UART was sufficient.

typically 4 of the 64 tuning values will produce corruption-free serial output.

[unreadable] x many
[very corrupt]
[slightly corrupt0]
[clean1]
[clean2]
[clean3]
[clean4]
[slightly corrupt5]
[very corrupt]
[unreadable] x many

if corrupt0 is better than corrupt5, use clean2, otherwise use clean3

I choose to USERSIG.write(0,clean3's value) and then in the real sketch at the top of setup:
```
  uint8_t cal = USERSIG.read(0);
  if(cal <64) _PROTECTED_WRITE(CLKCTRL_OSC20MCALIBA, cal);
```

serial_tuner.ino
```
uint8_t initval;
void setup() {
  initval=CLKCTRL_OSC20MCALIBA;
  //USERSIG.write(0,13); //INSERT CORRECT VALUE HERE and UNCOMMENT
  Serial.swap(); //standard uart pins are the crystal
  Serial.begin(115200);//, SERIAL_TX_ONLY);

  Serial.println("if this is corruption free the default tuning is acceptable")
  for(int j=0; j<5; j++) Serial.println("The quick brown fox jumps over the lazy dog.");
  Serial.println(); Serial.println();

  uint8_t cal = USERSIG.read(0);
  if(cal <64){
    _PROTECTED_WRITE(CLKCTRL_OSC20MCALIBA, cal);
    _NOP();_NOP();
    delay(100);
    Serial.println("if this is corruption free the current tuning is acceptable")
    for(int j=0; j<5; j++) Serial.println("The quick brown fox jumps over the lazy dog.");
  }
}

void loop() {
  for(int i=0; i<64; i++){
    _PROTECTED_WRITE(CLKCTRL_OSC20MCALIBA, i);
    _NOP();
    _NOP();
    delay(100);
    Serial.println();
    Serial.println();
    Serial.print("============= ");
    Serial.print(i);
    Serial.print(" (default was: ");
    Serial.print(initval);
    Serial.println(") =============");
    for(int j=0; j<5; j++) Serial.println("The quick brown fox jumps over the lazy dog.");
  }
  delay(10000);
}
```

this requires burning the sketch, waiting 7 seconds, inserting the proper value and re-burning. No second mcu or wiring required beyond connecting a serial adapter.

hmeijdam, 2022-05-25:
A few semicolons were missing in your serial_tuner.ino  sketch, but after fixing that I still get an error

`'USERSIG' was not declared in this scope
`
How did you compile your sketch?

full error dump if needed:
```
Arduino: 1.8.13 (Windows 10), Board: "ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204, ATtiny1614, 16 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), On all pins, like usual"

E:\Arduino\arduino-1.8.13-megatinycore\arduino-builder -dump-prefs -logger=machine -hardware E:\Arduino\arduino-1.8.13-megatinycore\hardware -hardware E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-megatinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-megatinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-megatinycore\libraries -libraries E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\libraries -fqbn=megaTinyCore:megaavr:atxy4:chip=1614,clock=16internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,startuptime=8,wiremode=mors,printf=default,attach=allenabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_222607 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_595047 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_627105\sketch_may25a.ino

E:\Arduino\arduino-1.8.13-megatinycore\arduino-builder -compile -logger=machine -hardware E:\Arduino\arduino-1.8.13-megatinycore\hardware -hardware E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-megatinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-megatinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-megatinycore\libraries -libraries E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\libraries -fqbn=megaTinyCore:megaavr:atxy4:chip=1614,clock=16internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,startuptime=8,wiremode=mors,printf=default,attach=allenabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_222607 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_595047 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_627105\sketch_may25a.ino

Using board 'atxy4' from platform in folder: E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr

Using core 'megatinycore' from platform in folder: E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr

Detecting libraries used...

"E:\\Arduino\\arduino-1.8.13-megatinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1614 -DF_CPU=16000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DARDUINO=10813 -DARDUINO_AVR_ATtiny1614 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.5.11\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=5UL -DMEGATINYCORE_PATCH=11UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy4 "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\variants\\txy4" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_222607\\sketch\\sketch_may25a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Generating function prototypes...

"E:\\Arduino\\arduino-1.8.13-megatinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1614 -DF_CPU=16000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DARDUINO=10813 -DARDUINO_AVR_ATtiny1614 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.5.11\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=5UL -DMEGATINYCORE_PATCH=11UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy4 "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\variants\\txy4" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_222607\\sketch\\sketch_may25a.ino.cpp" -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_222607\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE

"E:\\Arduino\\arduino-1.8.13-megatinycore\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_222607\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

[... cut ...]
C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_627105\sketch_may25a.ino: In function 'void setup()':

sketch_may25a:12:17: error: 'USERSIG' was not declared in this scope

   uint8_t cal = USERSIG.read(0);

                 ^~~~~~~

C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_627105\sketch_may25a.ino:12:17: note: suggested alternative: 'USERROW'

   uint8_t cal = USERSIG.read(0);

                 ^~~~~~~

                 USERROW

exit status 1

'USERSIG' was not declared in this scope
```

hmeijdam, 2022-05-25:
Fixed the sketch.

```
#include <USERSIG.h>

uint8_t initval;
void setup() {
  initval=CLKCTRL_OSC20MCALIBA;
//  USERSIG.write(0,13); //INSERT CORRECT VALUE HERE and UNCOMMENT
  Serial.swap(); //standard uart pins are the crystal
  Serial.begin(115200);//, SERIAL_TX_ONLY);

  Serial.println("if this is corruption free the default tuning is acceptable");
  for(int j=0; j<5; j++) Serial.println("The quick brown fox jumps over the lazy dog.");
  Serial.println(); Serial.println();

  uint8_t cal = USERSIG.read(0);
  if(cal <64){
    _PROTECTED_WRITE(CLKCTRL_OSC20MCALIBA, cal);
    _NOP();_NOP();
    delay(100);
    Serial.println("if this is corruption free the current tuning is acceptable");
    for(int j=0; j<5; j++) Serial.println("The quick brown fox jumps over the lazy dog.");
  }
}

void loop() {
  for(int i=0; i<64; i++){
    _PROTECTED_WRITE(CLKCTRL_OSC20MCALIBA, i);
    _NOP();
    _NOP();
    delay(100);
    Serial.println();
    Serial.println();
    Serial.print("============= ");
    Serial.print(i);
    Serial.print(" (default was: ");
    Serial.print(initval);
    Serial.println(") =============");
    for(int j=0; j<5; j++) Serial.println("The quick brown fox jumps over the lazy dog.");
  }
  delay(20000);
}
```

SpenceKonde, 2022-05-27:
2,5.12 will address this properly; there will be a 1 MHz tuned option, and it will work.

As well as the self-tuning sketch, that relies on the factory cal + factory cal error (on parts that have it - the 2-series dropped that from the sigrow, Early 2-series had it but more recent ones don;'t bother because it was always 0, 1, or -1

SpenceKonde, 2022-06-11:
May now be fixed in github version

closed 2022-06-17 by SpenceKonde