SpenceKonde/megaTinyCore #692: sprintf problems with serial.print/corrupted memory

vslinuxdotnet, 2022-04-19:
I have some problems with the sprintf() with serial.print to print the correct variable, seams that some kind of memory corrupted.
When I change to snprintf(), all problems was solved!
The device used is ATtiny1614.

Don't print 99:
```
    char s[3];
    sprintf(s,"%d",99);
    Serial.println(s);
```

Print 99:
```
    char s[3];
    snprintf(s,2,"%d",99);
    Serial.println(s);
```

Regards,

VS

SpenceKonde, 2022-04-19:
I'm wondering if this is considered correct behavior. I've fallen out of my chair when i saw some things that were "undefined behavior" in  C. 

Basically, does this represent a regression from the behavior of the classic core which is inconsistent with standard implementations, or is all of arduino like this?

vslinuxdotnet, 2022-04-19:
I don't think that is the correct behavior, because in the atmega328p it works fine.
So may the sprintf() have the snprintf() in the main function ?
I tested in arduino IDE in Linux v1.8.18 and v1.8.15.

SpenceKonde, 2022-04-19:
Hmm, yea, I just reproduced on the 1614. 

Does it work on megacorex? That's where I still most of the prinf stujff

SpenceKonde, 2022-05-01:
almost certainly impacts dxcore too. Begging anyopne who knows how this works for help

SpenceKonde, 2022-05-02:
Adding blocked tag because I have not the faintest idea what's going on under the hood :-(

MX682X, 2022-06-22:
tested right now with attiny1614 and core 2.5.10 and 2.5.11. I was not able to reproduce the error. in fact, snprintf only printed one 9 (the second char being \0 and removed in the print function). Also tested on AVR128DA48 on latest DxCore. Not reproduced either. I'm using an oscilloscope to check the output. Also tested on 9600 baud and 115200 baud.

testCode:
```
void setup() {
  Serial.begin(9600);
  char s[3];
   sprintf(s,"%d",99);
   //snprintf(s,2,"%d",99);
   Serial.println(s);

   delay(500);
   CPU_CCP = 0xD8;
   RSTCTRL.SWRR = 1;

}

void loop() {}
```

jvasileff, 2022-09-06:
I gave this a try and my results agree with @MX682X's - everything works as expected and `snprintf` generates `9`. I tested a 3226 and a 1617, and on the 1617, I tried all three compile time printf options.

If anyone wants to share complete test code that fails for them, I'd be happy to give it a try.

The code I used:

```cpp
void setup() {
  Serial.begin(115200); 
  delay(2000);
  Serial.println("-- begin --");
}

void loop() {
  char s[3];
  sprintf(s, "%d", 99);
  Serial.println(s);        // prints "99"

  char t[3];
  snprintf(t, 2, "%d", 99);
  Serial.println(t);        // prints "9"

  delay(500);
}
```

SpenceKonde, 2022-09-08:
Thanks. Sounds like there isn't actually a problem here.

closed 2022-09-08 by SpenceKonde