SpenceKonde/megaTinyCore #701: Documentation Notes - "Getting Started"

clarkbremer, 2022-05-08:
I just went through the installation and programmed my first ATtiny with no issues.  Some notes on the documentation in the "Getting Started" section:

- "The newest versions (1.8.17, 1.8.18, and 1.8.19) generate a `panic: no major version found` error and fail to compile any sketch for me."  **I'm using `1.8.19`, and had no difficulty.**
- "Manual installation - assuming you want support for the 2-series - is more complicated".   **I'm using an ATtiny824, and did not need to manually install anything.  Worked "out of the box"!**  (Used a $6 CH340 with Schottky diode on Windows).
- Question:  Which is preferred spelling/capitalization:  `ATTiny` or `ATtiny`?  I've seen both in the docs, and when I submit PRs with edits, I'd like to use the preferred nomenclature.

SpenceKonde, 2022-05-08:
1. I have had problems with 1.8.19. In fact. Something, somewhere it decides doesn't have a major version and panics, I had thought it was my core, maybe it's some other core. The error message doesn't';t tell me which one it is is! So I still consider these versions unfit because an unidentified board package can blow the whole thing up without telling you which one it was that did it, even though the same board packages work on older versions. .... 

2. Did you manually install megaTinyCore or use board manager? That applies only when manual installation is being performed. Manual installation would be creating a Hardware folder inside your sketchbook, and cloning the repo into there. Installing via board manager automatically installs the required toolchain. I just checked a 1.8.19 virgin portable installation, and it definitely has no iot824.h so there is no way it could compile if the core was manually installed *unless* board manager had been used to install some other core which includes a newer version of the toolchain, which I think is just DxCore. I don't think anyone else is distributing updated toolchains. 

3. Per Microchip and Atmel before them, it's the ATtiny. The ATTinyCore name is historical - when I took over that project, that's what it was named, and I didn't want to muddy the water bhy changing it. Any place in this documentation that uses ATTiny is in error unless "ATTiny" is the start of "ATTinyCore" as in "parts released before 2016 are not supported by megaTinyCore, but most of them are supported by ATTinyCore" 

Note that as I understand it the "part numbers" or names start with ATtiny, however if one is being pedantic, they are "tinyAVR devices", as in "The tinyAVR 0/1/2-series devices are numbered as "ATtiny" followed by a 3 or 4 digit number, with the 1's place indicating the pincount (2 = 8, 4 = 14, 6 = 20, 7=24 - any other number indicates the part is not a 0/1/2-series tiny, but rather an older "classic" tinyAVR), and the tens place being 0, 1, or 2, denoting the series and hence featureset, and the remaining 1 or 2 digits, the flash size in kiliobytes." (the ATtiny828 is a good example of a confusing part number - it's a classic tiny and a rather gimped one at that). Similarly, the ATtiny1634 looks like it would be a 16k part with 14 pins from a future 3-series, but it too is a classic tinyAVR. One wonders how they will handle that when they release a tinyAVR 3-series (I predict they won't, and that the 2-series will be the last set of parts to bear the ATtiny name). Note also that there is no space between ATtiny and the numbers; ATtiny 3224 is wrong, ATtiny3224 is correct.  

The phrase "classic AVR" and "modern AVR" is not official, but microchip has refused to give an official name to parts before and after the 2016 revolution (also not an official name, obviously). Despite the fact that most of the code on DxCore and megaTinyCore is the same, and most of it could be used unmodified to make a megaAVR 0-series core. The megaAVR 0-series was the only post-revolutionary series to feature the ATmega branding.  

Note that Microchip uses the term megaAVR to refer to all devices with part numbers starting with ATmega, both the classic ones like the 328p and the megaAVR 0-series like the 4809. Arduino on the other hand internally uses "avr" and "megaavr" to refer to what I call classic AVRs and modern AVRs (and we have to follow that convention otherwise libraries will think they're incompatible).

clarkbremer, 2022-05-08:
After reading the docs, I was anticipating a bumpy ride, since I had the latest version of IDE, and a "type 2" ATtiny chip.  Others might be frightened also, so perhaps we could clarify some.

1.  I think it must be something unique to your install, Spence.  I had no issues.
2.  I used board manager.  Again, worked just as expected.  It was not clear from the docs that it would work because I needed "type 2" support.  I do have a few other non-standard board packages installed, notably ESP32.  The comment in item 3 about waiting for the list to load might be more appropriate for item 4.  The install there took longer than I expected.  If it's all right with you, I'll add that comment.  I also think it would be nice to add a few more steps there about selecting which chip you have, and which programmer.
3. Noted.  ATtiny824.

SpenceKonde, 2022-05-15:
1, Well I've had numerous other users report the issue to me as well. so it's not just me, unfortunately. 

2. I will try to clarify that a board manager installation should not require any manual steps. 

Another terminology aside. Per Microchip, they're called "tinyAVR "2-series" not "type 2". They love the word "Series" - there's the tinyAVR 0-series, 1-series, and 2-series, and the AVR DA-series or Dx-series, there will be an EA-series. I don't think the word "type" is in their dialect of marketingese.

commit 9622b00d (2022-05-21): Clarify readme

Resolves #701
closed 2022-05-21 by SpenceKonde