SpenceKonde/megaTinyCore #715: Can't get events to work

CDeenen, 2022-05-25:
I'm trying to experiment with the event system on a 402, but I can't get it to work.

I've tried the example 'Simple_event.ino' with a switch on PA1 and an oscilloscope on PA2, but I can't get PA2 to change using the switch.
I've also tried triggering the user with `long_soft_event()`, which also didn't work. However, `soft_event()` did work.

SpenceKonde, 2022-05-26:
Thanks for reporting but without seeing your code so that I can reproduce the bug, I don't really have a path forward to resolve it. Can you please post the code you used?  At this point, I don't even know if you're usiong event library or configuring it manually!

CDeenen, 2022-05-26:
Thanks for your response. 
I'm using the 'Simple_event.ino' example from the library, see [here](https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/libraries/Event/examples/Simple_event/Simple_event.ino).

hmeijdam, 2022-05-26:
That example also fails me on a 204 and a 412. No signal on PA2

hmeijdam, 2022-05-26:
This should take the input from PA1 and show that as output on PA2 via the event system

```
void setup() {

  // configure ports / pins
  PORTA_PIN1CTRL = PORT_PULLUPEN_bm; // enable PortA Pin2 pullup
  // VPORTA_DIR |= PIN2_bm;   // set pin PA2 as output (not required. Portmux seems to set PA2 output)

  // configure event channel
  EVSYS_ASYNCCH0 = EVSYS_ASYNCCH0_PORTA_PIN1_gc;   // link event system async ch0 to pin PA1 (0x0B) (datasheet 14.5.3)
  EVSYS_ASYNCUSER8 = EVSYS_ASYNCUSER0_ASYNCCH0_gc; // link (0x08) Event out 0 (=AYNCUSER8) to asynch event ch0 (0x03) (datasheet 14.5.5)
  PORTMUX_CTRLA = PORTMUX_EVOUT0_bm; // enable event output 0 in Port A. This is always on pin PA2 ! (Chapter 5 datasheet)
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

SpenceKonde, 2022-06-06:
I've found and am working on correcting a large number of bugs in the event library

SpenceKonde, 2022-06-19:
@MCUdude - once I get this tested and confirmed working, you should probably pull in the changes, as the event library was crawling with tinyAVR bugs.

fijam, 2022-06-29:
I asked about it in February: https://github.com/SpenceKonde/megaTinyCore/discussions/644

Glad to see it's finally getting some attention.

SpenceKonde, 2022-07-02:
In my defense, you never replied to my follow-up question, so I assumed you'd figured out the problem and that there wasn't a bug.

SpenceKonde, 2022-07-23:
It is believed that the event bugs have been fixed.

closed 2022-07-23 by SpenceKonde