SpenceKonde/megaTinyCore #716: Linux Upload Fails

dattasaurabh82, 2022-05-26:
## System: 
**OS:** Ubuntu 22.04 LTS x86_64
**Kernel:** 5.17.8-051708-generic
**IDE:**  1.8.19 (This IDE version works on mac and win)

## Issue:
Upload failure. 
Guessing yaml lib is the culprit. 

## Arduino IDE verbose: 
```shell
Arduino: 1.8.19 (Linux), Board: "ATtiny3227/3217/1627/1617/1607/827/817/807/427/417, ATtiny1607, 5 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (no reset pin), 0ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), On all pins, new implementation."

/home/linuxbox/arduino-1.8.19/arduino-builder -dump-prefs -logger=machine -hardware /home/linuxbox/arduino-1.8.19/hardware -hardware /home/linuxbox/.arduino15/packages -tools /home/linuxbox/arduino-1.8.19/tools-builder -tools /home/linuxbox/arduino-1.8.19/hardware/tools/avr -tools /home/linuxbox/.arduino15/packages -built-in-libraries /home/linuxbox/arduino-1.8.19/libraries -libraries /home/linuxbox/Arduino/libraries -fqbn=megaTinyCore:megaavr:atxy7:chip=1607,clock=5internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=0,wiremode=mors,printf=default,attach=allenabled -vid-pid=0403_6001 -ide-version=10819 -build-path /tmp/arduino_build_37393 -warnings=default -build-cache /tmp/arduino_cache_316514 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/linuxbox/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/linuxbox/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/home/linuxbox/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=/home/linuxbox/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18 -prefs=runtime.tools.avr-gcc.path=/home/linuxbox/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino4b.path=/home/linuxbox/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.python3.path=/home/linuxbox/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/linuxbox/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -verbose /tmp/untitled920790078.tmp/sketch_may27b/sketch_may27b.ino
/home/linuxbox/arduino-1.8.19/arduino-builder -compile -logger=machine -hardware /home/linuxbox/arduino-1.8.19/hardware -hardware /home/linuxbox/.arduino15/packages -tools /home/linuxbox/arduino-1.8.19/tools-builder -tools /home/linuxbox/arduino-1.8.19/hardware/tools/avr -tools /home/linuxbox/.arduino15/packages -built-in-libraries /home/linuxbox/arduino-1.8.19/libraries -libraries /home/linuxbox/Arduino/libraries -fqbn=megaTinyCore:megaavr:atxy7:chip=1607,clock=5internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=0,wiremode=mors,printf=default,attach=allenabled -vid-pid=0403_6001 -ide-version=10819 -build-path /tmp/arduino_build_37393 -warnings=default -build-cache /tmp/arduino_cache_316514 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/linuxbox/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/linuxbox/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/home/linuxbox/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=/home/linuxbox/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18 -prefs=runtime.tools.avr-gcc.path=/home/linuxbox/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino4b.path=/home/linuxbox/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.python3.path=/home/linuxbox/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/linuxbox/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -verbose /tmp/untitled920790078.tmp/sketch_may27b/sketch_may27b.ino
Using board 'atxy7' from platform in folder: /home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11
Using core 'megatinycore' from platform in folder: /home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11
Detecting libraries used...
/home/linuxbox/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1607 -DF_CPU=5000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERA0 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_ATtiny1607 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.5.11\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=5UL -DMEGATINYCORE_PATCH=11UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/api/deprecated -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/variants/txy7 /tmp/arduino_build_37393/sketch/sketch_may27b.ino.cpp -o /dev/null
Generating function prototypes...
/home/linuxbox/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1607 -DF_CPU=5000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERA0 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_ATtiny1607 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.5.11\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=5UL -DMEGATINYCORE_PATCH=11UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/api/deprecated -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/variants/txy7 /tmp/arduino_build_37393/sketch/sketch_may27b.ino.cpp -o /tmp/arduino_build_37393/preproc/ctags_target_for_gcc_minus_e.cpp
/home/linuxbox/arduino-1.8.19/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_37393/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/linuxbox/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny1607 -DF_CPU=5000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERA0 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_ATtiny1607 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.5.11\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=5UL -DMEGATINYCORE_PATCH=11UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/api/deprecated -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/variants/txy7 /tmp/arduino_build_37393/sketch/sketch_may27b.ino.cpp -o /tmp/arduino_build_37393/sketch/sketch_may27b.ino.cpp.o
Compiling libraries...
Compiling core...
/home/linuxbox/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-gcc -c -g -x assembler-with-cpp -flto -MMD -mmcu=attiny1607 -DF_CPU=5000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERA0 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_ATtiny1607 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.5.11\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=5UL -DMEGATINYCORE_PATCH=11UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore/api/deprecated -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/cores/megatinycore -I/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/variants/txy7 /home/linuxbox/.arduino15/packages/megaTinyCore/ha [... cut ...]
  File "/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/tools/libs/yaml/constructor.py", line 204, in construct_mapping
    return super().construct_mapping(node, deep=deep)
  File "/home/linuxbox/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/tools/libs/yaml/constructor.py", line 126, in construct_mapping
    if not isinstance(key, collections.Hashable):
AttributeError: module 'collections' has no attribute 'Hashable'
AttributeError: module 'collections' has no attribute 'Hashable'
```

dattasaurabh82, 2022-05-28:
Found a solution (and explanation), may be temporary 😅  from here: [https://github.com/ablab/spades/issues/873](https://github.com/ablab/spades/issues/873)

Basically Edit the: [https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/tools/libs/yaml/constructor.py](https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/tools/libs/yaml/constructor.py) line 126

Note if user's want to edit this in their local system, at-least in my SYSTEM (as mentioned above), the path of this file is: 
```shell
/home/<USER>/.arduino15/packages/megaTinyCore/hardware/megaavr/<installed MegatIcyCore Version>/tools/libs/yaml/constructor.py
``` 

Edit line 126 in that `constructor.py`:
```shell
$ nano -l constructor.py

# change if not isinstance(key, collections.Hashable): to if not isinstance(key, collections.abc.Hashable):
# .
# OTHER THINGS IN THE FILE
# .

# if not isinstance(key, collections.Hashable):
if not isinstance(key, collections.abc.Hashable):

# .
# OTHER THINGS IN THE FILE
# .
```

Any permanent solution for this? 🤓

SpenceKonde, 2022-05-30:
Blame Python for this, they fucking changed how their language worked and broke tons of existing code. Arrogant twits. 

What I need in order to fix this is a way to detect whether it's a version that does or doesn't need this fix and react accordingly. See, I don't really know python,. I just stumble through it and work on parts of itthat are very close to things I Do know. But this is unrelated to anythign I know about python.

dattasaurabh82, 2022-06-02:
> What I need in order to fix this is a way to detect whether it's a version that does or doesn't need this fix and react accordingly

I will try something tomorrow and create a pull request, you can check afterwards

SpenceKonde, 2022-06-19:
Any luck with this? My posts on this got deletedon the arduino forums so i can't get help there. This is a critical issue and is blockingrelease ofthe next versions

SpenceKonde, 2022-07-07:
pymcuprog *definitely* must not be "updated" to the standard version!!

The stock pymcuprog implementation is not compatible with prog.py and even if it was, uploads will go at around 1/20th of the speed because it makes no attempt to optimize performance of the physical layer nor account for the fact that there is a minimum delay (USB latency) (0.5ms to 16ms depending on the driver and hardware), leaving on the response signature even during 512 byte repeated writes which ensures that each word of data takes a whole usb latency period. On the one hand, the version supplied with the core not only turns off the response signature during repeated transfers, it also combines successive commands that won't generate response signatures into a single transaction saving more USB latency (which at top speed, is about 50% of the time taken to upload to a tinyAVR with 128b pages (smaller pages = more time in latency) . 

And the delay generating library has also been modified to do a better job of making short delays on windows.

So the problem is in the version of the yaml library? Where do I get the correct version to use? See I didn't set that all up, I understand the phy and the microcontroller side, but python is largely outside of my area of expertise/ , or work in environments where everything is either globally installed (hopeless for this because of version differences, the reason we "vendorize" the libraries as you say is because otherwise we have no idea what versions are being used and whether they will play well together.  

This problem, as I see it, is essentially because the linux platforms don't get an actual set of python binaries packaged up like windows does, just links to the system installation, so version changes can shit over everything and turn working code into broken code without us changing anything. As happened here. I'd like to be able to have the python package pull in fully independent binaries for each platform so we could actually pick a version that is known to work with the codebase, like we do on windows.

SpenceKonde, 2022-07-23:
fix has been merged, now needs to be tested on windows

closed 2022-07-23 by SpenceKonde