SpenceKonde/megaTinyCore #722: Logic and Comparator libraries cannot be used together

TechAUmNu, 2022-06-05:
Logic and Comparator libraries both define enums with the same names in their .h 
enum output_t 
enum pinswap_t

SpenceKonde, 2022-06-06:
This i a duplicate of https://github.com/SpenceKonde/DxCore/issues/279

I am unsure how to correctly resolve it without introducing breaking changes - I am not comfortable with namespaces and the ++ part of C++ in general and have, in that issue, asked @mcudude who created those libraries how to fix it. We can't prefix them because that will trash any existing code!

SpenceKonde, 2022-06-06:
Adding blocked label because cannot fix without advise from someone more knowledabkle about C++

SpenceKonde, 2022-06-19:
@MCUdude  - this is another super critical problem, I've had at least three people complaining about it. If you could suggest a fix especially if it didn't break existing code, that  would be really helpful. I don't see any solution that won't break all existing code that uses Comparator, Logic, or ZCD (because IIRC that also names output the same.

MCUdude, 2022-06-19:
>  If you could suggest a fix especially if it didn't break existing code, that would be really helpful.

I already have!

SpenceKonde, 2022-07-23:
Changes checked into Logic, Event and Comparator

closed 2022-07-23 by SpenceKonde