SpenceKonde/megaTinyCore #736: 2.5.11 + Arduino IDE 2.0.0-rc8: Suggestions not working when selecting "w optiboot" targets

InstantMuffin, 2022-07-09:
Hello,

using XUbuntu 22.04 LTS, Arduino IDE 2.0.0-rc8 and megaTinyCore 2.5.11 editor suggestions do not work when, and only when I select a board type that has the "w/Optiboot" postfix.

For keywords that already exist in the sketch (like in this example "digitalWrite"), this is the behavior:
[image]

Otherwise nothing is being suggested.
Compilation (and upload) works fine.

I attached logfiles from the language server.

[languageserverlog.zip](https://github.com/SpenceKonde/megaTinyCore/files/9076215/languageserverlog.zip)

InstantMuffin, 2022-07-09:
Okay, a little update.
So as shown in the list above there no CPU targets for any of the 2nd generation MCUs.
The other non-optiboot entries respective to the ones with the error work because the CPU target is always a 1st generation MCU.

Looking at the boards.txt and compile_commands.json the CPU target is always the first CPU entry in boards.txt for the respective compile-target/MCU-family entry.
I tested this by checking the specified CPU target in compile_commands.json after selecting different entires under "Tools/Chip/" (which didn't change the target in compile_commands.json), and then by exchanging the order the CPU entries are defined under boards.txt.

InstantMuffin, 2022-07-10:
Mentioning https://github.com/arduino/arduino-ide/issues/31 as this is all the same bug it seems.

per1234, 2022-07-10:
>even though you select the "3226" under "Tools/Chip/" it uses "3216" in the entire params (for example "-mmcu=attiny3216").

This bug is being tracked here: https://github.com/arduino/arduino-ide/issues/1029

SpenceKonde, 2022-07-11:
Thanks for that info Per! 

Since this is a confirmed bug in the IDE, not a defect in the core, I'm going to close this one.

closed 2022-07-11 by SpenceKonde