SpenceKonde/megaTinyCore #738: Round produces unexpected results, extra () might help.

ieb, 2022-07-12:
https://github.com/SpenceKonde/megaTinyCore/blob/a0aa75cf63a0fce341cfbce118d65a41afb96bdb/megaavr/cores/megatinycore/api/Common.h#L69

double vd = round(value/precision)  as used in places like [1] has rounding errors.

    value:32.81000 precision:0.10000 vd:320.50, should be 328.50

there is a simple fix, if the code is your own 

double vd = round((value/precision))

1 https://github.com/ttlappalainen/NMEA2000/blob/c2de650e9abe943da7c87f6a6c03796c4f7b3234/src/N2kMsg.cpp#L587

SpenceKonde, 2022-07-14:
Please post full code that reproduces this (use latest release, serial is currently busted in master and I won'[t have time to look at it until firday at earliest

SpenceKonde, 2022-07-19:
I cannot reproduce this problem. 

```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(1000);
  float value = 32.81;
  float precision = 0.1;
  float vd = round(value/precision);
  Serial.println(vd);
}
```
produces the correct value, which is neither of the two that you listed - it gives 328.0 for me, which is what it should give, because the it's there to turn an floating point value into an integer. I'm not sure why we're screwing around with __type_of__ there to be honest, because there is only one type of the argument that makes sense (float aka double - the two are the same in Arduino-land). for any non-float, round(x) == x.  In any event it's cast to a which are integer types. The added 0.5 is just to make sure they round in the right direction.

closed 2022-07-19 by SpenceKonde