SpenceKonde/megaTinyCore #748: Check keywords.txt now spews hundreds of errors

SpenceKonde, 2022-07-23:
Nothing in the fucking code was changed, why is this doing this?

@per1234  do you have any idea what changed?

per1234, 2022-07-31:
@SpenceKonde do you mean these?:

https://github.com/SpenceKonde/megaTinyCore/runs/7483602119?check_suite_focus=true#step:4:9

```text
WARNING: Arduino IDE is not installed so unable to check for invalid reference links. Please call install_ide before running check_keywords_txt.
```

As indicated by the "WARNING" prefix they are warnings, not errors.

The `check_keywords_txt` function used by the "Check keywords.txt" workflow has an optional feature to check whether there is a reference page for the values in [the `REFERENCE_LINK` field of `keywords.txt`](https://arduino.github.io/arduino-cli/latest/library-specification/#reference_link). This warning is only telling you that check is not possible because the Arduino IDE was not installed on the GitHub Actions runner by calling the `install_ide` function. I omitted that intentionally because, in an application like this where the script is only being used for the `keywords.txt` check, the Arduino IDE installation is only needed for that single check and so it is questionable whether it is worth the time required to download and install the IDE.

So these warnings are expected for the configuration of this workflow and were there even before the legitimate failure occurred ([example](https://github.com/SpenceKonde/megaTinyCore/runs/7483621767?check_suite_focus=true#step:4:9)).

If you do decide to enable the feature, please note that, due to changes in later versions of the Arduino IDE, you must install Arduino IDE <=1.8.15: https://github.com/per1234/arduino-ci-script/issues/32

SpenceKonde, 2022-08-04:
Thanks for the explanation

Seeing as versions of 1.8.14 and later have serious regressions, I don't see that happening any time soon,....(8.14 added the bug with the "Panic no major version found" with no indication of what a major version of wasn't found, and 2..0.0 beta seems to have a lot of problems.......

per1234, 2022-08-04:
Yes, we discussed this on the forum:

https://forum.arduino.cc/t/panic-no-major-version-found/876644

The bug is being tracked here: https://github.com/arduino/arduino-cli/issues/1823 (I published it from the internal issue tracker to GitHub)

SpenceKonde, 2022-08-12:
issue is resolved

closed 2022-08-12 by SpenceKonde