SpenceKonde/megaTinyCore #751: problems with 16mhz derived speeds, request for <5mhz clock options derived from 20

ObviousInRetrospect, 2022-07-27:
My past experience with the tiny-1 series was 100% success with both the 16mhz and 20mhz derived clocks being good enough for serial communication to work.

I am accumulating examples of attiny1614s and now attiny3224s where serial works untuned on 20mhz derived speeds but require tuning for the 16mhz derived speeds. 100% of chips I have tested still work off the 20mhz derived without tuning.

megaTinyCore offers lots of clock options including tuning to hit some odd faster speeds. It offers 1 slow speed (1mhz) derived from the 16mhz clock. Sleep standby wake from serial seems to loose/corrupt data and sleep idle wastes power proportional to frequency so the lowest viable clock speed can be useful for devices that must listen on a uart in sleep idle.

When it was just attiny1614s I assumed I got a bad batch. But now that I am seeing 3224s doing the same thing any thoughts on adding a few slower speeds off of the 20mhz oscillator?

Of the options achievable, (2.5,1.25,0.625,0.3125,3.33,2,0.83,0.41) I would suggest the following priority order:

1.25mhz (20/16)
625khz (20/32)
2.5mhz (20/8)
312.5khz (20/64)
3.33mhz (20/6 - default clock post-reset)
2mhz (20/10)

For parity, completeness, and VCO devices it might also be worth adding 500khz, 250khz, and 2.66mhz derived off the 16mhz (and for when the trend reverses and the 16 starts being the accurate one).

SpenceKonde, 2022-07-27:
Take a look at boards.txt in latest version of repo.

But we can't do fractional speeds without a ton of work, - we rely on being
able to use can integer value for clock cycles per us everywhere. There are
no plans to work on such a potentially time consuming feature with so many
pressing priorities

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

ObviousInRetrospect, 2022-07-27:
on the new 2mhz, there is a bug in line 1346 of wiring.c:

      #elif (F_CPU == 2000000) // 16MHz prescaled by 16
        /* Clock DIV16 */
        _PROTECTED_WRITE(CLKCTRL_MCLKCTRLB, (CLKCTRL_PEN_bm | CLKCTRL_PDIV_16X_gc));
      #elif (F_CPU == 1000000) // 16MHz prescaled by 16
        /* Clock DIV16 */
        _PROTECTED_WRITE(CLKCTRL_MCLKCTRLB, (CLKCTRL_PEN_bm | CLKCTRL_PDIV_16X_gc));

it should either be CLKCTRL_PDIV_10X_gc (fused to 20) or CLKCTRL_PDIV_8X_gc (fused to 16). For the reason I reference here, CLKCTRL_PDIV_10X_gc would be far better.

And yes fractional clock speeds <<<<<<<<<< DD support working in dxcore :-).

ObviousInRetrospect, 2022-07-27:
looks like the 2mhz mode missed the 14-pin parts in boards.txt and might be tuned only for some others

ObviousInRetrospect, 2022-07-28:
pull-request #753 fixes this bug, adds non-tuned 2mhz to all 13 x 4 places I think it needs to go, and sets OSCCFG to use the 20mhz.

SpenceKonde, 2022-07-28:
Your observation of inaccurate 16 MHz is inconsistent with my data, please present supporting evidence in the form of frequency measurements against a stable timebase for all oscillator settings (preferably one line per part, 1 comma and no space between each calbyte value, that will maximize the ease with which I can bring it into excel to visualize the data and get some confidence that I'm not chasing a phantom.

ObviousInRetrospect, 2022-07-29:
Will collect data and report back if the serial tuner output showing 100% failure rate at 4mhz and 100% success rate at 5mhz across 9 samples isn't sufficient. If you have a t1614 or t3224 board handy would you mind doing a quick test to see that you get working serial at 115200 with no tuning enabled on both 4mhz and 5mhz? (I have unsurprisingly had no serial problems on the dx side).

I had hoped to use the delta between the at-boot calibration register value and the one chosen for serial but it appears the bootup ones are all >64 - do you know the mapping? 

Unfortunately I need to completely redesign my clock test setup that I was using for 32k crystals as it relies on working serial so the data might be a couple days out.

SpenceKonde, 2022-07-30:
I'm, trying to fix megaTinyTuner sketch to tune and if debug set, to output the speed at every calbyte setting in a format ready for processing using just 2-3 regexes to get a .csv file you can analyze in excel. and will let you know when I've got that working.... right now it works on 2-series but does nothing on 1-series.

SpenceKonde, 2022-07-30:
Okay this will take longer than planned, It wass not autosaved because it was an excample and windows rebooted itself and I lost all my wortk on tuning from yesterday

20 years ago, when I had a computer that rebooted itself of it's own accord on a monthly basis, I would beging replacing parts to find the hardware defect. Now, windows update simulates broken hardware  :-(

ObviousInRetrospect, 2022-07-30:
ok so I went ahead and built out an i2c debugging output channel so I can easily see WTF is going on even with serial dead.

I only took 2 data points because they were both about 25% fast.
```
51:51:57:34:33:20:77:15:06:23:FF:3F:10:05:FF, 10000000, 39, 99.84, 998
51:51:57:34:33:20:77:15:06:23:FF:3F:10:05:FF, 8000000, 46, 124.80, 1248
51:51:57:34:33:20:D7:84:00:30:FF:3F:10:06:FF, 10000000, 21, 100.97, 1009
51:51:57:34:33:20:D7:84:00:30:FF:3F:10:06:FF, 8000000, 52, 126.25, 1263
```

The fuses for both are in the pull request thread

ObviousInRetrospect, 2022-07-30:
I validated this by observing that changing:

```
  //Serial.begin(9600);  
  Serial.begin((uint16_t)(9600/1.248));  
```

I thought safe fuses were autoset (and serialupdi says they are being set) but I tried "burn bootloader" which did nothing (because this error is what you would get if the software thought it was scaling 16mhz but it was really scaling 20mhz).

ObviousInRetrospect, 2022-07-30:
header for the csv is:
serial, F_CPU, test_duration, %error, single_sample

SpenceKonde, 2022-07-30:
Oh if the speeds are 25% fast - that suggests that the fuses is set to 20 MHz but the compiler is assuming it's set to 16 MHz

ObviousInRetrospect, 2022-07-30:
yeah thought I said that

SpenceKonde, 2022-07-30:
sounds like the OSCCFG fuse isn't being set, then, by Atmel Studio.

ObviousInRetrospect, 2022-07-30:
Here are the fuses from the other thread. They are in fact the same.

51:51:57:34:33:20:D7:84:00:30:FF:3F:10:06:FF:FF
00:00:02:FF:00:C5:04:00:00 (5)
00:00:02:FF:00:C5:04:00:00 (4, alleged)

51:51:57:34:33:20:B7:F0:06:2E:FF:3D:0F:06:FF:FF
00:00:02:FF:00:C5:04:00:00 (5)
00:00:02:FF:00:C5:04:00:00 (4, alleged)

SpenceKonde, 2022-07-30:
Yeah the third fuse should be 0x01 for 16 MHz

ObviousInRetrospect, 2022-07-30:
Yeah, with 4mhz selected it is saying  Set fuses: ['2:0x02', '6:0x04', '8:0x00'] on upload

I wonder if this is Arduino 2.0 biting me somehow. Anyway sorry about all the noise. Apparently my 16mhz clocks were all off by 25% which would explain serial being broken. I think I had tried baud adjustments by up to +/-15% a while ago. Needed more.

We can put the clock accuracy issue to bed 

```
Sketch uses 9371 bytes (57%) of program storage space. Maximum is 16384 bytes.
Global variables use 483 bytes (23%) of dynamic memory, leaving 1565 bytes for local variables. Maximum is 2048 bytes.
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.3 - Jan 2022
Using serial port /dev/cu.usbserial-1110 at 230400 baud.
Delaying next op after each page write command by 1.0ms (-wd).
Target: attiny1614
Set fuses: ['2:0x02', '6:0x04', '8:0x00']
Action: write
File: /private/var/folders/jp/8cvzydc904n1td9j3y4ntcgm0000gn/T/arduino-sketch-70FEBC706FCB10E5CEBC41B1CC9C64C8/ina_bbk_reva_clkcal_fastisr_i2cout.ino.hex
Pinging device...
Ping response: 1E9422
Setting fuse 0x2=0x2
Writing literal values...
Verifying literal values...
Action took 0.04s
Setting fuse 0x6=0x4
Writing literal values...
Verifying literal values...
Action took 0.04s
Setting fuse 0x8=0x0
Writing literal values...
Verifying literal values...
Action took 0.03s
Finished writing fuses.
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.01s
Writing from hex file...
Writing flash...
```

ObviousInRetrospect, 2022-07-30:
hmm on a different machine that has both Arduino 1.8.19 and 2.0.0-rc5-nightly (on that machine they share ~/Library/Arduino15/) the 1.x sets the fuses correctly and 2.0 also Set fuses: ['2:0x02', '6:0x04', '8:0x00']

that was a fresh upgrade of tinymegacore (haven't used Arduino on this machine in a while).

so something is busted with Arduino 2.0.0. If you feel like it see if you can reproduce and then add a major warning. Arduino 2 can sit alongside 1.x at least on Mac and I haven't had problems with it screwing with 1.x

ObviousInRetrospect, 2022-07-30:
https://forum.arduino.cc/t/problems-with-megatinycore-in-ide-2/959081

ObviousInRetrospect, 2022-07-30:
looks like hierarchal inheritance is broken in 2.0 and a more specific setting from a previous option is sticking around when not set for the new selection instead of the default defined higher up being used.

Not sure what other fuses or settings this bug would affect.

do you want a pull request that adds a few hundred boards.txt lines? Needs one of these for every clock speed:
atxy4.menu.clock.16internal.bootloader.OSCCFG=0x01

SpenceKonde, 2022-07-30:
I am not targeting 2.0.0 betas due to the unacceptable number of gross bugs
in that version. If they are still fucking it up a few minor versions after
its out of beta/prerelease I will consider it changing my code, but I'm not
going to make changea that will be rendered moot when the fix the bug
hopefully before the decade it released.

Additionally, a change like that I boards.txt will not be accepted, it must
be made to create_boards_txt.py

Currently we recommend 1.8.13 (not 1.8.12, which had a catastrophic bug) or
1.8.14 (which added a critical regression they still havent fixed)

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

ObviousInRetrospect, 2022-07-30:
is create_boards_txt.py checked in? I get:

We couldn’t find any code matching 'create_boards_txt.py' in [SpenceKonde/megaTinyCore](https://github.com/SpenceKonde/megaTinyCore)

If it exists, it should be trivial to work around the bug and add such a property. I can understand not wanting to do the work but I would hope you would be open to someone sending a patch to remove a known issue. This particular bug may have soured me on 2.0 but I despise the 1.x "IDE" that lacks basic code editing features and runs glacially slowly if you dare to keep a half a dozen sketches open even on good hardware. 2.0 for all its warts is at least based on a real IDE. I may run them side by side and use 1.x to build and 2.x to edit.

If create_boards_txt.py is a future project, a far simpler munge_boards_txt.py that makes a specific change in all needed places would be faster to write than making the change by hand. Or at least that was my conclusion 3/4 of the way through the 2mhz untuned clock insertion.

SpenceKonde, 2022-07-31:
No it's not checked in because it isn't completed, but any changes would be
undone, really close to being able to cite over

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2022-07-31:
Closing this issue in specific. The spec for third party hardware packages hasnt changed, so this a bug on their part, and as 2.0.x is BETA. we will be making no moves to accommodate it

closed 2022-07-31 by SpenceKonde

per1234, 2022-07-31:
>this a bug on their part

Correct. As mentioned in the forum topic linked above, it is tracked here: https://github.com/arduino/arduino-cli/issues/1614

SpenceKonde, 2022-08-01:
Is 2.0.0 even able to work with hardware packages that are manually installed? I recall older versions were not able to and that was teh start of my "no work on 2.0.0 stuff until 2.0.0 is working" policy, since 2.0.0 was clearly far from finished.

Personally I **ONLY** install board manager packages in portable installs.

SpenceKonde, 2022-08-01:
Reported january and still not fixed?

So far 1.8.13  (the last good build) and 2.0.0 are sounding like a going from an old rustybut working station wagon to Ferrari, except that the Ferrarri had the transmission stolen and doesnt't have a gas pedal. Only one of those will gt you to work,.