SpenceKonde/megaTinyCore #767: What the fuck is wrong with Wire now?

SpenceKonde, 2022-08-11:


SpenceKonde, 2022-08-11:
How does this make any sense whatsoever? 
  /home/runner/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.11/libraries/Wire/src/Wire.cpp:272:9: error: prototype for 'uint8_t TwoWire::requestFrom(uint8_t, uint8_t)' does not match any in class 'TwoWire'
   uint8_t TwoWire::requestFrom(uint8_t  address,  uint8_t  quantity)

SpenceKonde, 2022-08-11:
I am staring in the face into 2+2 asd getting 5. Wire library may need to be dropped as O I am geting over a dozen emails a day asking for new versions.

ObviousInRetrospect, 2022-08-12:
is this still broken and is it checked in (maybe to a different branch)? I'd take a look if it is.

SpenceKonde, 2022-08-12:
Yeah nothing with wire compiles https://github.com/SpenceKonde/megaTinyCore/actions/runs/2843603540

SpenceKonde, 2022-08-12:
The overall cause is that i do not know C++

ObviousInRetrospect, 2022-08-12:
found a couple issues:

1. why does it say 2.5.11 in the path
2. wire.h/c define TwoWire but your errors say wire

id put a #error in the code you think you are running and see if it is actually the code

SpenceKonde, 2022-08-12:
the 2.5.11 compiles are the previous version to compare file size.

SpenceKonde, 2022-08-12:
but of course I have no idea if there;s a way to see those reports or like make a graph over time to see slow regressions in sketch size or the improvements of the newest version

ObviousInRetrospect, 2022-08-12:
found it:
void          begin(uint8_t  address, uint8_t receive_broadcast, uint8_t second_address);
void TwoWire::begin(uint8_t address, bool receive_broadcast, uint8_t second_address)

arg 2 bool uint8

ObviousInRetrospect, 2022-08-12:
im also not honestly sure what the side effects of putting functions like:
  void beginTransmission(int     address) {
      beginTransmission((uint8_t)address);
    }

in the .h file are I always put all implementation in the c++ file

ObviousInRetrospect, 2022-08-12:
[image]

SpenceKonde, 2022-08-12:
I think that did it! THANK YOU I was stuck on that for over a day.

I generally only put trivial code in .h files. The stuff I'd put on one line if I didn't run stylecheck on libraries which doesn't like that.

closed 2022-08-12 by SpenceKonde