SpenceKonde/megaTinyCore #773: I2C Wire.endTransmission() method freezes

jacobComp, 2022-08-19:
I use ATTiny 414 and I try to communicate with I2C protocol but program freezes on endTransmission() method. I call this method with no parameters.

SpenceKonde, 2022-08-19:
Please provide code for both the devices involved in order for this to be debugged. Also I need to know which version you are using because the head is very different from last release currently.

jacobComp, 2022-08-19:
#include <Wire.h>

void setup()
{
  Serial.begin(115200);
  Wire.begin(); 
  Serial.println("Started...");
}

byte x = 0;

void loop()
{
  Serial.println("Debug 1");
  Wire.beginTransmission(4); // transmit to device #4
  Serial.println("Debug 2");
  Wire.write("x is ");        // sends five bytes
  Wire.write(x);              // sends one byte  
  Wire.endTransmission();    // stop transmitting
  Serial.println("Debug 3");
  Serial.println(x);
  x++;
  delay(2000);
}

I wrote simply debug String to COMPort, this code gives Started... , Debug1, Debug2
Writing Wire.end() gives Debug3 too but I need return message to get action
Thank you.

jacobComp, 2022-08-19:
I think I solved it via enable Internal Pull-up resistor of I2C pins

Modac, 2022-08-19:
Are you using external pullups?

jacobComp, 2022-08-19:
> Are you using external pullups?

No, hardware design team did not place external pull-up, I realise it after open this issue

SpenceKonde, 2022-08-19:
As documented in both Wire.h and the main readme, external pullups are mandatory. We provide a method for debugging only to turn on the internal ones, but they are less than 1 third the strength required by the standard and just on the edge of working, an state (in boldface text as I recall) that it UsePullups() should be used only for debugging and if it fixes any issue there is a problem with your external ones. Such as them not existing. 

This has always been the case on AVRs, classics too. But the Arduino team would prefer to lie to you and pretend you didn't need external ones, resulting in a system that works with one or two devices and short wires and craps out in more complex scenarios with intermittent results. Were there not laws about such violence in the workplace, I would advise you slap your hardware design team members upside the head - it's the most basic parameter of the I2C connections

InstantMuffin, 2022-08-19:
Arduino world really tell you with subtlety that you don't need any pullups when using I2C, and it almost gave me the impression that arduinos are superior in that way to other microcontroller (systems).
Arduinos are a somewhat great start to get into the topic, but the community also teaches you a lot of wrong things, and often outdated accessories are rebranded into arduino-friendly devices, for example the LN298.

SpenceKonde, 2022-08-19:
@InstantMuffin that's because the most important employee at Arduino is the one who goes around and waters the rest of them twice a week. 

As for the obsolete parts, that's ubiquitous in hobby electronics because designs from the dawn of the field get repeated by people who are lucky to guess which end of the soldering iron to pick up in two tries, but are real good at writing books that make them sound like they know their stuff.

closed 2022-08-19 by SpenceKonde