SpenceKonde/megaTinyCore #794: Path to "stream.h" in "HardwareSerial.h" in master

jvasileff, 2022-09-10:
In `megaavr/cores/megatinycore/HardwareSerial.h` there is a new include:

```
#include "Stream.h"
```

but this should instead be:

```
#include "api/Stream.h"
```

right? I'm unable to compile against current master using my Makefile without making this change.

jvasileff, 2022-09-10:
The workaround of adding the `api` directory to the includes path:

```
INCLUDES_MEGATINYCORE += -I$(MEGATINY_DIR)/cores/megatinycore/api/
```

doesn't work on the default case-ignoring macOS filesystem, because, it appears, `api/String.h` conflicts with `<string.h>`.

SpenceKonde, 2022-09-11:
Does this issue reproduce when using the IDE, or is it unique to compiling via makefile? I do *not* see this issue on Windows compiling through the Arduino IDE. 

And I'm confused, does #include "api/Stream.h" work or not?

closed 2022-09-11 by SpenceKonde

jvasileff, 2022-09-11:
Thanks. Yes, the change you committed does fix compiling using a simple makefile. I haven't tried to compile using Arduino IDE.

jvasileff, 2022-09-11:
I tried using `2.6.0` with the Arduino IDE on macOS, and it works fine without the fix.

The reason it works on Windows and macOS filesystems is that the IDE compiles with the flag:

```
-I/Users/jvasileff/Documents/Arduino-Zeta/hardware/megaTinyCore/megaavr/cores/megatinycore/api/deprecated
```

and the "problem" file `String.h` is not in the **`deprecated/`** subdirectory.

I believe the IDE would fail if it tried to use `-I /.../megatinycore/api/`, just as my attempted workaround failed. Likewise, my makefile would have worked with `-I /.../megatinycore/api/deprecated` without the fix.

*TLDR* All is good, but having files in the core with names like `String.h` can cause confusion.

SpenceKonde, 2022-09-11:
thanks, all these damned third party compilation environments And I got my hands full testing on the IDE
Changing the name of String,h can't be done because it's part of the API an libraries depend on it. Direct complaints about naming of api files to the Arduino team.

SpenceKonde, 2022-09-11:
Now you have less than 24 hours to find a severe bug tt impacts arduino IDE users such that I'd need to respin 2.6.0before the release which will happene monday as soon as I get back to my new wireless cellphone charger because the charging port as crapped out on my phone and I can't to the 2FA to get the new toolchains available

SpenceKonde, 2022-09-12:
No shit it's fucking dumb we  have potential naming collisions. I don''t lmpw why you are looking at me instead of the AVRlibC people! Since thy actuallymade the decisions they might know why.

jvasileff, 2022-09-12:
Believe me, I have no interest in pointing fingers. But it is worth pointing out problems so they can be fixed (I understand, not possible in this case) or contained.

AlexWossi, 2022-09-12:
Hallo! Everything was OK on Friday, but today (on Monday) I've lost megaTinyCore in my Boards list and also in Boards Manager at PC at work. Some errors reported, but I'll be able to send a copy of the Error-messages only tomorrow CET 10:00 am. At my home PC temporary disconnected from Internet, no problems found with the same version of Arduino. Both PCs run Win10. The problem is, I have to send devices to the Customer in 2 days...

SpenceKonde, 2022-09-12:
Unrelated, I screwed up the stupid boards manager json by putting a , after the last element of a list, which causes the garbage parser arduino uses to act like the file doesn't exist

commit d416df30 (2022-09-11): Fix #794

Path should be "api/Stream.h"