SpenceKonde/megaTinyCore #797: megaTinyCore lost in my Boards list

AlexWossi, 2022-09-12:
Hallo! Everything was OK on Friday, but today (on Monday) I've lost megaTinyCore in my Boards list and also in Boards Manager at PC at work. Some errors reported, but I'll be able to send a copy of the Error-messages only tomorrow CET 10:00 am. At my home PC temporary disconnected from Internet, no problems found with the same version of Arduino. Both PCs run Win10. The problem is, I have to send devices to the Customer in 2 days...

jsormaz, 2022-09-12:
Same issue here, i think this is the same as #796 
That issue is marked closed but still occurring for me.

dr-rockzo, 2022-09-12:
I had it too, in the json there's an extra comma. I was able to edit it, then set the file read only, restart Arduino and it came back 

[image]

SpenceKonde, 2022-09-12:
Totally unrelated to 796. 

this error was introduced and saved this morning. I didn't have time to test it, but no problem, because Im not pushing it to the servers until I get all the files ready and add the new release the board manager file. Expecting it couldn't cause any problems and with a ferocious time deadline, to leave the house and get back home I left it be in that state. I hadn't counted on it getting auto-pushed to the webserver, and more importantly, that despite being written in json, which is a spectacular format for this sort of thing and is normally parsed liberally, and which almost every parser of has no trouble with ah, creators who have not used a literal interpretation of the standard. 

On the otherhand, for whatever reason, Arduino gets gets tripped up even the tinies deviation from the standard. Every programing languages lets you put a trailing , on the last element of a loop - json doesn't. God forbid we find a good open source json parser that itn't corrupted by the slightest typo. And I've run performance comparisons and so on. s usually faster, but you could find perverse data that blew up the time requirement.

closed 2022-09-12 by SpenceKonde